mu-editor/mu #942: make.py vs. make | Inconsistent behaviour

tmontes, 2019-10-14:
While exploring #941, I checked out the [`use-venv`](https://github.com/mu-editor/mu/tree/use-venv) branch on both Windows and macOS.

Then, essentially, ran `make win64` on the Windows system, and `make macos` on macOS. The Windows build progressed and built a Windows Mu installer. Interestingly, the macOS make stopped because `flake8` did not pass. Here's a snippet:
```
flake8
./mu/interface/dialogs.py:19:1: F401 'os' imported but unused
./mu/interface/dialogs.py:20:1: F401 'sys' imported but unused
./mu/interface/dialogs.py:22:1: F401 'csv' imported but unused
./mu/interface/dialogs.py:23:1: F401 'shutil' imported but unused
./mu/interface/panes.py:20:1: F401 'site' imported but unused
./mu/modes/python3.py:22:1: F401 'mu.logic.VENV_DIR' imported but unused
./mu/modes/python3.py:22:1: F401 'mu.logic.VENV_PYTHON' imported but unused
make: *** [flake8] Error 1
```

Why this happens: both "make targets" include a dependency on `check` which, among other things, runs `flake8`. On macOS, running `make`, `flake8` returns a non-successful status and `make` aborts. On Windows something is going on, because `flake8` complained about the same exact things, but the `win64` target did not abort.

Looks like an easy fix.

PS: Just wanted to log this here. Back to helping out with #941. :)

tjg-global, 2019-10-15:
In short: we return a non-zero returncode from check if any of the checks fail, but we don't do anything with it!

I rather thought we did check, but either this has changed recently or I'm just wrong. I'll put a patch together

tjguk, 2020-12-17:
Closed by #1195

closed 2020-12-17 by tjguk