mu-editor/mu #944: Mu crashes on pressing REPL button in mode Python3

nuptownboy, 2019-10-18:
Mu version latest stable 1.0.2 not the alpha although that did same thing and I uninstalled and ran the stable and same issue
Press REPL button and prog crashes on Windows 10 latest version update last week on my windows surface pr 2 with adequate mem/storage as I have Jupyter note and ST3 and pycharm plus run Dragon Prof voice recognition and Autocad light so it isn't a resource issue and see log below
Grateful if you can see if it is something obvious as I have tried a more complex file and it runs ok just not REPL!
2019-10-18 12:51:25,428 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2019-10-18 12:51:25,429 - root:114(run) INFO: uname_result(system='Windows', node='NigelBlanchard', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 69 Stepping 1, GenuineIntel')
2019-10-18 12:51:25,429 - root:115(run) INFO: Python path: ['C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\nigel\\AppData\\Local\\Mu', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python', 'C:\\Users\\nigel\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-10-18 12:51:25,429 - root:116(run) INFO: Language code: en_GB
2019-10-18 12:51:25,582 - mu.logic:587(__init__) INFO: Setting up editor.
2019-10-18 12:51:25,582 - mu.logic:607(__init__) DEBUG: Creating directory: C:\Users\nigel\AppData\Local\python\mu\site-packages
2019-10-18 12:51:25,584 - mu.logic:353(get_admin_file_path) DEBUG: Creating admin file: C:\Users\nigel\AppData\Local\python\mu\settings.json
2019-10-18 12:51:25,585 - mu.logic:609(__init__) INFO: Settings path: C:\Users\nigel\AppData\Local\python\mu\settings.json
2019-10-18 12:51:25,587 - mu.logic:353(get_admin_file_path) DEBUG: Creating admin file: C:\Users\nigel\AppData\Local\python\mu\session.json
2019-10-18 12:51:25,588 - mu.logic:610(__init__) INFO: Session path: C:\Users\nigel\AppData\Local\python\mu\session.json
2019-10-18 12:51:25,588 - mu.logic:611(__init__) INFO: Log directory: C:\Users\nigel\AppData\Local\python\mu\Logs
2019-10-18 12:51:25,589 - mu.logic:612(__init__) INFO: Data directory: C:\Users\nigel\AppData\Local\python\mu
2019-10-18 12:51:25,626 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2019-10-18 12:51:25,626 - mu.logic:629(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code
2019-10-18 12:51:25,627 - mu.logic:637(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\images
2019-10-18 12:51:25,654 - mu.logic:643(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\fonts
2019-10-18 12:51:25,655 - mu.logic:646(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\sounds
2019-10-18 12:51:25,701 - mu.logic:652(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\music
2019-10-18 12:51:25,702 - mu.logic:658(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\templates
2019-10-18 12:51:25,733 - mu.logic:661(setup) DEBUG: Creating directory: C:\Users\nigel\mu_code\static
2019-10-18 12:51:28,053 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\nigel\mu_code
2019-10-18 12:51:28,054 - mu.logic:684(restore_session) INFO: Restoring session from: C:\Users\nigel\AppData\Local\python\mu\session.json
2019-10-18 12:51:28,055 - mu.logic:685(restore_session) DEBUG: {}
2019-10-18 12:51:28,055 - mu.logic:1221(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'debugger', 'pygamezero']
2019-10-18 12:51:42,061 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\nigel\mu_code
2019-10-18 12:51:43,696 - mu.logic:742(restore_session) INFO: Starting with blank file.
2019-10-18 12:57:38,318 - mu.logic:885(get_dialog_directory) INFO: Using path for file dialog: C:\Users\nigel\mu_code
2019-10-18 12:58:42,146 - mu.interface.main:288(get_save_path) DEBUG: Getting save path: C:/Users/nigel/Documents/pymbook/helloworld.py
2019-10-18 12:58:42,146 - mu.logic:984(check_for_shadow_module) INFO: Checking path "C:/Users/nigel/Documents/pymbook/helloworld.py" for shadow module.
2019-10-18 12:58:42,146 - mu.logic:950(save_tab_to_file) INFO: Saving script to: C:/Users/nigel/Documents/pymbook/helloworld.py
2019-10-18 12:58:42,146 - mu.logic:951(save_tab_to_file) DEBUG: print("Hello World")

print("We are here using mu editor to write the code.")

2019-10-18 12:58:45,121 - mu.interface.panes:691(start_process) INFO: Running script: c:\users\nigel\documents\pymbook\helloworld.py
2019-10-18 12:58:45,121 - mu.interface.panes:693(start_process) INFO: Running with interactive mode.
2019-10-18 12:58:45,121 - mu.interface.panes:696(start_process) INFO: Command args: []
2019-10-18 12:58:45,121 - mu.interface.panes:741(start_process) INFO: Unable to set Python paths. Python's USER_SITE not enabled. Check configuration with administrator.
2019-10-18 12:58:45,122 - mu.interface.panes:755(start_process) INFO: Running with environment variables: [('PYTHONPATH', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs;C:\\Users\\nigel\\AppData\\Local\\Mu;C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\python36.zip;C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\DLLs;C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\lib;C:\\Users\\nigel\\AppData\\Local\\Mu\\Python;C:\\Users\\nigel\\AppData\\Local\\python\\mu\\site-packages;C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions')]
2019-10-18 12:58:45,122 - mu.interface.panes:758(start_process) INFO: Working directory: C:/Users/nigel/Documents/pymbook
2019-10-18 12:58:45,122 - mu.interface.panes:763(start_process) INFO: Python path: ['C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\nigel\\AppData\\Local\\Mu', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python', 'C:\\Users\\nigel\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-10-18 12:58:45,122 - mu.interface.panes:789(start_process) INFO: Runner: C:\Users\nigel\AppData\Local\Mu\Python\pythonw.exe
2019-10-18 12:58:45,122 - mu.interface.panes:790(start_process) INFO: Args: ['-i', 'c:\\users\\nigel\\documents\\pymbook\\helloworld.py']
2019-10-18 12:59:24,060 - mu.modes.python3:230(toggle_repl) INFO: Toggle REPL on.
2019-10-18 12:59:24,062 - mu.modes.python3:58(start_kernel) INFO: ['C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\nigel\\AppData\\Local\\Mu', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\DLLs', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python\\lib', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\Python', 'C:\\Users\\nigel\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-10-18 12:59:24,063 - mu.modes.python3:63(start_kernel) INFO: Starting iPython kernel with user defined envars: {}
2019-10-18 12:59:24,063 - mu.modes.python3:76(start_kernel) INFO: REPL PYTHONPATH: C:\Users\nigel\AppData\Local\Mu\pkgs;C:\Users\nigel\AppData\Local\Programs\Python\Python37\;C:\Users\nigel\AppData\Local\python\mu\site-packages
2019-10-18 12:59:24,420 - traitlets:257(start_kernel) DEBUG: Starting kernel: ['C:\\Python\\Python37\\python.exe', '-m', 'ipykernel_launcher', '-f', 'C:\\Users\\nigel\\AppData\\Local\\Temp\\tmpu177a477.json']
2019-10-18 12:59:24,430 - traitlets:149(launch_kernel) ERROR: Failed to run command:
['C:\\Python\\Python37\\python.exe', '-m', 'ipykernel_launcher', '-f', 'C:\\Users\\nigel\\AppData\\Local\\Temp\\tmpu177a477.json']
    PATH='C:\\Users\\nigel\\AppData\\Local\\Mu\\pkgs\\PyQt5\\Qt\\bin;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\ProgramData\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\Google\\Google Apps Migration\\;C:\\Program Files\\Microsoft SQL Server\\130\\Tools\\Binn\\;C:\\Program Files\\WinMerge;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files (x86)\\Common Files\\Seagate\\SnapAPI\\;C:\\Program Files (x86)\\Calibre2\\;C:\\Calibre2\\;C:\\Program Files\\PowerShell\\6\\;C:\\Program Files\\Sublime Text 3;C:\\Users\\nigel\\MyPythonScripts;C:\\Users [... cut ...]
2019-10-18 14:01:49,249 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\nigel\AppData\Local\python\mu\Logs\mu.log
2019-10-18 14:02:12,936 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\nigel\AppData\Local\python\mu\Logs\mu.log

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2021-03-24:
@nuptownboy the latest beta released has update how it uses the iPython REPL, would you be able to install and check if this is still a problem?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2

carlosperate, 2021-04-26:
Hi @nuptownboy, just friendly ping to see if you'd be able to check this in the latest beta release?

carlosperate, 2021-05-04:
We haven't heard back since the ticket was opened, so I'll close this one as "very likely fixed" in the current releases.

Thank you for the report, and please feel free to reopen if you find that the issue is still present in the latest beta releases.

closed 2021-05-04 by carlosperate