mu-editor/mu #946: Incompatible mpy file - si5351.mpy

bobbryce, 2019-10-20:
Background?
Just purchased the adafruit feather m0 express board, plus the si5351 clock board.
New user to these products and also new to circuitpython and new to Mu editor! However I am an engineer of many years and familiar with Python programming.
Tried the first few sample circuitpython examples and found that the feather m0 board is operating as expected.

What am I trying to do?
I was following the examples given on programming the Si5351. I connected the si5351 board to the feather board. After typing the first three lines of the example code (three imports) I saved to 'code.py' and was expecting zero errors. However it appears that after adding the si5351 import I was getting a repeatable error (no errors if this is left out). 
This was unexpected as the files have all been downloaded in the last few days. I chose to download version 4 as the latest stable version. I have not tried the version 5 alpha.

Code?
import board
import busio
import adafruit_si5351

Error listing?
code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
  File "adafruit_si5351.py", line 36, in <module>
MpyError: Incompatible .mpy file. Please update all .mpy files. See http://adafru.it/mpy-update for more info

Environment?
Windows 10 Home build 1803
Mu editor v1.0.2
Libraries - v4.x
File being suspected - adafruit_si5351.mpy is version 1.1.4 and is 5.7kB in size

Logs?
Starting Mu 1.0.2
2019-10-20 17:03:56,927 - root:113(run) INFO: uname_result(system='Windows', node='DESKTOP-M5RT383', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 42 Stepping 7, GenuineIntel')
2019-10-20 17:03:56,927 - root:114(run) INFO: Python path: ['C:\\Users\\Bob\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Bob\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Bob\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Bob\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Bob\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-10-20 17:03:56,927 - root:115(run) INFO: Language code: en_GB
2019-10-20 17:03:57,204 - mu.logic:543(__init__) INFO: Setting up editor.
2019-10-20 17:03:57,205 - mu.logic:562(__init__) INFO: Settings path: C:\Users\Bob\AppData\Local\python\mu\settings.json
2019-10-20 17:03:57,205 - mu.logic:563(__init__) INFO: Session path: C:\Users\Bob\AppData\Local\python\mu\session.json
2019-10-20 17:03:57,205 - mu.logic:564(__init__) INFO: Log directory: C:\Users\Bob\AppData\Local\python\mu\Logs
2019-10-20 17:03:57,205 - mu.logic:565(__init__) INFO: Data directory: C:\Users\Bob\AppData\Local\python\mu
2019-10-20 17:03:57,303 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-10-20 17:03:58,341 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Bob\mu_code
2019-10-20 17:03:58,342 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\Bob\AppData\Local\python\mu\session.json
2019-10-20 17:03:58,342 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': ['E:\\code.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-10-20 17:03:58,342 - mu.logic:715(_load) INFO: Loading script from: E:\code.py
2019-10-20 17:03:58,345 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2019-10-20 17:03:58,346 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2019-10-20 17:03:58,346 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2019-10-20 17:03:58,346 - mu.logic:789(_load) DEBUG: import board
import busio
import adafruit_si5351
2019-10-20 17:03:59,319 - mu.logic:649(restore_session) INFO: Loaded files.
2019-10-20 17:03:59,319 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-10-20 17:03:59,319 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-10-20 17:03:59,357 - mu.logic:1150(change_mode) INFO: Workspace directory: E:\
2019-10-20 17:04:03,858 - mu.modes.base:227(find_device) INFO: Found device on port: COM6
2019-10-20 17:04:03,858 - mu.modes.base:228(find_device) INFO: Serial number: 650E4334E415E40502020243B0F191FF
2019-10-20 17:04:03,874 - mu.modes.base:278(add_repl) INFO: Started REPL on port: COM6
2019-10-20 17:04:03,874 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2019-10-20 17:04:09,034 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\Bob\AppData\Local\python\mu\Logs\mu.log
2019-10-20 17:05:17,349 - mu.logic:871(save_tab_to_file) INFO: Saving script to: E:\code.py
2019-10-20 17:05:17,349 - mu.logic:872(save_tab_to_file) DEBUG: import board
import busio
import adafruit_si5351
2019-10-20 17:05:23,915 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\Bob\AppData\Local\python\mu\Logs\mu.log

dhalbert, 2019-10-20:
This is a CircuitPython issue, not a Mu issue. Could you close this and ask your question in https://forums.adafruit.com or in discord: https://adafru.it/discord? It sounds like you may have updated the library but not the version of CircuitPython on the board, but let's discuss this in the right place. Thanks!

bobbryce, 2019-10-20:
Thanks for quick response. Apologies for raising in wrong place and so will close here. As a newbie user, I 'assumed' this environment was all coming from the same place and so just clicked on 'help' - oops.

closed 2019-10-20 by bobbryce

ntoll, 2019-10-20:
@bobbryce, sorry to hear you've been having problems. As @dhalbert says, this is a CircuitPython issue rather than anything to do with Mu (thank you so much, as always, Dan for your eagle eyed responses to these sorts of things cropping up in the Mu project). Best of luck with your project and I'm absolutely confident that the folks in the CircuitPython forums / Adafruit channels will be able to help you out -- they're an amazing community of dedicated folks.