mu-editor/mu #948: mu-editor not in the repos (debian, ubuntu)

OMGdaDPS, 2019-10-27:
at the bottom of the page for the install instructions for the raspberry pi, it shows that I can install mu-editor by running the command in my terminal sudo apt-get install mu-editor, but it returns with an error saying that it cant be found. why is this so? a few months ago i was able to install it via this method, but now i cant. I am currently on Ubuntu 18.04.3.

ntoll, 2019-10-28:
This is a problem you'd have to bring up with Debian and Ubuntu. We don't have control over what they keep in their repositories. Sorry.

OMGdaDPS, 2019-10-28:
Right on, thank you. Is there another way to install it without having to install python and THEN install mu? Like maybe a script?

OMGdaDPS, 2019-10-28:
Also, maybe the site should be updated? If the tutorial to install isn't fully correct. Maybe it is idk, I'm just starting out with python.

abixadamj, 2019-10-28:
On Debian and deb-based distros You have to install mu-editor usung pip3. I do this and it works like a charm.

`pip3 install mu-editor`

But with 1.1 version there will be another, not resolved yet problem -> https://github.com/mu-editor/mu/issues/822

I still have no good solution - only to install mu-editor in virtual env per user. I am working on this, but my time is very limited, so there is no huge progress ;-)

OMGdaDPS, 2019-10-28:
wow that is a pretty informative thread! i like the way that you guys think, and i cannot wait until i can productively discuss things like that with you. I seen somewhere in that thread, someone spoke of a .deb installer for mu-editor? i may have misunderstood. 
  On another note... how helpful would it be if someone snapped the package up, or made a flatpak out of it, or something along those lines? 
  My only issue (which isnt really an issue, its just me) with installing all of these different things, is that it is hard for me to keep track of all of it. i get confused on the whole thing about different versions of python for this, different versions of pip for that, and then virtualenv this, or everything is all over your system lol. 
 I am still trying to get a grip on all of this, and ive just now become (i feel) fully adjusted afaik lol) to the command line in Linux ( Ubuntu specifically). so that was a hurdle to cross over. 
 Anyway, i guess ill close this and leave you with these thoughts. thank you guys for actually taking the time and respond to me. i am honestly striving to be where you guys are, its just i just now started (about 2 months ago), and before that i had to learn how to use Linux lol.
  :peace_symbol:

closed 2019-10-28 by OMGdaDPS

ZanderBrown, 2019-10-28:
Mu is already deb'ed https://repology.org/project/mu-editor/versions just not for Ubuntu 18.04 (but as said this is a Ubuntu issue)

Flatpak is an interesting ongoing problem because as ever PtQt is being akward

OMGdaDPS, 2019-10-28:
hmm, thats interesting. i wonder why the decision was made to package it for the non lts release, and not the lts release. do you know why?

ZanderBrown, 2019-10-28:
Generally when a package is added it's only in new releases

But as said that's a Ubuntu decision and I'm from a Fedora world

OMGdaDPS, 2019-10-28:
wait. there is a snap for it? i just found it...

OMGdaDPS, 2019-10-28:
although, its version 1.0.1. from a dev named bennuttall

ZanderBrown, 2019-10-28:
Well that is the lastest release and you know [who that is right?](https://bennuttall.com/)

But this is possibly getting a tad off-topic now

OMGdaDPS, 2019-10-28:
oh man! nice. and yeah, agreed. thanks for the chat. glad to see that there is a (sort of) platform agnostic way to install this! i installed it via pip3 via `pip3 install mu-editor`, but couldnt seem to find it anywhere though.. 
  again, thanks for the help you all!

abixadamj, 2019-10-28:
/usr/local/bin/mu-editor