mu-editor/mu #950: Mu 1.1.0 Alpha issue with Win10 and GUIZero

SciPiGuy, 2019-11-01:
I am currently writing some Python using the GUIZero module.  This module was detected as present on my 1.0.2 install of Mu, however when testing out the 1.1.0 Alpha that I downloaded this morning, it gives me the following output:

**Traceback (most recent call last):
  File "c:\users\steve\mu_code\superheroguiv2.py", line 2, in <module>
    from guizero import (
ModuleNotFoundError: No module named 'guizero'**

The Python and log for this event is pasted below.

from guizero import (
    App,
    Text,
    ButtonGroup,
    Combo,
    PushButton,
    TextBox,
    CheckBox,
    Picture,
)

app = App(title="The Super Splendiferous...", height=800, width=600)
app.bg = "yellow"
app.font = "Stencil"
dark_light = 1

#Function definitions for your events go here.
def make_hero_name():
    skin = ""
    adjective = bgp_adjective.value
    colour = txt_colour.value
    animal = cmb_animal.value

    if chk_skin_fur.value == 1:
        skin = skin + "furry "
    if chk_skin_scaly.value == 1:
        skin = skin + "scaly "
    if chk_skin_slip.value == 1:
        skin = skin + "slippery "

    hero = adjective + " " + colour + " " + skin + animal
    lbl_output.value = "You are...The " + hero

def dark_light_mode():
    global dark_light
    if dark_light == 1:
        dark_light = dark_light * -1
        app.bg = "black"
    else:
        dark_light = dark_light * -1
        app.bg = "white"

# Your GUI widgets go here
btn_make_name = PushButton(app, text=" Dark\Light Mode ", command=dark_light_mode)

message1 = Text(app, text="Choose your amazing adjective:")
bgp_adjective = ButtonGroup(
    app, options=["Happy", "Flappy", "Snappy", "Clappy"], selected="Happy"
)

message2 = Text(app, text="Colour:")
txt_colour = TextBox(app)

message3 = Text(app, text="Favourite animal:")
cmb_animal = Combo(
    app,
    options=[
        "Anteater",
        "Bandicote",
        "Coyote",
        "Duck",
        "Eagle",
        "Frog",
        "Gerbil",
        "Hyena",
        "Iguana",
        "Jellyfish",
        "Kakapo",
        "Llama",
        "Mole",
        "Newt",
    ],
    selected="Anteater",
)

message4 = Text(app, text="Skin Type:")
chk_skin_fur = CheckBox(app, text="Furry ", command=make_hero_name)
chk_skin_scaly = CheckBox(app, text="Scaly ", command=make_hero_name)
chk_skin_slip = CheckBox(app, text="Slippery ", command=make_hero_name)

btn_make_name = PushButton(app, text=" Go! ", command=make_hero_name)
lbl_output = Text(
    app, text="Hero name being forged...um any moment now...", bg="red", color="yellow"
)

picture = Picture(app, image="superhero2.gif")

# Show the GUI on the screen, start listening to events.
app.display()

2019-11-01 11:43:12,019 - mu.interface.panes:691(start_process) INFO: Running script: c:\users\steve\mu_code\superheroguiv2.py
2019-11-01 11:43:12,020 - mu.interface.panes:693(start_process) INFO: Running with interactive mode.
2019-11-01 11:43:12,020 - mu.interface.panes:696(start_process) INFO: Command args: []
2019-11-01 11:43:12,020 - mu.interface.panes:741(start_process) INFO: Unable to set Python paths. Python's USER_SITE not enabled. Check configuration with administrator.
2019-11-01 11:43:12,020 - mu.interface.panes:755(start_process) INFO: Running with environment variables: [('PYTHONPATH', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\pkgs;C:\\Users\\steve\\AppData\\Local\\Muv110;C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\python36.zip;C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\DLLs;C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\lib;C:\\Users\\steve\\AppData\\Local\\Muv110\\Python;C:\\Users\\steve\\AppData\\Local\\python\\mu\\site-packages;C:\\Users\\steve\\AppData\\Local\\Muv110\\pkgs\\IPython\\extensions')]
2019-11-01 11:43:12,020 - mu.interface.panes:758(start_process) INFO: Working directory: C:/Users/steve/mu_code
2019-11-01 11:43:12,020 - mu.interface.panes:763(start_process) INFO: Python path: ['C:\\Users\\steve\\AppData\\Local\\Muv110\\pkgs', 'C:\\Users\\steve\\AppData\\Local\\Muv110', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\python36.zip', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\DLLs', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\Python\\lib', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\Python', 'C:\\Users\\steve\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\steve\\AppData\\Local\\Muv110\\pkgs\\IPython\\extensions']
2019-11-01 11:43:12,020 - mu.interface.panes:789(start_process) INFO: Runner: C:\Users\steve\AppData\Local\Muv110\Python\pythonw.exe
2019-11-01 11:43:12,021 - mu.interface.panes:790(start_process) INFO: Args: ['-i', 'c:\\users\\steve\\mu_code\\superheroguiv2.py']
2019-11-01 11:43:15,366 - mu.logic:1161(show_admin) INFO: Showing admin with logs from C:\Users\steve\AppData\Local\python\mu\Logs\mu.log
2019-11-01 11:43:15,366 - mu.logic:163(installed_packages) INFO: Packages found: []
2019-11-01 11:47:02,567 - mu.logic:1192(show_admin) INFO: No admin settings changed.
2019-11-01 11:52:53,302 - mu.logic:1161(show_admin) INFO: Showing admin with logs from C:\Users\steve\AppData\Local\python\mu\Logs\mu.log
2019-11-01 11:52:53,303 - mu.logic:163(installed_packages) INFO: Packages found: []

tjg-global, 2019-11-01:
@SciPiGuy I can confirm that this happens on 1.1.0a2. However when I install via Mu's own 3rd-party packages tab, it installs succesfully and the module imports without error.

If I read you correctly, you (re)installed guizero via pip from the command-line, so probably into your existing Python installation, not the one which comes packaged with Mu.

So: workaround seems to be using Mu's package installer, accessible via the bottom right-hand cog and the third-party packages tab.

What I don't know was whether removing guizero from the bundled packages was intentional.

martinohanlon, 2019-11-29:
Removing guizero from the bundled packages was intentional.

guizero (and other packages) was originally included, mainly, because there was no way of installing packages with the installed versions of Mu. With the new alpha you can install packages using the third-party packages installer.

carlosperate, 2020-11-09:
As Martin has mentioned with the inclusion of the third-party packages installer some packages have been removed, as they can now be installed by the users.

I'll close this issue as resolved, but please feel free to reopen if you have any questions or there is still an issue.

closed 2020-11-09 by carlosperate