mu-editor/mu #953: Directory for reading and writing files

bddicken, 2019-11-07:
Hi everyone,

I teach into CS with python at the UofArizona, and recommend Mu to my (several hundred) students. In some ways, Mu is an awesome intro-level IDE, but there are a few pain-points, and one in-particular is file I/O for beginner python programmers.

As the devs are probably aware of already, it seems that any program, regardless of what directory it actually exists in, has its CWD set to the mu_code directory within the home directory. This because problematic for both importing custom modules, and caused some headache for students when it comes to reading and writing files.

For instance, I created the directory `test` at the path `/Users/bddicken/Desktop/test`, and put `test.py` and `test.txt` in there:

### /Users/bddicken/Desktop/test/test.py

```python
for line in open('test.txt', 'r'):
    print(line)
```

### /Users/bddicken/Desktop/test/test.txt

```
one
two
three
```

However, this will *not* open up the test.txt file in the same directory. Rather, it will look for it in the mu_code folder. Even if I try adding the following at the beginning of `test.py`:

```
import os,sys
cwd = os.path.dirname(os.path.realpath(__file__))
sys.path.insert(0, cwd)
```

Still, it looks for it in mu_code. If/when students run into problems along these lines, I can say thinga like "make sure that both of the files are in the mu_code directory". However, there are still several problems with this:

(A) Some students are just not very familiar with file systems, and how to ensure that their files are in this directory precisely
(B) Some students have more than one mu_code folder on their computer
(C) Some students like to organize their work into other directories and sub-directories, and I don't blame them! I think that is good, except for problems such as this

My questions are:

* Is this behavior intentional?
    * If so, why was it chosen?
    * If so, perhaps it should be discussed.
* If it is the devs' desired behavior, could we add a switch for turning this on/off?

Thanks in advance.

tmontes, 2019-11-08:
> My questions are:
> 
> * Is this behavior intentional?
>   
>   * If so, why was it chosen?
>   * If so, perhaps it should be discussed.
> * If it is the devs' desired behavior, could we add a switch for turning this on/off?

Hi @bddicken,

Thanks for using Mu. :) Indeed the current 1.0.2 Mu operates that way and, pretty much like you describe, makes some kinds of operations, like file I/O, non-intuitive, given the fact that when running Mu sets its CWD to the "mu_code" directory, regardless of the directory where the program to run is stored.

That has changed. The current alpha behaves differently in that it sets its CWD to the directory where the program to run is stored. With that, your sample code does work as you expect. Also, importing sibling modules should also work as expected. You can grab it at the top of https://codewith.mu/en/download.

:)

ntoll, 2019-11-08:
Hi @bddicken, thank you for getting in touch. @tmontes  you beat me to the answer..! ;-)

I'm sorry to hear Mu hasn't helped in various ways when it comes to the file system related aspects of organising one's code. What @tmontes says is true (i.e. the current alpha has changed behaviour that fixes the niggles you have encountered). It would be really helpful if you could try the alpha and let us know things are working well for your teaching environment. We love to get feedback and take comments and suggestions from teaching colleagues especially seriously (you folks are the experts!).

I'll try to address your questions, because (for historical reasons) I think the answers are quite interesting:

Is this behaviour intentional? Yes.
Why? Raspberry Pi education team complained that (school age) students would always be losing their work, ergo could we just have a known default directory (mu_code) where the teacher could guarantee to find that lost piece of homework. We were happy to oblige but this decision brought to light all the problems you mention and so the changes were implemented for 1.1.

I hope this makes sense! All feedback most welcome!

bddicken, 2019-11-08:
Thanks @tmontes and @ntoll. At what point will 1.1.0 become the stable release? Is there a plan to make it stable by January 2020? That's when I'll have the next wave of students download mu.