mu-editor/mu #954: Feature request: mode for Lego EV3 micropython

askvictor, 2019-11-12:
I've just discovered that there's a micropython port for the Lego EV3 system; they've got a VS Code extension as the official dev environment, but it would be awesome to be able to code through mu for students/teachers already familiar with it. I'm unsure what the communication mechanism between the computer and EV3 is; hopefully a file interface is available. I'll try to get my hands on an EV3 set and have a play with it soon.

dlech, 2019-11-13:
Sorry, there is no file interface. It uses the SSH protocol for transferring files and other communications. It also uses zerconf (aka bonjour, mDNS, avahi, etc.) for device discovery.

The source code for the VS Code extension is [here](https://github.com/ev3dev/vscode-ev3dev-browser/) for reference.

askvictor, 2019-11-13:
Thanks - I'm trying to find out a bit more about how this functions - I thought it was separate from the EV3dev project, but I'm now guessing that the micropython build is more of an 'officially sanctioned' ev3dev distribution? Is the source code to the official extension (https://marketplace.visualstudio.com/items?itemName=lego-education.ev3-micropython&ssr=false#overview) in the ev3dev project repo?

So if it talks over SSH + mDNS, would it be feasible to package an ssh client into mu? paramiko might do the job, though may be finicky on the Windows side. Other alternatives are using native ssh binaries (again, Windows may be annoying as this would require an 'optional feature' install through the settings), or bundle an external ssh client with mu and interact with that? @dlech - how does the VSCode extension handle ssh?

dlech, 2019-11-13:
> more of an 'officially sanctioned' ev3dev distribution?

Indeed it is.

>  Is the source code to the official extension in the ev3dev project repo?

No, however it doesn't do much. It just creates a new project folder from a template and contains an HTML version of the documentation. It uses the ev3dev extension for everything else (ev3dev extension is a dependency of the official extension).

> would it be feasible to package an ssh client into mu? paramiko might do the job

paramiko looks like it would do the job. 

> how does the VSCode extension handle ssh?

It uses what is probably the NodeJS equivalent of paramiko - a pure javascript implementation of the SSH protocol.

askvictor, 2019-11-13:
Thanks, good to know! Out of interest, do you know what the micropython port for the ev3 is based on? Is it the micropython-unix port along with the pybricks library?

dlech, 2019-11-13:
Basically, yes.

aivarannamaa, 2019-11-17:
> paramiko

I've heard good things about this alternative: https://github.com/ParallelSSH/ssh2-python