mu-editor/mu #956: Crashes on save/load

hmltech, 2019-11-23:
Help!

Mu crashes whenever I attempt to save or load a file on 5 laptops that I use for computer classes. They are all running Windows 10. I'll click save/load or double click the tab, the file explorer will pop up, things will hang for a moment, and then Mu will close. I have another pair of laptops running Windows 7 that do not have this issue.

I've tried uninstalling and reinstalling, but it persists.

This happens regardless of whether a device is plugged in when I open Mu and regardless of whether I've written any code or am just saving a blank file to test if it's crashing.

Here is the log from my most recent session:

`2019-11-23 14:18:36,787 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-11-23 14:18:36,802 - root:113(run) INFO: uname_result(system='Windows', node='Lab2-PC', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-11-23 14:18:36,802 - root:114(run) INFO: Python path: ['C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-11-23 14:18:36,802 - root:115(run) INFO: Language code: en_US
2019-11-23 14:18:39,358 - mu.logic:543(__init__) INFO: Setting up editor.
2019-11-23 14:18:39,374 - mu.logic:562(__init__) INFO: Settings path: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\settings.json
2019-11-23 14:18:39,389 - mu.logic:563(__init__) INFO: Session path: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\session.json
2019-11-23 14:18:39,389 - mu.logic:564(__init__) INFO: Log directory: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\Logs
2019-11-23 14:18:39,389 - mu.logic:565(__init__) INFO: Data directory: C:\Users\Public.Lab2-PC\AppData\Local\python\mu
2019-11-23 14:18:39,421 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-11-23 14:18:50,114 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Public.Lab2-PC\mu_code
2019-11-23 14:18:50,145 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\session.json
2019-11-23 14:18:50,145 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-11-23 14:18:50,145 - mu.logic:649(restore_session) INFO: Loaded files.
2019-11-23 14:18:50,145 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-11-23 14:18:50,145 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-11-23 14:18:51,243 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-11-23 14:18:52,138 - mu.interface.main:723(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.
2019-11-23 14:18:52,138 - mu.interface.main:724(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 C:\Users\Public.Lab2-PC\mu_code

...to store your code.
2019-11-23 14:19:00,212 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Public.Lab2-PC\mu_code
2019-11-23 14:19:00,822 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: C:\Users\Public.Lab2-PC\mu_code
2019-11-23 14:19:24,171 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-11-23 14:19:24,171 - root:113(run) INFO: uname_result(system='Windows', node='Lab2-PC', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-11-23 14:19:24,171 - root:114(run) INFO: Python path: ['C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Public.Lab2-PC\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-11-23 14:19:24,171 - root:115(run) INFO: Language code: en_US
2019-11-23 14:19:24,281 - mu.logic:543(__init__) INFO: Setting up editor.
2019-11-23 14:19:24,281 - mu.logic:562(__init__) INFO: Settings path: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\settings.json
2019-11-23 14:19:24,281 - mu.logic:563(__init__) INFO: Session path: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\session.json
2019-11-23 14:19:24,281 - mu.logic:564(__init__) INFO: Log directory: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\Logs
2019-11-23 14:19:24,281 - mu.logic:565(__init__) INFO: Data directory: C:\Users\Public.Lab2-PC\AppData\Local\python\mu
2019-11-23 14:19:24,312 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-11-23 14:19:24,735 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Public.Lab2-PC\mu_code
2019-11-23 14:19:24,735 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\Public.Lab2-PC\AppData\Local\python\mu\session.json
2019-11-23 14:19:24,735 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'adafruit', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-11-23 14:19:24,735 - mu.logic:649(restore_session) INFO: Loaded files.
2019-11-23 14:19:24,735 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-11-23 14:19:24,735 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-11-23 14:19:25,205 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-11-23 14:19:25,236 - mu.interface.main:723(show_message) DEBUG: Could not find an attached Adafruit CircuitPython device.
2019-11-23 14:19:25,236 - mu.interface.main:724(show_message) DEBUG: Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here:

 C:\Users\Public.Lab2-PC\mu_code

...to store your code.
2019-11-23 14:20:04,782 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Public.Lab2-PC\mu_code
2019-11-23 14:20:06,413 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\Public.Lab2-PC\AppData\Local\python\mu\Logs\mu.log
`

tmontes, 2019-11-24:
Hi @hmltech,

From the logs you shared, you are running Mu 1.0.2 in Adafruit's CircuitPython mode. AFAICT, Mu failed to find an attached device, per the logged messages:

> Python files for Adafruit CircuitPython devices are stored on the device. Therefore, to edit these files you need to have the device plugged in. Until you plug in a device, Mu will use the directory found here

Do you confirm that to be the case? More, did Mu display a dialog box like the following, telling you about that?

[image]

It should not crash, however, and, much like the dialog informs, the code should be saved to the "mu_code" directory, instead.

I tried reproducing the behaviour you described, by running Mu 1.0.2 on a Windows 10 VM with no attached CircuitPython device. It was installed from the installer package at https://github.com/mu-editor/mu/releases/download/1.0.2/mu-editor_1.0.2_win64.exe:

* Lauch Mu.
* Selected "Adafruit CircuitPython" mode.
* The informational dialog box above was displayed.
* Dismissed it by clicking the "OK" button.
* Typed in a single line of code in the "untitled" editor: `a_var = 42`.
* Clicked the "Save" button.
* Typed in a name "gh_issue_956.py" and clicked the "Save" button.
* The file was saved successfully.

I could not reproduce your failure, thus. I also tried loading the file back, which also worked fine.

There must be something going on with those systems. Here are a few ideas that may help understanding what's going on:

* Does file saving/loading work on those systems if Mu is running in "Python 3" mode, instead?
  I suspect it might fail too, but we'll never know until we try it.
* Is the `C:\Users\Public.Lab2-PC\` profile directory a "plain" directory stored on the local disk? Or could it be somehow network backed? (I seem to recall people having trouble when trying to save files onto OneDrive/GoogleDrive backed directories: could this be the case?)
* Could there be any kind of third party software component running on those systems that is somehow interacting badly with Mu? (antivirus? file synchronization? backups?).
* Not being aware of any particular changes that could affect this, I wonder if running the 1.1.0.alpha2 version would result in a different (hopefully better!) behaviour. If you're willing, you could try it out by grabbing the installer at https://github.com/mu-editor/mu/releases/download/1.1.0-alpha.2/mu-editor_1.1.0-alpha.2_win64.exe.

Thanks for reporting the failure. If possible, please share more details. Hopefully, we'll get to the bottom of this and improve Mu to provide a better end-user experience. :)

hmltech, 2019-11-27:
Yes, I didn't have a device plugged in at the time and I got the dialog box.

I tested in Python 3 mode, and it also crashed on both attempted load and attempted save. Exact same behavior as in CircuitPython mode.

Yes, that's just a plain directory.

I don't THINK there's any third party software that could be causing a problem. The closest I could come to thinking of anything is that these laptops do run Deep Freeze, but they're all currently thawed. I inherited these laptops from my predecessor, though, so I will dig around to see if there's something I'm missing.

I uninstalled 1.0.2 and installed 1.1.0.alpha2, and there was no change in behavior.

hmltech, 2019-11-27:
Additional info from playing around with it more:

I managed to get save to work ONCE after changing the directory Mu was saving to. Since then, Mu can successfully restore the session with my test file, and autosaving or hitting the save button work fine. However, trying to load a different file or doubleclicking the file name to try to save as continue to make it crash.

I know this is kind of a bizarre problem, so thank you for any help you can give me.

ntoll, 2019-11-27:
@hmltech can you attach the logs from the most recent crashes to this issue please (click on the cog icon to display them). That'll tell us exactly what the error was.

My feeling is that there's something odd on your computers since Mu doesn't crash like this for anyone else. The log should be revealing though. Thanks!

@tmontes I think we should seriously think of adding a, "Mu has encountered a serious problem" dialog that'll create all the details needed for generating a comprehensive crash report. Thoughts..?

hmltech, 2019-11-27:
It wouldn't surprise me if the issue was the computers... they've gone through many hands, unfortunately.

Since my last comment, I also got the same works-once-then-doesn't experience launching from the command line??? Weird.

Here's the log, which I think includes both success-then-failure sequences:

Starting Mu 1.0.2
2019-11-27 12:58:13,091 - root:113(run) INFO: uname_result(system='Windows', node='Lab2-PC', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-11-27 12:58:13,091 - root:114(run) INFO: Python path: ['C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-11-27 12:58:13,091 - root:115(run) INFO: Language code: en_US
2019-11-27 12:58:13,204 - mu.logic:543(__init__) INFO: Setting up editor.
2019-11-27 12:58:13,204 - mu.logic:562(__init__) INFO: Settings path: C:\Users\Lab2\AppData\Local\python\mu\settings.json
2019-11-27 12:58:13,204 - mu.logic:563(__init__) INFO: Session path: C:\Users\Lab2\AppData\Local\python\mu\session.json
2019-11-27 12:58:13,204 - mu.logic:564(__init__) INFO: Log directory: C:\Users\Lab2\AppData\Local\python\mu\Logs
2019-11-27 12:58:13,204 - mu.logic:565(__init__) INFO: Data directory: C:\Users\Lab2\AppData\Local\python\mu
2019-11-27 12:58:13,230 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-11-27 12:58:13,230 - mu.logic:590(setup) DEBUG: Creating directory: C:\Users\Lab2\Documents\Mu\images
2019-11-27 12:58:13,234 - mu.logic:597(setup) DEBUG: Creating directory: C:\Users\Lab2\Documents\Mu\fonts
2019-11-27 12:58:13,234 - mu.logic:600(setup) DEBUG: Creating directory: C:\Users\Lab2\Documents\Mu\sounds
2019-11-27 12:58:13,276 - mu.logic:605(setup) DEBUG: Creating directory: C:\Users\Lab2\Documents\Mu\music
2019-11-27 12:58:13,713 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Lab2\Documents\Mu
2019-11-27 12:58:13,713 - mu.logic:627(restore_session) INFO: Restoring session from: C:\Users\Lab2\AppData\Local\python\mu\session.json
2019-11-27 12:58:13,713 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-11-27 12:58:13,713 - mu.logic:649(restore_session) INFO: Loaded files.
2019-11-27 12:58:13,713 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-11-27 12:58:13,713 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-11-27 12:58:14,151 - mu.logic:680(restore_session) INFO: Starting with blank file.
2019-11-27 12:58:14,166 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Lab2\Documents\Mu
2019-11-27 12:58:16,204 - mu.logic:1077(show_admin) INFO: Showing logs from C:\Users\Lab2\AppData\Local\python\mu\Logs\mu.log
2019-11-27 12:58:47,868 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: C:\Users\Lab2\Documents\Mu
2019-11-27 12:59:02,710 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: 
2019-11-27 12:59:07,303 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: C:\Users\Lab2\Documents\Mu
2019-11-27 12:59:11,989 - mu.interface.main:263(get_save_path) DEBUG: Getting save path: C:/Users/Lab2/Documents/Mu/test.py
2019-11-27 12:59:11,989 - mu.logic:905(check_for_shadow_module) INFO: Checking path "C:/Users/Lab2/Documents/Mu/test.py" for shadow module.
2019-11-27 12:59:11,989 - mu.logic:871(save_tab_to_file) INFO: Saving script to: C:/Users/Lab2/Documents/Mu/test.py
2019-11-27 12:59:11,989 - mu.logic:872(save_tab_to_file) DEBUG: # Write your code here :-)
a_var = 42
2019-11-27 12:59:16,660 - mu.logic:1053(quit) DEBUG: Session: {'theme': 'day', 'mode': 'python', 'paths': ['C:\\Users\\Lab2\\Documents\\Mu\\test.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2019-11-27 12:59:16,660 - mu.logic:1054(quit) DEBUG: Saving session to: C:\Users\Lab2\AppData\Local\python\mu\session.json
2019-11-27 12:59:16,660 - mu.logic:1068(quit) INFO: Quitting.

2019-11-27 12:59:20,523 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2019-11-27 12:59:20,523 - root:113(run) INFO: uname_result(system='Windows', node='Lab2-PC', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 94 Stepping 3, GenuineIntel')
2019-11-27 12:59:20,523 - root:114(run) INFO: Python path: ['C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\Python\\python36.zip', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\Python', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs', 'C:\\Users\\Lab2\\AppData\\Local\\Mu\\pkgs\\IPython\\extensions']
2019-11-27 12:59:20,523 - root:115(run) INFO: Language code: en_US
2019-11-27 12:59:20,617 - mu.logic:543(__init__) INFO: Setting up editor.
2019-11-27 12:59:20,617 - mu.logic:562(__init__) INFO: Settings path: C:\Users\Lab2\AppData\Local\python\mu\settings.json
2019-11-27 12:59:20,617 - mu.logic:563(__init__) INFO: Session path: C:\Users\Lab2\AppData\Local\python\mu\session.json
2019-11-27 12:59:20,617 - mu.logic:564(__init__) INFO: Log directory: C:\Users\Lab2\AppData\Local\python\mu\Logs
2019-11-27 12:59:20,617 - mu.logic:565(__init__) INFO: Data directory: C:\Users\Lab2\AppData\Local\python\mu
[... cut ...]
2019-11-27 13:14:54,195 - mu.logic:789(_load) DEBUG: # Write your code here :-)
a_var = 42
2019-11-27 13:14:54,634 - mu.logic:649(restore_session) INFO: Loaded files.
2019-11-27 13:14:54,635 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2019-11-27 13:14:54,635 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2019-11-27 13:14:54,653 - mu.logic:1150(change_mode) INFO: Workspace directory: C:\Users\Lab2\Documents\Mu
2019-11-27 13:15:06,320 - root:112(run) INFO:

tmontes, 2019-11-28:
@hmltech,

Thanks for your patience. There is clearly something very unexpected going on. Please help clarifying: you say you succeeded in saving to the non-default "C:\Users\Lab2\Documents\Mu\" directory once, in a file called "test.py". The logs show that. What about the following scenarios:

* Loading "C:\Users\Lab2\Documents\Mu\test.py", changing it, and saving it. Works or crashes?
* Saving a new editor tab to the non-default "C:\Users\Lab2\Documents\Mu\" directory, under a new name, say "another.py". Works or crashes?
* If both cases lead to a crash, what happens if you save a new editor tab to a new non-default directory, say, "C:\Users\Lab2\Documents\AnotherMu\"? Works or crashes?
* If that works, what happens when you save a new editor tab as a new file in that new non-default directory? Works or crashes?

Please try to be as accurate as possible in your checks: I'm trying to guide us in figuring out if any behavioural patterns emerge from this. :)

More:

I looked a the code that handles file saving in Mu: it is prepared to handle both `OSError` and `UnicodeEncodeError` exceptions. I suspect that some other exception is being raised in your case that unfortunately results in a Mu crash. With that in mind, I suggest we try to force that exception via the REPL. Could you please try:

* Launch Mu (ideally 1.1.0.alpha2).
* Close any open editor tabs.
* Ensure it is in Python 3 mode.
* Bring up the REPL, by clicking the REPL button.
* Once in the REPL, please type in each of the following lines, one at a time:

  * `import os`
  * `os.getcwd()` -- expected to output `'C:\Users\Lab2\mu_code'`.
  * `f = open('issue956.txt', 'wt')` -- may either work, generate an exception, or crash Mu.
  * Continue if the previous line produced no output on the REPL, meaning it worked.
  * `f.write('hello there')` -- as above, this may work, generate an exception, or crash Mu.
  * If it works, 11 is returned, the number of "letters" written. Continue if that's the case.
  * `f.close()` -- again, this may work, generate an exception, or crash Mu.

I expect this procedure to fail at some point: hopefully with an exception that helps understanding what might be going on. A potentially useful variation of the above is trying to save/create a file in a different directory: for that, you could use a full path in the first argument of the `open` function call, keeping in mind that a raw string should be used to ensure the back-slashes aren't interpolated by Python -- `r'C:\Users\Lab2\Documents\Mu\issue956.txt'`, for example.

Again, thanks for your patience and help in diagnosing this and improving Mu for other users.

> @tmontes I think we should seriously think of adding a, "Mu has encountered a serious problem" dialog that'll create all the details needed for generating a comprehensive crash report. Thoughts..?

@ntoll, this is a very interesting and useful idea. We should create an issue for that! :)