mu-editor/mu #958: PyQtChart wheel change causes error in win_installer.py

carlosperate, 2019-11-24:
I've been looking at the current AppVeyor failures on recent PRs like this one, that haven't touched the CI config:

https://ci.appveyor.com/project/carlosperate/mu/builds/29084193

I'm writing here my current findings as I might not be able to raise a PR today, but maybe somebody else could continue looking into it. 

The error:

```
Checking for wheel availability at PyPI.
...
- PyQtChart==5.12.0 Traceback (most recent call last):
  File "win_installer.py", line 300, in <module>
    run(bitness, repo_root)
  File "win_installer.py", line 264, in run
    installer_exe = create_pynsist_cfg(venv_python, repo_root, pynsist_cfg)
  File "win_installer.py", line 196, in create_pynsist_cfg
    wheels = pypi_wheels_in(requirements)
  File "win_installer.py", line 155, in pypi_wheels_in
    if any(r.package_type == "wheel" for r in releases):
TypeError: 'NoneType' object is not iterable
```

The code:
https://github.com/mu-editor/mu/blob/ee600ff16753194f33e39975662aba438a8f5608/win_installer.py#L152-L155

Neither the `win_installer.py` code or the dependency version have changed for quite a long time, so something external had to change to produce this error now.

I tried to see the output from that snippet of code locally and indeed we are getting a `None` value returned:

```python
>>> import yarg
>>> package = yarg.get("PyQtChart")
>>> package.release("5.12.0")
>>>
```

Tracked the last successful build and I notice that the `pip freeze` output for PyQtChart has changed!
From `5.12` (as indicated in setup.py) to `5.12.0`.
https://ci.appveyor.com/project/carlosperate/mu/builds/28868244

And that does work:

```python
>>> package.release("5.12")
[<Release 5.12>, <Release 5.12>, <Release 5.12>, <Release 5.12>]
```

I checked the version of pip in case something has changed in the `freeze` command, but we are using the same version in both builds.

So then I check PyPI and found something fishy:

[image]

Version 5.12 was released in February, yet the wheels were uploaded 2 days ago? 🤔

I can't confirm, but I think RiverBank has deleted the "...cp35.cp36.cp37..." wheels and uploaded new  "...cp35.cp36.cp37.cp38..." for the Python 3.8 release, and in doing so changed the wheel metadata identifying itself as version `5.12.0` (I have checked and the wheels do have this version string) instead of the original `5.12`.

carlosperate, 2019-11-24:
https://www.riverbankcomputing.com/pipermail/pyqt/2019-November/042353.html

carlosperate, 2019-11-25:
The 5.12 wheels cannot really be updated, so it looks like the only solution on our side is to update PyQt as done in https://github.com/mu-editor/mu/pull/959.

tmontes, 2019-11-28:
@carlosperate, thanks for your detailed analysis. I've just pushed PR #961 with an improved `win_installer.py` that now iteratively tries to find "funny" wheels such as the case of `PyQtChart` now.

Note how it figured it out and properly logged it in [this AppVeyor build log line](https://ci.appveyor.com/project/carlosperate/mu/builds/29193757#L689).

While I agree that updating Mu's dependencies should be done, our tooling was also somewhat fragile and comes out a little bit better with this. :)

I'd love to hear your feedback on the PR.

carlosperate, 2019-11-29:
Great, thanks Tiago! I'm out of time today, but I'll give it a review tomorrow 👍

dybber, 2020-06-13:
I guess this can be closed, now where #961 is merged?

carlosperate, 2020-06-14:
Yes, thanks everyone!

closed 2020-06-14 by carlosperate