mu-editor/mu #967: Just a minor french spelling mistake to correct in mu.mo "Ecris ton programme"

feria35, 2019-12-05:
Hello,

I am trying to correct the welcome message of Mu Editor
which has French error translation in the sentence : '# Write your program here ;-)'

The problem is : Mu Editor write " # Ecrit ton programme ici ;-) "
But it should be  " # Ecris ton programme ici ;-) "  because it is the imperative.

The correction is well made in 'mu.po' ( mu/mu/locale/fr/LC_MESSAGES/mu.po )
but it is not in **'mu.mo' line 82** ( mu/mu/locale/fr/LC_MESSAGES/mu.mo ).
And Windows seems to load 'mu.mo'.

There is no pen to correct it in Github, so that's why I am writing an issue.

My OS is : Windows 10 (64 bits)
I tried Mu Editor (32 bit and 64 bit version) : mu-editor_1.1.0-alpha and mu_2019-12-03_01_12_win-pip_27ddb8f_64bit, mu_2019-12-03_01_15_win-pip_27ddb8f_32bit

Thank you.
Feria35

rossjones, 2019-12-17:
@feria35 Normally the .mo file is compiled from the .po file, but this is a step that most likely happens during a release. The .po is fixed in github so this should show up as fixed in the next release.

feria35, 2019-12-18:
@rossjones Thank you very much for your reply.
Well, I think it will be repaired with a pull request (because at the moment the error is still there). Anyway, thank you very much for your contribution.

And I have another question : 
As the  .po file is correct since the the May 15, 2018 and the last release date is much later (Jul 5, 2019) ; why the .mo file has not been recompiled ?
Thank you very much for your explaination.

merwok, 2020-05-13:
I noticed this too, then many more mistakes.
I am updating the French translation and will send a PR.

carlosperate, 2021-12-27:
Hi @feria35 @merwok, it's been a while and a lot of the Mu strings have changed, but the project is now ready again to accept translations, if you'd like to volunteer a translation that would be really welcomed!

More info in:
- https://github.com/mu-editor/mu/discussions/1931

carlosperate, 2022-03-02:
Hi @feria35 @merwok, we'll synchronise translations in this discussion:
- https://github.com/mu-editor/mu/discussions/1931

To keep things tidy I'll close this issue, but if you can still provide French updates to the translation it would be really appreciated!

closed 2022-03-02 by carlosperate