mu-editor/mu #968: Autocompletion in method

JLFra, 2019-12-06:
Could you add autocompletion in the argument of methods please. Autocompletion works, but not in methods?
Thanks
Julien

ZanderBrown, 2019-12-06:
Could you clarify " in the argument of methods "?

Do you mean between the ( and ) of a function/method call? In which case I assume you're looking for suggestions of local variables which would require the autocompletion to be far cleverer than it is at the moment

JLFra, 2019-12-07:
Yes, exactly. Autocompletion works perfectly outside method, but not inside.

carlosperate, 2020-11-09:
Is this because the autocompletion is showing the function docs?

[image]

In this case, pressing escape closes the current info and suggest a variable:
[image]

carlosperate, 2022-07-16:
This behaviour is defined by QScintilla, so not much we can do within Mu itself, so we can consider the current behaviour acceptable.

closed 2022-07-16 by carlosperate