mu-editor/mu #972: Permissions Issue on Catalina

DaveKT, 2019-12-10:
Hello, 

Ran into a permissions issue this AM.

1. Downloaded the latest stable version of mu.
2. Try launching app on OS X Catalina 10.15.1 (19B88)
3. App Crashes.

From the mu.log file. Looks like it trying to write to a protected directory.

Starting Mu 1.0.2
2019-12-10 11:54:10,191 - root:112(run) INFO: uname_result(system='Darwin', node='Gauss.local', release='19.0.0', version='Darwin Kernel Version 19.0.0: Thu Oct 17 16:17:15 PDT 2019; root:xnu-6153.41.3~29/RELEASE_X86_64', machine='x86_64', processor='i386')
2019-12-10 11:54:10,191 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2019-12-10 11:54:10,191 - root:114(run) INFO: Language code: en_US
2019-12-10 11:54:10,389 - mu.logic:543(__init__) INFO: Setting up editor.
2019-12-10 11:54:10,390 - mu.logic:562(__init__) INFO: Settings path: /Users/DTassara/Library/Application Support/mu/settings.json
2019-12-10 11:54:10,390 - mu.logic:563(__init__) INFO: Session path: /Users/DTassara/Library/Application Support/mu/session.json
2019-12-10 11:54:10,390 - mu.logic:564(__init__) INFO: Log directory: /Users/DTassara/Library/Logs/mu
2019-12-10 11:54:10,390 - mu.logic:565(__init__) INFO: Data directory: /Users/DTassara/Library/Application Support/mu
2019-12-10 11:54:10,401 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2019-12-10 11:54:10,402 - mu.logic:590(setup) DEBUG: Creating directory: /Volumes/CIRCUITPY/images
2019-12-10 11:54:10,412 - root:94(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 144, in run
    editor.setup(setup_modes(editor, editor_window))
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/logic.py", line 591, in setup
    os.makedirs(images_path)
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/os.py", line 220, in makedirs
PermissionError: [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/images'

DaveKT, 2019-12-10:
Quick update. So I had installed the new version of mu with a Circuit Playground connected (hence the path). As soon as I disconnected the playground I was able to launch MU. seems there is an order of operations -> 1. Launch mu. 2. Connect Circuit Playground. Closing since I'm not certain if this is unique to me or not.

closed 2019-12-10 by DaveKT

sid-patwardhan, 2020-09-12:
This issue is occurring for me. Were you able to find a fix?
(using mu-editor for the first time)

carlosperate, 2020-09-17:
@sid-patwardhan were you able to open Mu with the circuitpython device disconnected?
If not, could you open a new issue describing all the steps you took and any Mu log you could retrieve?

sid-patwardhan, 2020-09-18:
@carlosperate, I did manage to resolve this. Turns out that the issue (for me) is likely related to a security feature of the macOS (Catalina). By default applications appear to not have read or write permissions into a mounted volume (and certain other directories ... even a directory listing through Terminal popped up an OS dialog, requiring me to approve access to removable volume).

I think mu-editor was unable to create 'images' directory on the mounted volume CIRTUITPY at startup because of this access restriction... and was crashing right away. I have managed to allow access to mounted volumes, and the mu-editor is working correctly now. 👍