mu-editor/mu #974: Proposed enhancement: API for third-party plugins

aroberge, 2019-12-15:
Mu's approach of focusing on a positive user experience, including ease of installation and providing a self-contained solution is, in my opinion, the near-perfect choice for its target audience. However, I do believe that it comes at a great cost in that any addition (including new modes) must be carefully vetted and require a great deal of time from the developers to incorporate into the existing code and to maintain afterwards.  This is a burden that, imo, should be lessen as much as possible.

Furthermore, and with apologies for the negative phrasing, it does result in a bit of a "walled garden" which makes it difficult for others to experiments with possible additions to Mu's fonctionality. Investing time to develop some potential addition to Mu with no guarantee that the work will not be "wasted" as it might never get approval for integration in Mu is a discouraging thought; maintaining a separate fork of a project like Mu is not something viable, in my opinion.

I believe that a useful addition would be to have an API for plugins, each of which could be installed separately from Mu.  The learning curve for creating a plugin would likely be much gentler than that required to add to the existing code for Mu.

Since Mu comes with its own pre-packaged Python installation, I do realize that this might present some challenges; to be honest, I am not sure of what would be the best approach to do this, and I am not suggesting that creating such an API would be an easy task for Mu developers.  However, if it were possible to create such plugins, I am confident that tentative solutions for at least one of the existing issues (https://github.com/mu-editor/mu/issues/963) would be promptly investigated! ;-)

**Perhaps** the minimal functionality that would be required initially for most plugin writers is simply the following:
1. Ability to select a different Python installation with command line arguments provided by the user.  This would become automatically appended as a new available "mode", but using the default UI for Python 3.
2. Run the code as a subprocess, capturing stdout and stderr and redirecting them to the normal "window" at the bottom.

I'm wondering if such an approach would also provides an "easy" solution for https://github.com/mu-editor/mu/issues/941.

ntoll, 2019-12-15:
@aroberge yes. I agree with everything you state and we'd had (informal) discussions about adding a "plugin" story for version 2.0 (then the walled garden problem goes away -- I feel this keenly).

There's a LOT of work going on / to be done to update the installation story, to make Mu play nicely with virtualenvs and the addition of further "official" modes (web mode especially is highly prized by teachers). The trick is to weave in the plugin story into this complicated project management.

aroberge, 2019-12-15:
I wish I could help, contributing to Mu's code is almost certainly beyond my ability level.