mu-editor/mu #978: Mu_Editor does not open after Mac OS Catalina upgrade

GrNeOp, 2019-12-27:
after upgrading to MacOS Catalina, Editor stop working altogether  can't even open the app.
it just blink once at the dock... and that is it...
MacBook pro mid 2018 A1990

tmontes, 2019-12-27:
Hello @GrNeOp,

We're sorry to hear that Mu is no longer working for you, after your OS upgrade.

Here are a few things you could try, in order of growing complexity, to help us diagnose your case and, hopefully, make Mu work again soon:

1. Try reinstalling Mu 1.0.2:
   * Use the Finder to locate Mu and delete it.
   * Download the 1.0.2 version from https://codewith.mu/en/download
   * Install it again.

2. Try reinstalling Mu 1.1.0.alpha2:
   * As above but download version 1.1.0.alpha2 instead.

3. Share any Mu logs that may be present in your system:
   * In the Finder menu click "Go" > "Go To Folder...".
   * Type in "~/Library/Logs/mu".
   * Please share the most recent file you find there.

4. Launch Mu from a Terminal:
   * Launch the Terminal application.
   * Type in `/Applications/mu-editor.app/Contents/MacOS/mu-editor` and hit RETURN.
   * If Mu does not work (I don't expect it will), probably some output text will be displayed beneath that. Please share it back here.

Again, thanks for using Mu and for helping us figuring out what may need fixing. :-)

Tairraos, 2019-12-28:
My Mu1.0.2 is worked fine.
MacOS Catalina 10.15.2 / MacBook Pro (15-inch, 2017)

carlosperate, 2020-03-22:
As we have issue https://github.com/mu-editor/mu/issues/997 for what it looks like the same problem, I'll close this one to avoid having two parallel threads.

Even though this one is older, other issues are already linking to https://github.com/mu-editor/mu/issues/997, so it makes more sense to keep the conversation there.

@GrNeOp if you have a chance to follow @tmontes instructions in [his comment above](https://github.com/mu-editor/mu/issues/978#issuecomment-569263240) and post the results to https://github.com/mu-editor/mu/issues/997 it would help us diagnose and fix the issue.

Thanks!

closed 2020-03-22 by carlosperate