mu-editor/mu #979: Debugging windows don't retain formatting changes

Jake-theSnake, 2019-12-28:
Hi! New to Mu! New to programming. Just started using the "Debugger" in Python 3 mode. When the debugger stops at a breakpoint, the "Names" and "Values" pane opens up to the right. So far so good. But, I like to resize the "Names" and "Values" column (so there's more space between the two), and expand any expandable items (like a dictionary, for instance), so that I can see everything quickly and clearly at a glance. However, when I click on the "Step Over" button to execute the next line of code, my reformatting goes away. The columns are now scrunched up again, and any expanded items get collapsed. So I end up reformatting that window each time I step through my code.
Is there a way to retain my formatting choices while stepping through my code?
Windows 10, btw, if that makes a difference.

carlosperate, 2019-12-30:
Hi Jake,

Thank you for your report, I think this might be related to https://github.com/mu-editor/mu/issues/490, do you think it's the same issue?

Jake-theSnake, 2019-12-30:
Similar, but not exactly the same.  The problem I'm seeing is similar, in
that it occurs in "Debugging" mode. However, my issue is twofold:

1) While in "Debugging" mode, if I resize the "Name" and "Value" columns
(to put some whitespace in between the visible elements), when I step
through the program, the resizing I do disappears, and the columns
return to their default state.

2) Again while in debugging mode, if I expand any elements that have
sub-elements (like a dictionary, for example), once I step through the
program, those expanded elements collapse.

See the attached video for reference.

I am actually not seeing the issue where the whole "Debugging" pane resizes
itself.

devdanzin, 2021-01-22:
The issue happens because we clear `Window.debug_model` on [each update](https://github.com/mu-editor/mu/blob/50347029a43a30e605c629008cf5edb02daa27c5/mu/interface/main.py#L715):
```python
        self.debug_model.clear()
        self.debug_model.setHorizontalHeaderLabels([_("Name"), _("Value")])
```

We could instead just delete the rows we added, and that keeps the headers and width choices:
```python
        self.debug_model.setHorizontalHeaderLabels([_("Name"), _("Value")])
        # Remove rows so we keep the same column layouts if manually set
        while self.debug_model.rowCount() > 0:
            self.debug_model.removeRow(0)
```

We can also remember what column widths were last used and use them in next `DebugInspector` invocation (reset on quit).

Keeping expanded elements open seems a bit more involved. The simplest option is to keep a list of seen dict names, record their expanded status before updating and expanding them anew when we create the new debug content. I just don't know how to get the expanded status yet. It'll lead to false positives for name clashes, but I guess tracking identity might be a bit much.

devdanzin, 2021-01-22:
Here's a demo of fixing 1) and 2) above. 

https://user-images.githubusercontent.com/74280297/105441841-28646180-5c48-11eb-835e-8b7b6a1702c0.mp4

Sorry about the black border!

commit b98e1f28 (2021-01-31): Merge pull request #1244 from devdanzin/fix_debug_layout_reset

Remember debugger layout and expanded items (fixes #979)
closed 2021-01-31 by ntoll