mu-editor/mu #981: CI issue

Tairraos, 2020-01-02:
All recent PR failed, caused same windows build issue, i test on local windows, but everything passed. Could you check what's happen?

carlosperate, 2020-06-11:
Hi @Tairraos,

Is this the issue you mean from the Windows CI?

```
- PyQtChart==5.12.0 Traceback (most recent call last):
  File "win_installer.py", line 300, in <module>
    run(bitness, repo_root)
  File "win_installer.py", line 264, in run
    installer_exe = create_pynsist_cfg(venv_python, repo_root, pynsist_cfg)
  File "win_installer.py", line 196, in create_pynsist_cfg
    wheels = pypi_wheels_in(requirements)
  File "win_installer.py", line 155, in pypi_wheels_in
    if any(r.package_type == "wheel" for r in releases):
TypeError: 'NoneType' object is not iterable
Check
Clean
Tidy
flake8
coverage
Building 64-bit Windows installer
Command exited with code 1
```

This was due to an edge case on how PyQt was repackaged. This PR addresses the problem: https://github.com/mu-editor/mu/pull/961
 
Alternatively the error will probably go away if update the version of PyQt used.

I'll close this issue as there was already a pending PR to fix it, but please do reopening if you were referring to a different problem.

closed 2020-06-11 by carlosperate

carlosperate, 2020-06-11:
The Windows CI should be working again now, thanks @tjguk and @tmontes!