mu-editor/mu #982: Flashing fails after upgrade to Catalina

deejaygraham, 2020-01-02:
Mu editor had been working fine with version 1.0.9 (I think). Upgraded Mac OS recently and editor still works fine but flashing a program fails with an error message: "There was a problem flashing the microbit, please do not disconnect during flashing..."

Tried several different programs that had worked in the past with the same result. Tried upgrading to newer versions of mu and now running 1.1.0.alpha.2 with the same behaviour. Microbit shows up as a usb device as usual in Finder.

Thanks.

Log is below:

2020-01-02 20:37:08,815 - mu.modes.microbit:154(flash) INFO: Preparing to flash script.
2020-01-02 20:37:08,825 - mu.modes.microbit:164(flash) DEBUG: Python script:
2020-01-02 20:37:08,825 - mu.modes.microbit:165(flash) DEBUG: b'from microbit import *\n\nwhile True:\n\n    if button_a.is_pressed():\n        display.show(Image.HAPPY)\n        pin13.write_digital(1)\n        pin14.write_digital(0)\n        sleep(500)\n        pin13.write_digital(0)\n        pin14.write_digital(0)'
2020-01-02 20:37:08,841 - mu.modes.microbit:212(flash) INFO: Path to micro:bit: /Volumes/MICROBIT
2020-01-02 20:37:08,842 - mu.modes.base:261(find_device) INFO: Found device on port: cu.usbmodem14102
2020-01-02 20:37:08,842 - mu.modes.base:262(find_device) INFO: Serial number: 9900023431864e450019100a0000001400000000cc4d28bd
2020-01-02 20:37:08,842 - mu.modes.microbit:217(flash) INFO: Serial port: /dev/cu.usbmodem14102
2020-01-02 20:37:08,842 - mu.modes.microbit:218(flash) INFO: Device serial number: 9900023431864e450019100a0000001400000000cc4d28bd
2020-01-02 20:37:08,842 - mu.modes.microbit:243(flash) INFO: Checking target device.
2020-01-02 20:37:09,385 - mu.modes.microbit:247(flash) INFO: {'sysname': 'microbit', 'nodename': 'microbit', 'release': '1.0.1', 'version': 'micro:bit v1.0.1+b0bf4a9 on 2018-12-13; MicroPython v1.9.2-34-gd64154c73 on 2017-09-01', 'machine': 'micro:bit with nRF51822'}
2020-01-02 20:37:09,388 - mu.modes.microbit:266(flash) INFO: Board MicroPython: 1.0.1
2020-01-02 20:37:09,388 - mu.modes.microbit:268(flash) INFO: Mu MicroPython: 1.0.1
2020-01-02 20:37:09,391 - mu.modes.microbit:420(copy_main) INFO: Copying main.py onto device
2020-01-02 20:37:09,391 - mu.modes.microbit:430(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'from microbit import *\\n\\nwhile True:\\n\\n    if button_a.is_pressed(')", "f(b'):\\n        display.show(Image.HAPPY)\\n        pin13.write_digital')", "f(b'(1)\\n        pin14.write_digital(0)\\n        sleep(500)\\n        pi')", "f(b'n13.write_digital(0)\\n        pin14.write_digital(0)')", 'fd.close()']
2020-01-02 20:37:10,429 - mu.modes.microbit:365(flash) WARNING: Could not copy file to device.
2020-01-02 20:37:10,430 - mu.modes.microbit:366(flash) ERROR: Could not enter raw REPL.
2020-01-02 20:37:10,430 - mu.modes.microbit:367(flash) INFO: Falling back to old-style flashing.
2020-01-02 20:37:10,473 - mu.modes.microbit:75(run) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-01-02 20:37:10,484 - mu.modes.microbit:447(flash_failed) ERROR: [Errno 1] Operation not permitted: '/Volumes/MICROBIT/micropython.hex'
2020-01-02 20:37:10,532 - mu.interface.main:763(show_message) DEBUG: There was a problem flashing the micro:bit.
2020-01-02 20:37:10,532 - mu.interface.main:764(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2020-01-02 20:38:12,579 - mu.logic:1161(show_admin) INFO: Showing admin with logs from /Users/derek/Library/Logs/mu/mu.log
2020-01-02 20:38:12,580 - mu.logic:163(installed_packages) INFO: Packages found: []

aidancuckoo, 2020-01-04:
I second that - please fix this bug!

deejaygraham, 2020-01-05:
Yep that worked for me. 

Open a terminal, sudo /Applications/mu-editor.app/Contents/MacOS/mu-editor, type password. 

You may get a prompt that terminal needs to access to a remote drive (the microbit). Flashes and works as before.

filipejsbrandao, 2020-01-11:
I'm having the same issue and using the terminal is not solving anything. There is one difference in my case from the OP, I installed MU after having macOS Catalina. Here is my log:

-----------------

Starting Mu 1.0.2
2020-01-11 00:14:58,287 - root:112(run) INFO: uname_result(system='Darwin', node='iMac-de-Filipe.local', release='19.2.0', version='Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64', machine='x86_64', processor='i386')
2020-01-11 00:14:58,287 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2020-01-11 00:14:58,287 - root:114(run) INFO: Language code: en_US
2020-01-11 00:14:58,785 - mu.logic:543(__init__) INFO: Setting up editor.
2020-01-11 00:14:58,786 - mu.logic:562(__init__) INFO: Settings path: /Users/Filipe/Library/Application Support/mu/settings.json
2020-01-11 00:14:58,786 - mu.logic:563(__init__) INFO: Session path: /Users/Filipe/Library/Application Support/mu/session.json
2020-01-11 00:14:58,786 - mu.logic:564(__init__) INFO: Log directory: /Users/Filipe/Library/Logs/mu
2020-01-11 00:14:58,786 - mu.logic:565(__init__) INFO: Data directory: /Users/Filipe/Library/Application Support/mu
2020-01-11 00:14:58,798 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-01-11 00:14:58,976 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/Filipe/mu_code
2020-01-11 00:14:58,988 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/Filipe/Library/Application Support/mu/session.json
2020-01-11 00:14:58,988 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['/Users/Filipe/mu_code/Hello.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2020-01-11 00:14:58,989 - mu.logic:715(_load) INFO: Loading script from: /Users/Filipe/mu_code/Hello.py
2020-01-11 00:14:58,989 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-01-11 00:14:58,989 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-01-11 00:14:58,989 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2020-01-11 00:14:58,989 - mu.logic:789(_load) DEBUG: from microbit import *
display.scroll("I am a micro:bit.")
2020-01-11 00:14:59,068 - mu.logic:649(restore_session) INFO: Loaded files.
2020-01-11 00:14:59,068 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-01-11 00:14:59,068 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-01-11 00:14:59,078 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/Filipe/mu_code
2020-01-11 00:15:04,766 - mu.modes.microbit:239(flash) INFO: Preparing to flash script.
2020-01-11 00:15:04,766 - mu.modes.microbit:249(flash) DEBUG: Python script:
2020-01-11 00:15:04,766 - mu.modes.microbit:250(flash) DEBUG: b'from microbit import *\ndisplay.scroll("I am a micro:bit.")'
2020-01-11 00:15:04,812 - mu.modes.microbit:297(flash) INFO: Path to micro:bit: /Volumes/MICROBIT
2020-01-11 00:15:04,813 - mu.modes.base:227(find_device) INFO: Found device on port: cu.usbmodem142102
2020-01-11 00:15:04,813 - mu.modes.base:228(find_device) INFO: Serial number: 9901000051114e4500637001000000200000000097969901
2020-01-11 00:15:04,813 - mu.modes.microbit:302(flash) INFO: Serial port: /dev/cu.usbmodem142102
2020-01-11 00:15:04,813 - mu.modes.microbit:303(flash) INFO: Device serial number: 9901000051114e4500637001000000200000000097969901
2020-01-11 00:15:04,813 - mu.modes.microbit:328(flash) INFO: Checking target device.
2020-01-11 00:15:05,955 - mu.modes.microbit:364(flash) WARNING: Could not detect version of MicroPython.
2020-01-11 00:15:05,993 - mu.modes.microbit:381(flash) INFO: Flashing new MicroPython runtime onto device
2020-01-11 00:15:16,403 - mu.modes.microbit:505(copy_main) INFO: Copying main.py onto device
2020-01-11 00:15:16,403 - mu.modes.microbit:515(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', 'f(b\'from microbit import *\\ndisplay.scroll("I am a micro:bit.")\')', 'fd.close()']
2020-01-11 00:15:17,468 - mu.modes.microbit:532(flash_failed) ERROR: Could not enter raw REPL.
2020-01-11 00:15:17,478 - mu.interface.main:723(show_message) DEBUG: There was a problem flashing the micro:bit.
2020-01-11 00:15:17,478 - mu.interface.main:724(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
2020-01-11 00:15:36,770 - mu.logic:1077(show_admin) INFO: Showing logs from /Users/Filipe/Library/Logs/mu/mu.log

ntoll, 2020-01-26:
Hi folks, this isn't so much a problem with Mu, as a difference in behaviour in Catalina.

Basically, Catalina requires applications to have "permission" to access various parts of the filesystem and the way Mu interacts with the micro:bit  in this sense is problematic. However, there is a fix (which I've included in all the documentation), which I'll repeat here so folks can find it easily via this issue.

To have the correct permissions, you need to run Mu as an administrator user, by opening up the `terminal` program and typing the following commands (which may require you to re-enter your password):

```
$ cd /Applications
$ sudo open mu-editor.app
```

This it'll just work.

Since this is a working solution I'll close this issue. Thanks for taking the time to report it..!

closed 2020-01-26 by ntoll

tmontes, 2020-02-09:
Dear All,

In the context of #1000, @DaveSprague came up with a macOS Catalina procedure that may be suitable in this case --  instead of having to run Mu via `sudo`, he suggests adjusting System Preferences to grant additional permissions to Mu.

I wonder if that procedure -- [documented here](https://github.com/mu-editor/mu/issues/1000#issuecomment-583336514) -- also fixes the micro:bit flashing failure on macOS Catalina.