mu-editor/mu #984: If you stop a running Python3 program from mu, atexit-registered code is not run

romilly, 2020-01-14:
Some code (especially I2C code) requires a clean shutdown to avoid the need for a reboot. It seems that atexit-registered code is only run when a Python session ends normally, and it is not invoked after SIGKILL.

Zander suggested that the Stop button should send SIGTERM and only invoke SIGKILL after a timeout. That sounds as if it would fix the problem.

tmontes, 2020-01-14:
@romilly,

Thanks for raising the issue. AFAICT, addressing it in full will be a two step-process:

* The current "stop" button -- assuming Python 3 mode -- does indeed send a SIGKILL signal to the running process on Unix-like systems. In fact, what the button actually does is call [`QProcess.kill`](https://doc.qt.io/qt-5/qprocess.html#kill) which does that under the covers. Having it send a SIGTERM instead seems as simple as replacing this [`QProcess.kill` call](https://github.com/mu-editor/mu/blob/master/mu/modes/python3.py#L216) with a [`QProcess.terminate`](https://doc.qt.io/qt-5/qprocess.html#terminate) -- this is a Mu fix.

* Then, you will probably need to update your code to ensure that, not only the current `atexit` handing is run when you expect it to run, **but also** that it runs whenever the process running your program gets the said SIGTERM. For that, you will need to register a signal handler with [`signal.signal`](https://docs.python.org/3/library/signal.html#signal.signal) -- this is a fix to your program.

Note that most of this does not apply if the platform is Windows. But from your words I gather that not to be the case.

Hoping this helps addressing the issue.

tmontes, 2020-01-14:
PS: I took the approach I mentioned above for a quick spin. It turns out, not unexpectedly, that there are some closing/cleaning-up race conditions that need to be addressed for a consistent and predictable "stop" behaviour. Key motives:

* The sub-process is running under `python -I program.py`. This means that even if a handled SIGTERM stops your program, the interactive interpreter will still be running. That, in turn, will have to be, again, terminated.

* Then there is the possibility that before exiting either your program or `python -I` produce output (or not). This output will probably need to be consumed, even if not displayed in the running output pane.

Sounds like a "fun" little project, improving this!

romilly, 2020-01-14:
I'm now clearer on the problem, and possible solutions. Thanks!

I will probably have to wrap the underlying Pimoroni library (or sweet-talk
them into changing it) in order to hide the issue from the beginners who
will be writing/running code that uses the library in mu.

I may be able to take a look at the changes to mu myself later this month,
in which case I will attempt to submit a PR. Testing it is going to be fun
manual!