mu-editor/mu #985: Add REPL timeout / increase default.

WayneKeenan, 2020-01-17:
If you are reporting a bug, we would like to know:

* What you were trying to do,

List files on an ESP32 board on macOS.

* What steps you took to make this happen,

Click Files

* What you expected to happen,

Show files 
* What actually happened,

`OSError: Could not enter raw REPL.`

* Why this difference is problematic (it may not be a bug!),

Can't see files.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

This is occurring for me because there is code in boot.py that is taking a few seconds (5ish) to complete before presenting the REPL prompt. When I deleted boot.py all is good.

Is there a way for a user to configure the timeout before Mu gives up?
Or can the default be increased please?

related:  https://github.com/mu-editor/mu/issues/858

WayneKeenan, 2020-01-17:
Simple `boot.py` test cases:

Works:
```
from utime import sleep_ms
sleep_ms(500)
```

Fails:
```
from utime import sleep_ms
sleep_ms(1000)
```

dybber, 2020-02-26:
I had the problem with M5Stack devices.

15 seconds is perhaps a bit long to wait for an error message to appear - in the case where the problem is not boot.py

Does your boot.py really take that long to run? I think we shouldn't increase it more than necessary. If the error appears again for someone else we can increase it some more.

Perhaps we should consider indicating in the UI that Mu is waiting with some kind of Busy indicator

WayneKeenan, 2020-02-26:
15 secs was a bit finger in the air.  I think the `boot.py` script took about ~6 seconds (when connecting to my main wifi AP), but the connection setup time varied between different access points and there's environmental factors that introduces variable delays to connection setup time, sometimes.

Perhaps defaulting it, say to 5, and adding it as a configurable setting in the settings file or as a UI config option, rather than having to (re)build mu to change it.

dybber, 2020-03-24:
Hi @WayneKeenan 

Okay, actually I think you shouldn't put wifi-connectivity stuff in the boot.py, but in main.py. Let me try to explain why.

Both boot.py and main.py run on startup but serves different purposes:
 - boot.py is for setting up various configuration options, that should be the same across any program you might wish to write
 - main.py is for executing your startup program: that is running your actual application

In the case of wifi, what if the wifi-connection is lost at some point? Your application should be aware of the wifi-connectivity and reconnect if necessary. I mentioned M5Stack's firmware has the same issue, but I also think it's problematic that they put so much stuff in their boot.py, which should have been in their main.py.

It seems that Mu is unable to interrupt boot.py, but is able to interrupt the execution of main.py, so it will work if you do it that way at least. But perhaps you have a different situation than I'm imagining. I just want to make sure it's really necessary to introduce this delay for users.

WayneKeenan, 2020-03-28:
Isn't wanting to have the Wifi connection to be the same for all or any `main.py` that might be uploaded a valid use case?

e.g. teachers could setup boot.py with WiFi and then students only need to update `main.py` (keeps maintenance and docs simple)

However, the real reason I would like WiFi setup in `boot.py` is because the `boot.py` I am dealing with is not mine but part of the ESP32 port of EduBlocks, and that's the way that program is currently structured.

WayneKeenan, 2020-03-28:
Increasing the timeout doesn't inherently add a delay to all `boot.py` sctipts, does it?

federicomm, 2020-07-01:
The Raw REPL problem, that I have, is exactly the one described in the (*)reference : ESP32 early LOG "cpu_start: Pro cpu up."
The board that I am testing on Windows10 with Mu 1.1.0.alpha.2 is an Espressif ESP32 WROOM-32.
To confirm that we can have an delay around 16 sec before the IOError message ( see microfs.py 
- enter_raw_repl(serial) ) I have added at the start and at the end of the read_until instruction 
the print of the local time - as you can easily view inspecting one of the attachment) and also a print 
of the processed serial data that give us an immediate vision of what happening before the 
micropython boot. The micropython boot.py and the main.py weight for a negligible amount of time while the
ESP32 RAM-TEST involves the processing of about 2900 bytes by the read_until method - This is, in my opinion,  the main
the reason of the IOError('could not enter raw repl')  .
The work-around (see the picture) has been the  stop of the IDF DRAM test reading at its start.
from microfs.py
...............................................
PY2 = sys.version_info < (3,)
RAM_TEST = True
...............................................
def enter_raw_repl(serial):  (**) see my W600-PICO vs Mu 1.1.0.alpha.2 post
        ...............................................
        serial.write(b'\r\x01') # ctrl-A: enter raw REPL
		
        # added for ESP32 - start - Stop reading of IDF RAM Test
        if RAM_TEST == True:
            data = read_until(serial, 1, b'cpu_start: Pro cpu up.')
            time.sleep(1)
            serial.write(b'\r\x01') # ctrl-A: enter raw REPL
        # added for ESP32 - end
		
        data = read_until(serial, 1, b'raw REPL; CTRL-B to exit')
		...............................................

The board raw REPL is achieved in  about 3 sec.
To do this an RAM_TEST boolean flag is set to True. Clearly this solution ( flag True) represent 
a penalty for ESP32 boards without the IDF DRAM test and mainly for all the ESP8266 board that 
with the flag set to False in about 1sec shown the device file list while with the flag set to 
True needed from 8 to 16 sec.

The other sulution for the board involves the use of the IDF tool to modify in the sdkconfig the
 debug mode - 
- the flash of a dummy RTOS example application
- the flash erasing
- the reflash of an micropython image.

FileSystemNavigator
I suggest the following two changes( see attacment for details), to make easier the interactions between the device an local files panes.
- Device pane - Sorted file list by Name
( micro.fs - def ls(serial=None): 
-Local pane- Add the home Dir to the pane head. ( panes.py class FileSystemPane(QFrame):

(*)ref :
https://www.esp32.com/viewtopic.php?t=7881

(**)
https://github.com/mu-editor/mu/issues/1052

[image]
[image]
[image]
[image]

dybber, 2020-09-07:
I don't know if this makes sense in your situation, but just for the record if anyone else reads this, I want to explain how I use wifi in MicroPython when I have kids doing projects that require e.g. collecting data from API's, logging data, etc.

I've created a small wifi-module, that takes care of the details, which makes it possible for students to just do:
```python
import wifi

while True:
    wifi.connect("wifi-ssid", "wifi-password")
    dostuff()
    wifi.disconnect()
    time.sleep(10)
```

In this way the wifi-setup is not something completely magical that just happens, but the students have to do it themselves, but it's still so simple that everyone can understand it, and in this way the Wifi-connection will happen in `main.py`, not in `boot.py`.

I wouldn't use this module for anything but teaching, as it prints some debug-prints, to allow students to see whether connection succeeds, but for educational purposes this simple module works fine:

**wifi.py**
```python
import time
import network

# Connect to a specific wifi network
def connect(essid, password, timeout=30000):
    wifi = network.WLAN(network.STA_IF)
    wifi.active(True)
    if not wifi.isconnected():
        print("Connecting to WiFi network...")
        wifi.connect(essid, password)
        # Wait until connected
        t = time.ticks_ms()
        while not wifi.isconnected():
            if time.ticks_diff(time.ticks_ms(), t) > timeout:
                wifi.disconnect()
                print("Timeout. Could not connect.")
                return False
        print("Successfully connected to " + essid)
        return True
    else:
        print("Already connected")
        return True

def disconnect():
    wifi = network.WLAN(network.STA_IF)
    wifi.disconnect()

def isconnected():
    wifi = network.WLAN(network.STA_IF)
    return wifi.isconnected()

def ipaddress():
    wifi = network.WLAN(network.STA_IF)
    ip, subnetmask, gateway, dns = wifi.ifconfig()
    return ip
```

WayneKeenan, 2020-09-08:
Please see my previous comment: 
`However, the real reason I would like WiFi setup in boot.py is because the boot.py I am dealing with is not mine but part of the ESP32 port of EduBlocks, and that's the way that program is currently structured.`

dybber, 2020-10-01:
Fixed by @WayneKeenan's PR #1123, thanks!

closed 2020-10-01 by dybber