mu-editor/mu #987: Up Arrow in REPL for Previous Command Problem

the-stanely, 2020-01-21:
In REPL on Windows 10, problem with up arrow for previous command.

First up arrow is fine.  Subsequent keystrokes don't erase the previous command before printing the new one.  Instead, what looks like terminal control characters are printed, followed by the previous command.

The character sequence is like: [10D

[nnD  where nn is the decimal number of characters to the beginning of line.  So it looks like the terminal is asked to left-delete the previous nn characters.  But the sequence is not being interpreted in windows and instead is printed.

Here's an example:

print(mac)[10Dprint mac[9Dmac = ubinascii.hexlify(network.WLAN().config('mac'),':').decode()

dybber, 2020-04-10:
Hi @the-stanely 

I have used some afternoons working on this and I believe I have solved the issue.

Please try if you can reproduce the error when using the following branch:
https://github.com/dybber/mu/tree/bugfix/replcursor_movement

The branch also contains fixes for several other issues regarding cursor movement in the MicroPython REPL, so it should now be possible to do selections, move the cursor with the mouse, and in general it should just be much more stable. It seems to work on Mac, but would be nice if you could test on Windows.

I still need to write test case, before I can send it in as a pull request though.

the-stanely, 2020-04-12:
Yes, the up/down cursor keys work well and I can not reproduce the errors.  Thank you.  The up arrow lets me go back to 8 most recent commands.  Down arrow goes forward through the commands just like it's supposed to.  This is great, because the left/right arrows work correctly now too!

I did find one unexpected feature.  When you press ENTER when inside the REPL command line being edited, a CR/LF appears to be inserted before sending the line to the interpreter.  This is unexpected because when editing a line in CMD or sh, a CR sends the line for processing and doesn't actually insert a CR/LF but adds it to the end of the line.  Here's a REPL example.  Note the edit cursor ( | ) position...

>>> print(8); print(9); print(1|0); print(11)

Pressing <Enter> yields:

>>> print(8); print(9); print(1
0); print(11)8
9
0
11

That's all I found.  Otherwise it works great.  I also tested back and forward deletes and no problems there.

dybber, 2020-04-13:
Thanks, Stan! I can reproduce the CR/LF problem and will try to find time for a debugging session later this week. I want to push forward to get the ESP-mode out of alpha.

dybber, 2020-04-13:
I just pushed a fix, where I move the cursor to the end of the terminal before sending carriage return. I hope that doesn't break anything else.

mindforger, 2020-05-18:
i am not sure if this is the same issue i am having right now, using win10 mu editor x64 

in the repl console i have artifacts showing up messing up my line when using arrow key up and down

> np[0] = (255,255,255,128)
> np[0] = (255,255,255,128)[25Dnp] = (255,255,255,128)[23Dnp] = (0,255,255,128)[21Dnp.write()

how can i get a compiled version or compile a version for myself of this branch to test?

dybber, 2020-05-18:
Hi @mindforger, not completely the same issue, but I fixed that as well in the above issue, so it should be fixed as soon as it's merged and we make a new release.

mindforger, 2020-05-18:
will be waiting then :) (if i know how i'd tested it)

closed 2020-09-29 by dybber

the-stanely, 2020-09-29:
👍

PR #1026 (2020-09-29): Support mouse movement, selection, and fix several bugs in micro:bit & ESP-mode REPL
The micro:bit, CircuitPython and ESP-mode REPL have suffered from several bugs, which this pull requests addresses:
 - Partial reception of multibyte VT100 commands
 - Support for selecting text with SHIFT + cursors
 - Support for mouse movement and mouse text selection
 - General stability

This fixes #987 and #840.