mu-editor/mu #989: After closing running Pygame Zero window "Stop" button doesn't change back to "Play"

MrCreightonD, 2020-01-23:
I clicked Play to run my Pygame Zero game. 
I closed the Pygame Zero game using the windows X button.
Game window closes.
Stop button is still presented in the Mu interface instead of Play button.

tjg-global, 2020-01-23:
I admit that this is slightly confusing, but it is intended behaviour. (FWIW it makes very slightly less sense in Pygame modes than in Python / micro:bit modes).

The idea is that, even once the program has finished you can still explore variables and/or look at tracebacks and so on. So the [Stop] button really means [Close the lower window] (but that doesn't really fit).

I'm fairly sure -- without actually bothering to look -- that we have an outstanding feature request to suggest changing the button to read something like "Close" in this situation. Feel free to hunt for it and add your support to the idea.

tjg-global, 2020-01-23:
(By the way, thank you for taking the trouble to report this rather than just grumble under your breath and not bother to say anything...)

dybber, 2020-09-05:
I think @tjg-global assessment here is right, this is the intended behavior, and I will thus close this issue.

closed 2020-09-05 by dybber