mu-editor/mu #991: Problem with Tidy

abixadamj, 2020-01-27:
When I click Tidy butto, the mu crashes:

```
adasiek@mint-desktop:~/mu$ ./run.py 
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 9796, resource id: 13473228, major code: 40 (TranslateCoords), minor code: 0
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 9967, resource id: 13920875, major code: 40 (TranslateCoords), minor code: 0
Traceback (most recent call last):
  File "/home/adasiek/mu/mu/logic.py", line 1618, in tidy_code
    from black import format_str, FileMode, PY36_VERSIONS
ModuleNotFoundError: No module named 'black'

```
Linux Mint 19.3, mu clone from this repo 1 hour ago.

ntoll, 2020-01-27:
Which version of Mu, have you recently updated the required packages..? We've added some new dependencies (including black).

tjguk, 2020-01-27:
In my experience, this happens when you have an existing dev environment and then don't do a `pip install --upgrade ...`

abixadamj, 2020-01-27:
OK, missing packages....
```

 "flake8 >= 3.7.8",
    "pycodestyle >= 2.5.0, < 2.6.0",
    "pyflakes >= 2.1.0, < 2.2.0",
    "pyserial==3.4",
    "qtconsole==4.4.3",
    "pgzero==1.2",
    "appdirs>=1.4.3",
    "semver>=2.8.0",
    "nudatus>=0.0.3",
    'black>=18.9b0;python_version > "3.5"',
    "Flask==1.0.2",

Everything works ;-)
```

closed 2020-01-27 by abixadamj

ntoll, 2020-01-27:
I do this **all the time** :smiley: