mu-editor/mu #993: pip3 install -U mu-editor on Linux Mint 193. Error.

abixadamj, 2020-01-29:
I just did:
`root@mint-desktop:~#pip3 install -U mu-editor`

An error occured:
```
Collecting qscintilla==2.11.4 (from mu-editor)
  Could not find a version that satisfies the requirement qscintilla==2.11.4 (from mu-editor) (from versions: 2.9.4, 2.10, 2.10.1, 2.10.2, 2.10.3, 2.10.4, 2.10.7, 2.10.8, 2.11, 2.11.1, 2.11.2, 2.11.3)
No matching distribution found for qscintilla==2.11.4 (from mu-editor)
```

When I try to install manually, I see only version 2.11.3 is avaible:
```
root@mint-desktop:~# pip3 install -U qscintilla
Collecting qscintilla
  Downloading https://files.pythonhosted.org/packages/66/9c/858a1dd55e6a98c617779164f790d30c06ce1ce140735d76e647a4ca4289/QScintilla-2.11.3-cp35.cp36.cp37.cp38-abi3-manylinux1_x86_64.whl (2.7MB)

```

@ntoll - any sugestions?

ntoll, 2020-01-29:
The current version is definitely 2.11.4. See:

https://pypi.org/project/QScintilla/

No idea why you might be getting this message. Linux Mint will use exactly the same wheels as I do, but I see no errors. :-/

abixadamj, 2020-01-29:
Seems to be like https://github.com/pypa/warehouse/issues/3962 - I will wait until tomorrow.

abixadamj, 2020-01-29:
@ntoll 
It seems that I had old version of pip ;-)

[](root@mint-desktop:~# pip3 install -U pip
Collecting pip
  Downloading https://files.pythonhosted.org/packages/54/0c/d01aa759fdc501a58f431eb594a17495f15b88da142ce14b5845662c13f3/pip-20.0.2-py2.py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 950kB/s 
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed pip-20.0.2
root@mint-desktop:~# pip3 install QScintilla --upgrade
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Collecting QScintilla
  Downloading QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 1.2 MB/s 
Collecting PyQt5-sip<13,>=12.7
  Downloading PyQt5_sip-12.7.0-cp36-cp36m-manylinux1_x86_64.whl (260 kB)
     |████████████████████████████████| 260 kB 20.0 MB/s 
Requirement already satisfied, skipping upgrade: PyQt5>=5.10.1 in /usr/local/lib/python3.6/dist-packages (from QScintilla) (5.11.3)
ERROR: pyqt5 5.11.3 has requirement PyQt5_sip<4.20,>=4.19.11, but you'll have pyqt5-sip 12.7.0 which is incompatible.
ERROR: mu-editor 1.0.2 has requirement matplotlib==2.2.2, but you'll have matplotlib 3.1.2 which is incompatible.
ERROR: mu-editor 1.0.2 has requirement pycodestyle==2.4.0, but you'll have pycodestyle 2.5.0 which is incompatible.
ERROR: mu-editor 1.0.2 has requirement pyflakes==2.0.0, but you'll have pyflakes 2.1.1 which is incompatible.
ERROR: mu-editor 1.0.2 has requirement qscintilla==2.10.8, but you'll have qscintilla 2.11.4 which is incompatible.
Installing collected packages: PyQt5-sip, QScintilla
  Attempting uninstall: PyQt5-sip
    Found existing installation: PyQt5-sip 4.19.19
    Uninstalling PyQt5-sip-4.19.19:
      Successfully uninstalled PyQt5-sip-4.19.19
  Attempting uninstall: QScintilla
    Found existing installation: QScintilla 2.10.8
    Uninstalling QScintilla-2.10.8:
      Successfully uninstalled QScintilla-2.10.8
Successfully installed PyQt5-sip-12.7.0 QScintilla-2.11.4
)

abixadamj, 2020-01-29:
Then I can update mu-editor ;-)
[image]

closed 2020-01-29 by abixadamj

dhalbert, 2020-01-29:
I had what at first looked like a different problem, but it had the same underlying cause. I was getting this error:
```
$ pip3 install mu-editor
Collecting mu-editor
  Using cached https://files.pythonhosted.org/packages/fa/2f/6f7eb7e836a528d6959e5c79491d90faef2df7a78e9d953b84451551a98e/mu_editor-1.0.3-py3-none-any.whl
Collecting pyqt5==5.14.1 (from mu-editor)
  Using cached https://files.pythonhosted.org/packages/3a/fb/eb51731f2dc7c22d8e1a63ba88fb702727b324c6352183a32f27f73b8116/PyQt5-5.14.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.6/tokenize.py", line 452, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-build-xbjmta56/pyqt5/setup.py'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-xbjmta56/pyqt5/
```
I removed `~/.cache/pip/`, and then I had the problem above of not finding the latest version of `qscintilla`. I upgraded pip and that problem went away.