mu-editor/mu #994: issue on windows install

machinehistories, 2020-01-30:
I ran into a few issues on the install of the latest release. I uninstalled the previous version and then I began the new installation. Perhaps there is a specific upgrade path that I didn't follow correctly. Many thanks for such a great tool I look forward to getting the latest release back up and running.

[image]

[image]

[image]

tmontes, 2020-01-30:
Interesting that the installer seems to be failing at writing the files in a directory to which you should have write permissions. I wonder if the files are somehow locked.

I know that Windows (at times?) prevents writing to files that are "in use". Maybe Mu is still running?

Suggestion:
* Ensure Mu is not running.
* Go to the Windows software uninstall panel and confirm Mu is not there. If it is, go ahead an uninstall it (maybe, once more). If not, disregard.
* Try installing again.

My 2c. :)

ntoll, 2020-01-30:
This looks like file permissions on your machine -- i.e. Windows is not allowing Mu's installer to write to the location it references. :-/

I've uninstalled / reinstalled many times on my Windows machine and never seen this problem.

Perhaps you had the old version of Mu running (in which case it was using the referenced files and so you'd be stopped from overwriting them)..?

In any case, I'd reboot the machine (so you can be certain Mu isn't running, and delete the folder `c:\Users\jason\AppData\Local\Mu`. Then the installer should work.

Best of luck and please let us know how you get on..! :+1:

ntoll, 2020-01-30:
Hah..! @tmontes that's exactly it.

I'm closing the issue because it appears to be a local configuration issue rather than anything to do with Mu per-se.

Please feel free to re-open if you don't think so, or we've missed something.

closed 2020-01-30 by ntoll