mu-editor/mu #996: Feature Request: Expand "Files" capabilities.

the-stanely, 2020-01-31:
One of the most outstanding features of the Mu editor is that it performs some of the features of ampy through its 'Files' pane.  It would be useful if this capability was expanded to include:

* navigate paths (up and down) so libraries installed with upip into /lib could be seen
* show file size info
* maybe copy files from target to development folder (I'm a little concerned about this one because I use the minifier, and others who do and aren't paying attention might overwrite their code with the minified version.)

dybber, 2020-05-25:
Additionally it would be nice to be able to create folders on the device (requested in #1022)

KKawase0104, 2020-06-02:
Hello.
Maybe, I had implemented something close.
If it's the same as your image, I'm going to implement it in ESP32 mode.

[image]

dybber, 2020-06-02:
@KKawase0104 looks good. Is that just a sketch, or something you already implemented?

Another point regarding the file manager, is that currently is a bit difficult to understand how to use it. It's only drag and drop based right now, and you have to "guess" or be told that you can drag files from one side to the other. Would be nice if it was somehow made more "intuitive", or just support multiple ways of copying files. Here are some alternatives to drag and drop:

 - There could be buttons between the two file panes ( <-- and -->) that you could push to transfer a selected file
 - Users could CTRL/CMD+C CTRL/CMD+V to copy and paste files
 - User could right click a file, and choose a menu item "Copy to device"

I don't know which of these are the right to add, I have just experienced students who get stuck because they can't figure out how to use the file manager.

ZanderBrown, 2020-06-02:
I think a context item makes sense

dybber, 2020-06-10:
Just thought more about this, and I think that it would also make it more intuitive for users if a  icons are used to distinguish files from folders, that would make it seem more like a traditional file manager. I think the best way to go is to mimick a traditional file manager as much as possible.

KKawase0104, 2020-06-28:
Sorry for the delay. I made for the ESP32.
7bf330a

dybber, 2020-09-19:
Hi @KKawase0104, could you perhaps send this in as a pull-request, then we can take the discussion there?

KKawase0104, 2020-09-27:
@dybber 
Thank you for your comment.
I will rebase and pull-request.