mu-editor/mu #997: Mac OSX 10.15 crash when open the app

swortain, 2020-02-03:
I install the app download from the website, but the app crash when I open the app.
Sorry, I don't know what I can provide... It just crashed...

ntoll, 2020-02-03:
What version of Mu are you using? What version of OSX are you on?

ntoll, 2020-02-03:
Ahh, sorry, I see you're on Catalina.

swortain, 2020-02-04:
yes, It's 10.15.2 of OSX
I tried 1.0.2\1.0.3\1.1.0, but all of them will crash on my mac

swortain, 2020-02-04:
> Ahh, sorry, I see you're on Catalina.

thx for the reply.

rOGI420, 2020-02-05:
I have the exact same issue.

Latest version of Mu, Mac OS X 10.11 (El Capitan).

When I run Mu from terminal I get the following:

`(base) Igors-MacBook-Pro:~ igor.cornejo$ mu-editor
Traceback (most recent call last):
  File "/Users/igor.cornejo/opt/anaconda3/bin/mu-editor", line 5, in <module>
    from mu.app import run
  File "/Users/igor.cornejo/opt/anaconda3/lib/python3.7/site-packages/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: dlopen(/Users/igor.cornejo/opt/anaconda3/lib/python3.7/site-packages/PyQt5/QtCore.abi3.so, 2): Symbol not found: __os_activity_create
  Referenced from: /Users/igor.cornejo/opt/anaconda3/lib/python3.7/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Users/igor.cornejo/opt/anaconda3/lib/python3.7/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore`

I'm also using Anconda, would that have anything to do with it?

Thanks!

tmontes, 2020-02-06:
@rOGI420,

Your failure is clearly different. Note how at some point the output you quoted shows:

```
Symbol not found: __os_activity_create
Referenced from: /Users/igor.cornejo/opt/anaconda3/lib/python3.7/site-packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore
(which was built for Mac OS X 10.13)
```

...this means that you should be running macOS 10.13, at least.

tmontes, 2020-02-07:
Linking in @rOGI420 created issue: #1002.

carlosperate, 2020-03-22:
@swortain thanks for the report. If you could provide additional information by following these steps @tmontes put together it would help us diagnose the problem.

From https://github.com/mu-editor/mu/issues/978#issuecomment-569263240:
> We're sorry to hear that Mu is no longer working for you, after your OS upgrade.
> 
> Here are a few things you could try, in order of growing complexity, to help us diagnose your case and, hopefully, make Mu work again soon:
> 
>     1. Try reinstalling Mu 1.0.2:
>        
>        * Use the Finder to locate Mu and delete it.
>        * Download the 1.0.2 version from https://codewith.mu/en/download
>        * Install it again.
> 
>     2. Try reinstalling Mu 1.1.0.alpha2:
>        
>        * As above but download version 1.1.0.alpha2 instead.
> 
>     3. Share any Mu logs that may be present in your system:
>        
>        * In the Finder menu click "Go" > "Go To Folder...".
>        * Type in "~/Library/Logs/mu".
>        * Please share the most recent file you find there.
> 
>     4. Launch Mu from a Terminal:
>        
>        * Launch the Terminal application.
>        * Type in `/Applications/mu-editor.app/Contents/MacOS/mu-editor` and hit RETURN.
>        * If Mu does not work (I don't expect it will), probably some output text will be displayed beneath that. Please share it back here.
> 
> 
> Again, thanks for using Mu and for helping us figuring out what may need fixing. :-)

@mimiflynn @GrNeOp if this is something you could do as well that would help us too, the more data we have the better.

twn39, 2020-10-10:
I have the same issue, error message:

```
2020-10-10 14:41:50,075 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
```
**Mac system: macOS Majove  10.14.6**
** Mu version: 1.0.3**

But I launch Mu from terminal, it works.

carlosperate, 2020-10-12:
Thanks for the report @twn39.
Could you try this alpha version (a build from the latest commit in the master branch) and see if the problem still persist there?
https://s3-eu-west-2.amazonaws.com/mu-builds/osx/mu-editor_2020-10-08_22_06_master_e6adf68.zip

twn39, 2020-10-14:
@carlosperate  It works, thanks.

carlosperate, 2021-03-04:
@swortain are you still having this issue?
I believe all the other issues covered in this ticket have been resolved, but I am not sure we were able to resolve yours.

carlosperate, 2021-03-16:
If you are still having this issue can you try the latest release? https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2

As all the other problems discussed here were resolved I'll close this as resolved as well, but please do reopen it if you are still having issues with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate