mu-editor/mu #998: Mu not able to flash micro:bit on macOS Catalina

brahmafear, 2020-02-03:
Attempting to flash program to micro:bit while running on macOS Catalina (10.15) fails with error "There was a problem flashing the micro:bit."

This happens on v1.0.3 and on 1.1.0alpha2.
This happens on multiple machines - all running Catalina.
This happens even when app is started via sudo ("sudo open /Applications/mu-editor.app").

I understand that this is due to Catalina's new security model but hoping for some sort of workaround.

brahmafear, 2020-02-03:
So far, have only found this:
Click Flash with micro:bot not connected.  
Select folder to save.
Connect micro:bit.
Drag micropython.hex from step 2 to micro:bit.

technodono, 2020-02-13:
Version: Mac OSX Catalina with Mu 1.1.0 alpha 2 (might work on the non alpha version too)

I found a fix by making this script which launches the applications script in the apps files itself:
`mu-editor.app/contents/mac_osx/mu-editor`
Or are least the directory looks something like that... the script can be downloaded here:
[Mu-Editor.command.zip](https://github.com/mu-editor/mu/files/4197684/Mu-Editor.command.zip)
But for it to work you need to do 5 simple steps
1. Make sure you have a version of mu installed and make sure it's named Mu-Editor or it may not work
2. Drag the file into your applications folder
3. open terminal and run `cd /applications`
4. run `chmod a+x Mu-Editor.command` - if you didn't do this properly the script will not open due to insufficient file permissions
5. Now you can open the file and run it from anywhere on your laptop not just your applications folder and this should work and flash to your microbit\

If this script doesn't work make sure the mu app you installed is named to mu-editor.app and when doing the terminal commands make sure the script is named Mu-Editor

ntoll, 2020-02-13:
@technodono hurrah..! I'll reference this in our docs.

brahmafear, 2020-02-13:
I can see the intention -- launching the app via Terminal causes macOS to prompt, "Terminal has requested access to file son a removable volume" which should work around the restriction.  

But this workaround is not working for my tests.  I still receive an error about unable to flash.  I will test on more machines to verify.

technodono, 2020-02-13:
Hmm... I don't know the specifics but after doing a whole bunch of things like changing the permissions of the microbit disk and stuff that's all I could find that worked on my machine... Maybe it won't work for everyone... I just dunno yet. Another option is to give full disk access to terminal and see if that flashes to the microbit

brahmafear, 2020-02-13:
I tested with full disk permission for Terminal - still didn't work for
me.  I will continue to test and report back.

technodono, 2020-02-13:
Alright... well that's all the knowledge I know as far as this app goes. Hopefully a better solution is found

technodono, 2020-02-14:
Hang on, are you launching the app or the executable because my program launches the mu-editor executable inside the app: 
[image]
You can get to it by right clicking on the app and > show contents
[image]
then go into contents > MacOS
and the open the mu-editor file because what it seems from your reply is that you are just launching the app from terminal and not the executable in the app.

If you did do that then I have no idea how you'd fix it... however I have a script which uses sudo to run app as admin which I can attach if you want to try that out

brahmafear, 2020-02-14:
I've attempted all the ways I can think of - with and without sudo.  

The "app" mu-editor inside mu-editor.app/Contents/MacOS is really just a shell script which launches its embedded version of python3 and sets up necessary modules.  I've even tried launching using macOS's own python3 (installed via xcode).  

I'm going to try to setup a new machine with a fresh installation of Catalina tomorrow to re-test this.

technodono, 2020-02-14:
Good luck! in the meantime I'll classify this as a half-solution because it hasn't worked for you yet... also why did  ("sudo open /Applications/mu-editor.app"). not work for me in terminal but ("sudo open /Applications/mu-editor.app/contents/MacOS/mu-editor") did work for me... if you don't have a clue that's fine

carlosperate, 2020-02-14:
I'm not sure I'll be able to be of much help in this case, as I've purposely not upgraded to Catalina because of a couple of things (like this one). But since there are new access restrictions in this version of macOS, I'm hoping you can at least grant Mu access to the removable volume using the Security and Privacy settings:

https://nektony.com/duplicate-finder-free/folders-permission

Could anybody try this?

brahmafear, 2020-02-14:
@technodono You need to add "-a" to your open command.  "sudo open -a /Applications/mu-editor.app"

@carlosperate I had previously found same site about Catalina permissions.  Users can't manually add a new application to that list - the app has to request it.

technodono, 2020-02-14:
Thanks for the correction @brahmafear and @carlosperate yes you can do that but it still doesn’t work with the app for me

brahmafear, 2020-02-15:
Got the .command file to work on a new Catalina installation.  Flashing and REPL work fine.

technodono, 2020-02-15:
Woohoo! Problem solved. What was the issue? Or do you not know

mooselips, 2020-10-06:
This fixed worked for me!  I have been struggling with this for almost a year and have not been able to use the Micro:Bits with my students.  Now I just have to:

1. Figure out how to deploy the fix via JAMF.
2. Test it on another machine.  The mu-editor app now opens from the Applications folder, but I so muddled things up on this computer because I have tried every fix in the past that I need to test just doing this script on a device.  I also want to test just chmod a+x /Applications/mu-editor/Contents/MacOS/mu-editor to see if that works (location is from memory...don't copy and paste).

Mike

> Version: Mac OSX Catalina with Mu 1.1.0 alpha 2 (might work on the non alpha version too)
> 
> I found a fix by making this script which launches the applications script in the apps files itself:
> `mu-editor.app/contents/mac_osx/mu-editor`
> Or are least the directory looks something like that... the script can be downloaded here:
> [Mu-Editor.command.zip](https://github.com/mu-editor/mu/files/4197684/Mu-Editor.command.zip)
> But for it to work you need to do 5 simple steps
> 
> 1. Make sure you have a version of mu installed and make sure it's named Mu-Editor or it may not work
> 2. Drag the file into your applications folder
> 3. open terminal and run `cd /applications`
> 4. run `chmod a+x Mu-Editor.command` - if you didn't do this properly the script will not open due to insufficient file permissions
> 5. Now you can open the file and run it from anywhere on your laptop not just your applications folder and this should work and flash to your microbit\
> 
> If this script doesn't work make sure the mu app you installed is named to mu-editor.app and when doing the terminal commands make sure the script is named Mu-Editor
> 
> It seems a bit messy but it's a solution that works (for me at least.) You may still have to grant disk permissions in privacy settings though

technodono, 2020-10-07:
> This fixed worked for me! I have been struggling with this for almost a year and have not been able to use the Micro:Bits with my students. Now I just have to:
>
>     1. Figure out how to deploy the [...]

Unfortunately it's the solution that will have to do as I have no idea how to fix it otherwise... glad it worked for you!

indigo51, 2020-11-19:
Tried the above but still no joy flashing microbits.  Is there a clear series of steps someone has I can follow?  Workshops coming up and I am sure some will have Catalina.

technodono, 2020-11-19:
if you really want to avoid the issue, you should save the file as a .hex and manually drag&drop the exported file into a microbit @indigo51

technodono, 2020-11-19:
currently I don't think this problem is yet fixed :/

carlosperate, 2020-11-19:
@tmontes is this something that would be resolved with the new pup app signing system?

andreagrandi, 2020-12-13:
I've tried the `sudo open -a /Applications/mu-editor.app` but unfortunately it doesn't work for me, I still get the same error

carlosperate, 2021-03-16:
This should be fixed in the latest release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 🎉 

I'll close this as resolved, but please do reopen it if you are still having this issue with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate

sunruoshi, 2021-03-22:
My version is Mu 1.1.0 beta.2 still have this problem. Cannot flash. OS version 11.2.3(20D91) Big Sur.

carlosperate, 2021-03-22:
Thanks for testing @sunruoshi, could you open Mu, try to flash the micro:bit and then add a copy of your log to this issue?
https://github.com/mu-editor/mu/issues/1383

You can get the logs by clicking on the cog icon in the bottom right of the editor window.

sunruoshi, 2021-03-23:
> Thanks for testing @sunruoshi, could you open Mu, try to flash the micro:bit and then add a copy of your log to this issue?
> #1383
> 
> You can get the logs by clicking on the cog icon in the bottom right of the editor window.

2021-03-23 09:27:52,476 - root:259(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.2
2021-03-23 09:27:52,476 - root:260(run) INFO: uname_result(system='Darwin', node='Sams-MacBook-Pro.local', release='20.3.0', version='Darwin Kernel Version 20.3.0: Thu Jan 21 00:07:06 PST 2021; root:xnu-7195.81.3~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2021-03-23 09:27:52,476 - root:261(run) INFO: Python path: ['/', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python38.zip', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/lib-dynload', '/Users/sam/.local/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/setuptools-49.6.0-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/pip-20.2.2-py3.8.egg', '/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/IPython/extensions']
2021-03-23 09:27:52,476 - root:262(run) INFO: Language code: en_US
2021-03-23 09:27:52,477 - mu.settings:220(load) WARNING: No settings file found at /Users/sam/Library/Application Support/mu/settings.json; skipping
2021-03-23 09:27:52,728 - mu.virtual_environment:434(ensure_and_create) INFO: Added log handler.
2021-03-23 09:27:52,728 - mu.virtual_environment:438(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-03-23 09:27:52,874 - mu.virtual_environment:489(ensure_path) INFO: Virtual Environment found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202
2021-03-23 09:27:52,880 - mu.virtual_environment:502(ensure_interpreter) INFO: Interpreter found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/python
2021-03-23 09:27:52,920 - mu.virtual_environment:534(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-03-23 09:27:52,925 - mu.virtual_environment:566(ensure_pip) INFO: Pip found at: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/pip
2021-03-23 09:27:52,925 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pygame
2021-03-23 09:27:53,042 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: numpy
2021-03-23 09:27:53,272 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-03-23 09:27:53,309 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: flask
2021-03-23 09:27:53,584 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: serial
2021-03-23 09:27:53,624 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-03-23 09:27:53,663 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: nudatus
2021-03-23 09:27:53,720 - mu.virtual_environment:548(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-03-23 09:27:53,791 - mu.virtual_environment:451(ensure_and_create) INFO: Virtual environment already exists.
2021-03-23 09:27:54,911 - mu.logic:758(__init__) INFO: Setting up editor.
2021-03-23 09:27:54,911 - mu.logic:778(__init__) INFO: Log directory: /Users/sam/Library/Logs/mu
2021-03-23 09:27:54,911 - mu.logic:779(__init__) INFO: Data directory: /Users/sam/Library/Application Support/mu
2021-03-23 09:27:54,912 - mu.logic:793(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, pyboard, debugger, pygamezero, lego, pico
2021-03-23 09:27:54,912 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-23 09:27:54,956 - mu.modes.base:61(get_default_workspace) INFO: Using workspace /Users/sam/mu_code from settings file
2021-03-23 09:27:54,956 - mu.logic:1523(change_mode) INFO: Workspace directory: /Users/sam/mu_code
2021-03-23 09:27:54,956 - mu.logic:867(restore_session) DEBUG: <SessionSettings from /Users/sam/Library/Application Support/mu/session.json>
2021-03-23 09:27:54,961 - mu.logic:978(_load) INFO: Loading script from: /Users/sam/mu_code/compass.py
2021-03-23 09:27:54,962 - mu.logic:316(read_and_decode) DEBUG: Trying to decode with utf-8
2021-03-23 09:27:54,962 - mu.logic:319(read_and_decode) INFO: Decoded with utf-8
2021-03-23 09:27:54,962 - mu.logic:332(read_and_decode) DEBUG: Detected newline '\n'
2021-03-23 09:27:54,962 - mu.logic:1077(_load) DEBUG: # Write your code here :-)
from microbit import *

compass.calibrate()

while True:
    bearing = compass.heading()
    if bearing < 45 or bearing > 315:
        display.show("N")
    elif bearing > 45 and bearing < 135:
        display.show("E")
    elif bearing > 135 and bearing < 225:
        display.show("S")
    elif bearing > 225 and bearing < 315:
        display.show("W")
    else:
        display.show(" ")

2021-03-23 09:27:55,069 - mu.logic:889(restore_session) INFO: Loaded files.
2021-03-23 09:27:55,070 - mu.logic:892(restore_session) INFO: User defined environment variables: []
2021-03-23 09:27:55,070 - mu.logic:897(restore_session) INFO: Minify scripts on micro:bit? False
[... cut ...]
2021-03-23 09:28:29,697 - mu.logic:1396(show_admin) INFO: Showing admin with logs from /Users/sam/Library/Logs/mu/mu.log
2021-03-23 09:28:29,697 - mu.virtual_environment:697(installed_packages) INFO: Discovering installed third party modules in venv.
2021-03-23 09:28:29,697 - mu.virtual_environment:198(run) DEBUG: About to run blocking: /Users/sam/Library/Application Support/mu/mu_venv-38-20210322-154202/bin/pip, ['list', '--disable-pip-version-check'], 30.0
2021-03-23 09:28:30,115 - mu.virtual_environment:717(installed_packages) INFO: []

carlosperate, 2021-03-23:
Thank you @sunruoshi. Can you post your log in this ticket GitHub issue instead? https://github.com/mu-editor/mu/issues/1383

This one is close and will be easier to track this problem over there.

sunruoshi, 2021-03-23:
> Thank you @sunruoshi. Can you post your log in this ticket GitHub issue instead? #1383
> 
> This one is closed and will be easier to track this problem over there.

Sure. Posted.