mu-editor/mu #1000: runs properly with "make run" but segmentation fault with running from MacOS package

DaveSprague, 2020-02-04:
I've made a version of mu-editor adding the changes from [PR #485](https://github.com/mu-editor/mu/pull/485) and it works fine on MacOS Catalina when I run it via "make run" but after building the MacOS package I get a segmentation fault that seems to be related to Qt?

I'd appreciate any help you can give in tracking down what's causing it. 

Here's the dump from the seg fault:

Process:               python3 [15878]
Path:                  /Applications/mu-editor.app/Contents/MacOS/mu-editor
Identifier:            mu.codewith.editor.mu-editor
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           python3 [15878]
User ID:               501

Date/Time:             2020-02-04 07:53:56.688 -0500
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        D4FF6FB5-C72C-9E86-12F0-B177CB0312A3

Sleep/Wake UUID:       F0AD5FA4-4C49-4C34-AB7E-5F4A44E0CA10

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       41000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [15878]

VM Regions Near 0x20:
--> 
    __TEXT                 0000000105b70000-0000000105d2d000 [ 1780K] r-x/rwx SM=COW  /Applications/mu-editor.app/Contents/Resources/python/bin/python3

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   python3                       	0x0000000105c1472a unicode_fromformat_write_str + 26
1   python3                       	0x0000000105beb5da PyUnicode_FromFormatV + 2058
2   python3                       	0x0000000105bec264 PyUnicode_FromFormat + 148
3   sip.so                        	0x000000010729612b sip_api_init_module + 646
4   QtCore.so                     	0x00000001069ab1dd PyInit_QtCore + 333
5   python3                       	0x0000000105c6b731 _PyImport_LoadDynamicModuleWithSpec + 657
6   python3                       	0x0000000105c6b1f1 _imp_create_dynamic + 257
7   python3                       	0x0000000105bc81c2 PyCFunction_Call + 258
8   python3                       	0x0000000105c48d04 _PyEval_EvalFrameDefault + 25844
9   python3                       	0x0000000105c4ce7c _PyEval_EvalCodeWithName + 3436
10  python3                       	0x0000000105c4d801 fast_function + 657
11  python3                       	0x0000000105c4c011 call_function + 401
12  python3                       	0x0000000105c48988 _PyEval_EvalFrameDefault + 24952
13  python3                       	0x0000000105c4d758 fast_function + 488
14  python3                       	0x0000000105c4c011 call_function + 401
15  python3                       	0x0000000105c48988 _PyEval_EvalFrameDefault + 24952
16  python3                       	0x0000000105c4d758 fast_function + 488
17  python3                       	0x0000000105c4c011 call_function + 401
18  python3                       	0x0000000105c48988 _PyEval_EvalFrameDefault + 24952
19  python3                       	0x0000000105c4d758 fast_function + 488
20  python3                       	0x0000000105c4c011 call_function + 401
21  python3                       	0x0000000105c48988 _PyEval_EvalFrameDefault + 24952
22  python3                       	0x0000000105c4d758 fast_function + 488
23  python3                       	0x0000000105c4c011 call_function + 401
24  python3                       	0x0000000105c48988 _PyEval_EvalFrameDefault + 24952
25  python3                       	0x0000000105c4d9fd _PyFunction_FastCallDict + 477
26  python3                       	0x0000000105b7c453 _PyObject_FastCallDict + 307
27  python3                       	0x0000000105b7d456 _PyObject_CallMethodIdObjArgs + 486
[... 362 lines cut ...]
 
                                VIRTUAL   REGION 
REGION TYPE                        SIZE    COUNT (non-coalesced) 
===========                     =======  ======= 
Activity Tracing                   256K        1 
Dispatch continuations            8192K        1 
Kernel Alloc Once                    8K        1 
MALLOC                            46.7M       58 
MALLOC guard page                   16K        4 
MALLOC_LARGE (reserved)            512K        2         reserved VM address space (unallocated)
STACK GUARD                         12K        3 
Stack                             17.0M        3 
__DATA                            22.8M      265 
__DATA_CONST                        20K        1 
__FONT_DATA                          4K        1 
__LINKEDIT                       356.2M       27 
__OBJC_RO                         32.0M        1 
__OBJC_RW                         1780K        2 
__TEXT                           161.7M      262 
__UNICODE                          564K        1 
mapped file                       27.7M        2 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                            675.3M      640 
TOTAL, minus reserved VM space   674.8M      640

DaveSprague, 2020-02-04:
A bit more info.  I also get this segmentation fault when I take the current master of mu-editor and do a "make MacOS".

tmontes, 2020-02-04:
@DaveSprague,

I tried reproducing your failure but couldn't. Here's what I just verified both under macOS Mojave 10.14.6 and on a macOS Catalina 10.15.2 VM:

* Checkout current master.
* Create and activate new virtual environment running Python 3.6 from www.python.org.
* `pip install -e ".[dev]"`
* **hack**: uninstalled `docutils` as there's currently a versioning conflict between indirect Mu docs dependencies and Mu macOS packaging dependencies: `pip uninstall docutils`.
* `make run` works -- but there's a better, simpler way, I'd argue: just type `mu-editor`. Both work.
* Package macOS application bundle with `make macos`.
* The Application Bundle under `./macOS` works on both systems.

Again, I recall this is from current master, with several outdated dependencies, including the older Qt 5.12.1. But it does not crash.

Can you please try to retrace your steps based on mine, above, and see where you may be doing something different (or what might be different in your environment)?

DaveSprague, 2020-02-04:
Thanks tmontes, I was running python v. 3.8.1 so that was the problem.  I followed your steps and I got it to work.

The only other change I had to make was to remove the flake8 from the definition of "check:" in the Makefile.  When flake8 runs get its many many errors/warnings and stops the script from completing and generating the MacOS package.  Once I disabled flake8 from running it generated a MacOS app that runs just fine.  Thanks for you help
Dave

DaveSprague, 2020-02-04:
Now that I have the version of the mu-editor, with #485 integrated, running in both Mac app form and from using mu-editor or "make run" in the development folder, I'm now seeing another problem.  I can open a separate issue for this if that makes more sense.

When I run the mu-editor in the development director with 'mu-editor' or 'make run' it works fine but when I run it in it's the Mac App (created with "make macos", it crashes when it the Run button is clicked and it attempts to copy the code file from the local filesystem to the CIRCUITPY Volume.

This is the error message in the log file after a crash:
2020-02-04 14:12:21,663 - root:105(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/modes/circuitpython.py", line 265, in run
    copyfile(pathname, dst)
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/shutil.py", line 121, in copyfile
PermissionError: [Errno 1] Operation not permitted: '/Volumes/CIRCUITPY/code.py'
2020-02-04 14:12:29,048 - root:122(run) INFO: 

Any recommendations on how to fix this?  I confirmed that I can manually write files to the CIRCUITPY Volume just fine. 

Dave

DaveSprague, 2020-02-05:
I was able to fix the segmentation fault that happened when the circuit python mode would copy the file to the CIRCUITPY Volume with the name code.py

This was caused by Catalina required explicit permission for apps to access certain folders.  I had to go under Settings | Security & Privacy | Privacy | Full Disk Access and add the mu-editor app to the whitelist.

tmontes, 2020-02-06:
@DaveSprague, thanks for sharing your progress and, in particular, the manual way of telling macOS Catalina that Mu should be able to write to the CircuitPython "external volume".

I wonder if such manual adjustment will still be needed once we address #940, focusing on the updated requirements for macOS application packaging on Catalina. :)

DaveSprague, 2020-02-06:
Hi Tiago, I'll do some testing on the current code base (without the
modifications for CircuitPython Run button) to verify that the manual
security change is necessary and sufficient to get it working.

I do still have a question about the flake8 test.  When I run it, it gets
many many errors or warnings.  I believe most of them are related to line
length.  I might try to add a flag/option to the flake8 call to remove
those errors/messages to see if there's something else.  I appears that the
flake8 check is preventing "make run" from completing and also keeps "make
macos" from completing.  I had to remove flake8 from the "check" definition
in Makefile in order to generate the macos package.

Dave

tmontes, 2020-02-06:
> Hi Tiago, I'll do some testing on the current code base (without the modifications for CircuitPython Run button) to verify that the manual security change is necessary and sufficient to get it working.

Thanks in advance for that. It will be useful information that we can share with other macOS Catalina users.

> I do still have a question about the flake8 test. When I run it, it gets many many errors or warnings. I believe most of them are related to line length. I might try to add a flag/option to the [...]

I find that interesting. Here's my experience, both with the current master, at commit e3b7191e223a02fbff296fa0ecc834382e16e2b4 and with the proposed branch in #485 at commit 5c6056df6c59b15bf006169b458f975badf1dc0a:

```
$ git checkout master
$ git log --oneline | head -5
e3b7191 Merge pull request #992 from tomviner/patch-1
8eb9c5e Update release.rst
68e8055 Release 1.0.3 changelog
ee600ff Merge pull request #951 from knowledgejunkie/use-https-urls-in-appstream-data
841ed33 Where available, provide HTTPS URLs in AppStream metadata
$ git status
On branch master
Your branch is up to date with 'mine/master'.

nothing to commit, working tree clean
$ make flake8
flake8
$
```

No messages at all, from the `flake8` check in the current `master`. Let's now check the proposed branch in #485:

```
$ git checkout fmorton-master 
Switched to branch 'fmorton-master'
Your branch is up to date with 'fmorton/master'.
$ git log --oneline | head -5
5c6056d modify comment
a7e23a4 improve test coverage
4523de0 improve test coverage
1458d12 improve circuitpython test coverage
ad18689 catch up from june 2018--needs more test coverage
$ git status
On branch fmorton-master
Your branch is up to date with 'fmorton/master'.

nothing to commit, working tree clean
$ make flake8
flake8
$
```

Again, no output from flake8, so all looking good here.

Of note:
* The commands above were run on a bash shell on macOS 10.14.6.
* A Python 3.6 virtual environment was activated, with Mu installed in development mode, including  the `dev` extra: `pip install -e .[dev]`.
* Commands were run with the root of the repository as the current working directory.
* For completeness, refer to the existing `setup.cfg` file that already tunes the `flake8` check behaviours.

Something must be different on your system to justify your observations. Feel free to share either more details about it or any findings you believe are relevant.

DaveSprague, 2020-02-06:
#485

DaveSprague, 2020-02-07:
Here's the steps to generate a mac app that has the Circuit Python Run Button ([PR#485](https://github.com/mu-editor/mu/pull/485)) that saves your program file locally on your computer but then copies it to the CIRCUITPY volume with a name of "code.py" so it will run on the Circuit Python board.

It assumes you're running some version of python 3.6 (I believe any later version will work as well).

git clone https://github.com/mu-editor/mu.git
cd mu
python3 -m venv .venv
_{note: if you name your virtual env something other than ".venv" then edit setup.cfg and add that folder to the ignore list for flake8}_
source .venv/bin/activate
pip install --upgrade pip
pip install -e ".[dev]"
pip uninstall botocore
pip install botocore
_the above two lines fix the incompatibilty between botocore and docutils_
git pull origin pull/485/head
make macos

Now move the mu-editor.app file from the macos folder, under the mu-editor folder, to the Mac applications folder (or you can just run it from the macos folder if you want).

If you're running Catalina then you might have to give the app permission to access the CIRCUITPY volume.  (You'll know you need to do this if the app crashes when you try to "Run" the program, the mu log file will say there was a Permission error when copying the file to the CIRCUITPY volume.) 

To do that go under Settings | Security and Privacy, Select the Privacy tab at the top, in the left column select "Full Disk Access" and then on the right side click the box next to mu-editor to give it permission.  If those options are grayed out, you need to select the padlock in the lower left of the dialog box and enter you system password before you can chance the settings.

It assumes you're running some version of python 3.6

git clone https://github.com/mu-editor/mu.git
cd mu
python3 -m venv .venv

//{note: if you name your virtual env something other than ".venv" then edit setup.cfg and add that folder to the ignore list for flake8}

source .venv/bin/activate
pip install --upgrade pip
pip install -e ".[dev]"
pip uninstall botocore
pip install botocore
git pull origin pull/485/head
make macos

Now move the mu-editor.app file from the macos folder to the Mac applications folder (or you can just run it from the macos folder if you want).

If you're running Catalina then you might have to give the app permission to access the CIRCUITPY volume.  To do that go under Settings | Security and Privacy, Select the Privacy tab at the top, in the left column select "Full Disk Access" and then on the right side click the box next to mu-editor to give it permission.  If those options are grayed out, you need to select the padlock in the lower left of the dialog box and enter you system password before you can chance the settings.

closed 2020-02-07 by DaveSprague

tmontes, 2020-02-07:
@DaveSprague, thanks for sharing your progress and success. I suppose that your "macOS Catalina" instructions might be useful in the current docs... :)

@ntoll, we haven't kept in sync for a while, but I noticed that in the recent [1.0.3 announcement](https://madewith.mu/mu/releases/2020/01/26/mu-1-0-3.html) at https://madewith.mu/ macOS Catalina users are guided to run `sudo open mu-editor.app` to "mitigate" disk access permissions, IIUC. I guess the notes by @DaveSprague above (which I'll quote next) may be a better approach for a few reasons:

* No need to use the "intimidating" terminal thing.
* `sudo` will not work if the user is not an administrative user.
* `sudo` works if the user is an administrative user but can go awfully wrong if, for some reason, beginners don't type the same exact command as described. ;-)

Here's what @DaveSprague noted, slightly reformatted/edited for easier reading

> If you're running Catalina then you might have to give the app permission to access the CIRCUITPY volume. To do that:
> * Launch System Preferences.
> * Select "Security & Privacy".
> * Select the "Privacy" tab at the top.
> * In the left column select "Full Disk Access".
> * Then, on the right side, click the box next to mu-editor to give it permission.
> If those options are grayed out, you need to select the padlock in the lower left of the dialog box and enter you system password before you can chance the settings.

What do you say about including these notes somewhere (even maybe updating the 1.0.3 announcement post)?

PS: Not sure if these steps will still be needed once we address #940. We'll get to that! :)

ntoll, 2020-02-11:
@tmontes hey hey... many thanks for this @DaveSprague..!

My "real work" is extraordinarily busy at the moment. But I'll try getting these instructions into the website ASAP.

ntoll, 2020-02-11:
OK... you can find the update here:

https://codewith.mu/en/howto/1.0/install_macos#catalina-permissions

Thanks for the heads up..!