mu-editor/mu #999: Installation in ubuntu 18.04 python 3.8

waqarrashid33, 2020-02-04:
I am been trying to install the mu-editor on ubuntu 18.04 but so far haven't been successful. I am getting the following error and I think I need to install some dependencies. It was giving error about SDL but then I searched on internet and somehow installed the .so file 
```
    ERROR: Command errored out with exit status 1:
     command: /usr/local/bin/python3.8 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-amxp4hv0/pygame/setup.py'"'"'; __file__='"'"'/tmp/pip-install-amxp4hv0/pygame/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-amxp4hv0/pygame/pip-egg-info
         cwd: /tmp/pip-install-amxp4hv0/pygame/
    Complete output (18 lines):    
    
    WARNING, No "Setup" File Exists, Running "buildconfig/config.py"
    Using UNIX configuration...
    
    Hunting dependencies...
    SDL     : found 1.2.15
    FONT    : not found
    IMAGE   : not found
    MIXER   : not found
    PNG     : found
    JPEG    : not found
    SCRAP   : found
    PORTMIDI: not found
    PORTTIME: not found
    FREETYPE: found 21.0.15
    Missing dependencies
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```
Below is the output for `python3 setup.py egg_info`. I think the problem is probably straight forward but I am not well versed in python so I would appreciate any help.

```
python3 setup.py egg_info
running egg_info
creating mu_editor.egg-info
writing mu_editor.egg-info/PKG-INFO
writing dependency_links to mu_editor.egg-info/dependency_links.txt
writing entry points to mu_editor.egg-info/entry_points.txt
writing requirements to mu_editor.egg-info/requires.txt
writing top-level names to mu_editor.egg-info/top_level.txt
writing manifest file 'mu_editor.egg-info/SOURCES.txt'
package init file 'mu/debugger/__init__.py' not found (or not a regular file)
reading manifest file 'mu_editor.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'mu_editor.egg-info/SOURCES.txt'
```

Best regards,
Waqar Rashid

ntoll, 2020-02-04:
Hey hey... this is probably because PyGame isn't (yet?) out for Python 3.8.

waqarrashid33, 2020-02-04:
> Hey hey... this is probably because PyGame isn't (yet?) out for Python 3.8.

Ok. Let me try with another version of python3.

ntoll, 2020-02-04:
I develop Mu on 18.04 with Python 3.7. ;-) Works for me. :+1: 

But this is a good heads up. Perhaps check in on the PyGame project and see where they're at with Python 3.8 integration. Perhaps Mu needs to bump up a version in the deps..?

drammock, 2020-02-05:
I confirm the same failure installing into a python 3.8 conda environment, and success when installing into a python 3.7 environment.

easytarget, 2020-05-05:
Fedora32 has Python3.8 as a default, so after upgrading I needed to use python 3.7 to get my mu install working:
```$ sudo dnf install python37
$ source venv37/bin/activate
```
After this I was able to `pip install -e ".[dev]"` as usual and mu is working.
Note: I also had to `sudo dnf install SDL-devel`, but that may be just my (ad-hoc) base system.
Nor have I checked the official package; (at least on F31) it is quite elderly and has severe copy/paste  and other issues.

ZanderBrown, 2020-05-05:
fc32 ships 1.0.2 (as in fc30/fc31) but looks like fc33 will jump to 1.0.3 https://apps.fedoraproject.org/datagrepper/raw?package=mu

carlosperate, 2021-03-24:
Thanks everyone! The latest beta release has updated the dependencies to be compatible with Python 3.5 to 3.8, so it should work now in this cases.

I'll close this issue, but if anybody is still having issue please feel free to reopen it or to open a new one.

closed 2021-03-24 by carlosperate