mu-editor/mu #1001: matplotlib version - Future request

abixadamj, 2020-02-05:
@ntoll 
Every time I install mu-editor from pip, I have to upgrade matplotlib, maybe it is time to leave 2.2.2 as requirement...

```
root@mint-desktop:~# pip3 install matplotlib -U
Collecting matplotlib
  Downloading matplotlib-3.1.3-cp36-cp36m-manylinux1_x86_64.whl (13.1 MB)
     |████████████████████████████████| 13.1 MB 1.3 MB/s 
Requirement already satisfied, skipping upgrade: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib) (2.4.6)
Requirement already satisfied, skipping upgrade: kiwisolver>=1.0.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib) (1.1.0)
Requirement already satisfied, skipping upgrade: cycler>=0.10 in /usr/local/lib/python3.6/dist-packages (from matplotlib) (0.10.0)
Requirement already satisfied, skipping upgrade: python-dateutil>=2.1 in /usr/local/lib/python3.6/dist-packages (from matplotlib) (2.8.1)
Requirement already satisfied, skipping upgrade: numpy>=1.11 in /usr/local/lib/python3.6/dist-packages (from matplotlib) (1.18.0)
Requirement already satisfied, skipping upgrade: setuptools in /usr/local/lib/python3.6/dist-packages (from kiwisolver>=1.0.1->matplotlib) (44.0.0)
Requirement already satisfied, skipping upgrade: six in /usr/local/lib/python3.6/dist-packages (from cycler>=0.10->matplotlib) (1.13.0)
ERROR: mu-editor 1.0.3 has requirement matplotlib==2.2.2, but you'll have matplotlib 3.1.3 which is incompatible.
Installing collected packages: matplotlib
  Attempting uninstall: matplotlib
    Found existing installation: matplotlib 2.2.2
    Uninstalling matplotlib-2.2.2:
      Successfully uninstalled matplotlib-2.2.2
Successfully installed matplotlib-3.1.3
```

abixadamj, 2020-02-05:
Of course after that upgrade mu-editor works ;-) So 3.1.3 with mu 1.0.3 is OK.

tmontes, 2020-02-06:
@abixadamj,

Thanks for highlighting that. Two notes:

* AFAICT, the `matplotlib==2.2.2` requirement in 1.0.3 is not really necessary, being a left-over from some earlier packaging specification. The current alpha versions, along with the `master` branch, no longer define any dependency on `matplotlib`.

* Regarding your particular environment: you seem to be using `pip` to install / upgrade `matplotlib` into the system-wide shared Python site packages, by running it as root and, from what it seems, outside of an isolated virtual environment. If you are installing `mu-editor` in the same way -- that is, running `pip install mu-editor` as root -- you are bound to face such conflicts (probably harmless in this particular case, but that will lead to all sorts of failures in other cases). With that in mind, as per the installation [docs here](https://codewith.mu/en/howto/1.0/install_with_python), you should install Mu onto an isolated Python virtual environment. In that case, even though `matplotlib==2.2.2` is brought in, it won't conflict with any other version of it you may have installed in some other Python environment.

Does this make sense to you? Hope so! :)

abixadamj, 2020-02-06:
@tmontes - I must use system-wide install because I use LTSP system, where I have ~600 users on 1 server, so create 600 venv's it is not an option for me.

tmontes, 2020-02-07:
> @tmontes - I must use system-wide install because I use LTSP system, where I have ~600 users on 1 server, so create 600 venv's it is not an option for me.

...understood and agreed: creating ~600 venvs each with its own Mu installation is not the way to go. But that won't stop you from creating an admin-managed, system-wide Python virtual environment for Mu, will it? AFAICT, that will give you an isolated environment where Mu can be installed and used from, even in a network shared filesystem, which is something I suppose will be at play in your system. Have you thought about that? :)

abixadamj, 2020-02-07:
@tmontes -WoW !!!! system wide venv? I did not know about it! Can you point me some docs? I will learn, it will be usefull for other projects...

tmontes, 2020-02-07:
> @tmontes -WoW !!!! system wide venv? I did not know about it! Can you point me some docs? I will learn, it will be usefull for other projects...

Well, what can I say, a venv is a venv -- simplistically, just a bunch of files under a given path. Who can read or write to those files is an OS-level thing. With that, if, say, `root` creates a world-readable venv, then every user should be able to use it (read from it) even though they won't be able to change it (`pip install` into it). :-)

Picking up on your sample shell session, I guess something like this might work:

```
root@mint-desktop:~# mkdir -p /path/readable/by/all/users/
root@mint-desktop:~# python3.6 -m venv /path/readable/by/all/users/mu-venv
root@mint-desktop:~# /path/readable/by/all/users/bin/python -m pip install --upgrade pip
root@mint-desktop:~# /path/readable/by/all/users/bin/python -m pip install mu-editor
```

Then, any user should be able to launch Mu with:

```
user@host:~$ /path/readable/by/all/users/bin/mu-editor
```

Notes:
* Adding a launcher icon for user desktop environments left as an exercise! ;-)
* There might be some details related to the way LTSP shares files, etc., that imply minor adjustments to the above.

PS: I have written a few [words on the fundamentals of virtual environments here](https://tmont.es/2018/virtual-environment-fundamentals/), maybe they will help you get a better grasp of why there really is no difference between a "user created venv" and an "admin-managed venv". Let me know if you find that useful. :)

dybber, 2020-06-10:
I guess this can be closed?

dybber, 2020-06-12:
That is, without the matplotlib dependency, users will just manually have to install matplotlib through the "Third party packages" tab to be able to do this: 
[image]

I'm all for removing non-critical dependencies, I even think the Jupyter-dependencies are a bit overkill in the first place. The code could be structured to use a simple REPL without jupyter installed and then move to this REPL if users decides to install all those dependencies jupyter brings in - but that's a different story...

closed 2020-07-14 by abixadamj