mu-editor/mu #1002: OS X 10.11 (El Capitan) Not Supported

rOGI420, 2020-02-07:
Hi All;

Sadly I need to report that mu-editor, version 1.0.3, is not supported in El Capitan.

When running the app, the application launches but then closes.

When launching using the command line, I get an error message expecting OS X version 10.13.

The official website says OS X 10.11 is supported, not correct.

Thanks!

tmontes, 2020-02-07:
@rOGI420, thanks for highlighting that. I've just submitted a fix for the documentation. If you like, you can track its progress here: https://github.com/mu-editor/mu-editor.github.io/pull/54.

tmontes, 2020-02-07:
PS: If you're interested in running Mu on you 10.11 El Capitain system, you can try out 1.0.2 or the current 1.1.0alpha2 which should work. :)

carlosperate, 2020-10-26:
PR has been merged and the website is now updated.

closed 2020-10-26 by carlosperate