mu-editor/mu #1003: macOS 10.15.2 with Chinese, Can't run 1.1.0.alpha1, 1.1.0.alpha2, 1.0.3

JohnGuan, 2020-02-07:
* What you were trying to do,
Try to open the mu-editor

* What steps you took to make this happen,
just open

* What you expected to happen,
may macOS Chinese lang not support by mu

* What actually happened,
flash quit

JohnGuan, 2020-02-07:
1.0.3 can run with Terminal

JohnGuan, 2020-02-07:
2020-02-07 19:37:02,039 - root:111(run) INFO: 

-----------------

Starting Mu 1.0.2
2020-02-07 19:37:02,041 - root:112(run) INFO: uname_result(system='Darwin', node='Johns-MBP.lan', release='19.2.0', version='Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64', machine='x86_64', processor='i386')
2020-02-07 19:37:02,041 - root:113(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2020-02-07 19:37:02,041 - root:114(run) INFO: Language code: zh_CN
2020-02-07 19:37:02,502 - mu.logic:543(__init__) INFO: Setting up editor.
2020-02-07 19:37:02,503 - mu.logic:562(__init__) INFO: Settings path: /Users/johnguan/Library/Application Support/mu/settings.json
2020-02-07 19:37:02,504 - mu.logic:563(__init__) INFO: Session path: /Users/johnguan/Library/Application Support/mu/session.json
2020-02-07 19:37:02,504 - mu.logic:564(__init__) INFO: Log directory: /Users/johnguan/Library/Logs/mu
2020-02-07 19:37:02,504 - mu.logic:565(__init__) INFO: Data directory: /Users/johnguan/Library/Application Support/mu
2020-02-07 19:37:02,520 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2020-02-07 19:37:02,867 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/johnguan/mu_code
2020-02-07 19:37:03,037 - mu.logic:627(restore_session) INFO: Restoring session from: /Users/johnguan/Library/Application Support/mu/session.json
2020-02-07 19:37:03,037 - mu.logic:628(restore_session) DEBUG: {'theme': 'day', 'mode': 'pygamezero', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 5}
2020-02-07 19:37:03,037 - mu.logic:649(restore_session) INFO: Loaded files.
2020-02-07 19:37:03,037 - mu.logic:653(restore_session) INFO: User defined environment variables: []
2020-02-07 19:37:03,037 - mu.logic:657(restore_session) INFO: Minify scripts on micro:bit? False
2020-02-07 19:37:03,179 - mu.logic:680(restore_session) INFO: Starting with blank file.
2020-02-07 19:37:03,201 - mu.logic:1150(change_mode) INFO: Workspace directory: /Users/johnguan/mu_code
2020-02-07 19:37:07,936 - mu.logic:811(get_dialog_directory) INFO: Using path for file dialog: /Users/johnguan/mu_code
2020-02-07 19:37:21,376 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: /Users/johnguan/Downloads/何老师-turtle例题/猫头鹰.py
2020-02-07 19:37:21,376 - mu.logic:715(_load) INFO: Loading script from: /Users/johnguan/Downloads/何老师-turtle例题/猫头鹰.py
2020-02-07 19:37:21,377 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2020-02-07 19:37:21,377 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2020-02-07 19:37:21,378 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\r\n'
2020-02-07 19:37:21,378 - mu.logic:789(_load) DEBUG: import turtle
turtle.pensize(5)
turtle.color("darkslategrey","darkslategrey")
turtle.begin_fill()
turtle.circle(100)
turtle.end_fill()
turtle.left(90)
turtle.forward(170)
turtle.left(90)
turtle.forward(55)

#左边耳朵
turtle.begin_fill()
turtle.right(40)
turtle.forward(13)
turtle.left(40)
turtle.forward(60)
turtle.circle(5,120)
turtle.forward(75)
#右边耳朵
turtle.left(65)
turtle.forward(190)
turtle.left(50)
turtle.forward(60)
turtle.circle(5,125)
turtle.forward(80)
turtle.end_fill()

#右边眼睛
turtle.color("black","white")
turtle.penup()
turtle.forward(30)
turtle.left(90)
turtle.forward(50)
turtle.pendown()
turtle.begin_fill()
turtle.circle(35)
turtle.end_fill()

turtle.left(90)
turtle.penup()
turtle.forward(30)
[... 83 lines cut ...]
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Users/johnguan/Desktop/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Users/johnguan/Desktop/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2020-02-07 22:12:54,452 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2020-02-07 22:12:58,832 - root:95(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/Applications/mu-editor.app/Contents/MacOS/mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.2', 'console_scripts', 'mu-editor')()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 111, in run
    setup_logging()
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 66, in setup_logging
    print(_('Logging to {}').format(LOG_FILE))
UnicodeEncodeError: 'ascii' codec can't encode characters in position 0-4: ordinal not in range(128)
2020-02-07 22:15:17,766 - root:112(run) INFO:

tmontes, 2020-02-09:
@JohnGuan, thanks for sharing the details of the failure you are observing.

A few notes:
* The log above is for Mu 1.0.2 but a similar failure is expected under 1.0.3.
* Not sure about 1.1.0alpha2 which may also fail for you but that will most probably be failing due to some other underlying condition.

If you want/need to stick with 1.0.2 / 1.0.3, please refer to [this note here](https://github.com/mu-editor/mu/issues/1000#issuecomment-583336514) where I guided another user in "hacking" their Mu installation in order to avoid the same exact failure in your log. With it, both 1.0.2 and 1.0.3 should run by "clicking in the Finder".

Regarding 1.1.0alpha2, I'm not sure. If you prefer to use that instead, please share the failure log, if any. :)

JohnGuan, 2020-02-20:
I can run 1.0.3 by "Hacking" app.py now, but it is a little hard to my student. I hope the problem can be fixed.

dybber, 2020-12-09:
@JohnGuan Could you try if the problem still exist in the development version of Mu available at http://mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/ ?

At least the line `print(_('Logging to {}').format(LOG_FILE))` no longer exist in the source code anymore, but perhaps you will get a similar error in a different location

dybber, 2021-02-02:
Just tried this myself, setting my preferred language to Chinese (simplified). I don't get this error, not in the current master and not with the most recent alpha release. I also tried creating this small script to replicate the error, and I don't get the issue there either:

```python
import gettext

language_code = 'zh'
localedir="./mu/locale"
LOG_FILE = "/Users/johnguan/Library/Logs/mu/mu.log"

gettext.translation("mu",
                    localedir=localedir,
                    languages=[language_code],
                    fallback=True).install()

print(_('Logging to {}').format(LOG_FILE))

```

However I'm on Mojave, and I think it's a Catalina issue, so someone on Catalina should try and confirm that the issue is solved in the current master.

tmontes, 2021-02-02:
@dybber,

...a bit in a rush, but IIRC, the issue cannot be replicated with your example. Underlying triggers:

* Path to log file contains non-ASCII bytes -- maybe Chinese user name, for example.
* Launching from Finder does not set a few env vars (that launching from the terminal does).
* Thusly Python defaults to US-ASCII text/binary codec, instead of UTF-8, for example!

dybber, 2021-02-02:
No, the path to the log-file is clearly visible in the pasted logs, where I copied it from: ```2020-02-07 19:37:02,504 - mu.logic:564(__init__) INFO: Log directory: /Users/johnguan/Library/Logs/mu``` 

So that shouldn't be the issue. My thought was that it must be an issue with how we use the `gettext`-module, or some kind of interaction with that particular version of `gettext` and Catalina.

I didn't know the that the ENV-vars could be the cause, so that's something to investigate.

tmontes, 2021-02-02:
> No, the path to the log-file is clearly visible in the pasted logs, where I copied it from:

Oh, of course, you are correct. The env vars are probably the culrpit, along with the embedded Python version...

Here's my observation/test combinations:

* Launch from Finder vs from Terminal.
* Mu 1.1.0a2 (embedded Python 3.6) vs 1.1.0b1 (embedded Python 3.7)

Test procedure:

* REPL
* `import os`
* `os.environ['LC_CTYPE']`

macOS Mojave:

* Finder launches 1.1.0a2 -- `LC_CTYPE` **not defined**.
* Terminal launches 1.1.0a2 -- `LC_CTYPE` is defined.
* Finder or terminal launches 1.1.0b2 -- `LC_CTYPE` is defined.

The way Python sets the default encoding has varied and been improved across revisions and platforms. However, it depends on a few env vars and, IME, the `LC_CTYPE` is relevant in the 1.1.0a2 Mu case.

From the experiment above, I feel that the original issue from @JohnGuan may be fixed in the current beta.

PS: Still a bit in a rush, but trying to help out with clearer details! :)

dybber, 2021-02-02:
Just tried to run the same scripts, after unsetting various environment variables, and the problem still doesn't show up:

```
$ unset LC_CTYPE
$ unset LC_ALL
$ unset LANG
$ python mu/test_script_issue1003.py
记录日志到 /Users/johnguan/Library/Logs/mu/mu.log
```

But again, I'm on Mojave and haven't been able to replicate the error anyways, so I still suspect something Catalina-related that I'm not able to debug.

ntoll, 2021-07-10:
Since we're unable to replicate this with the latest versions of Mu, I'm closing it. @JohnGuan please feel free to re-open if the most recent releases still have this problem. Thank you everyone for your contributions..! :+1:

closed 2021-07-10 by ntoll