mu-editor/mu #1004: Flask server not shut down on quit

mlibby, 2020-02-07:
* What you were trying to do?
Quit Mu. :)

* What steps you took to make this happen:
Start the simple example "Hello, Mu" Flask example, verify that it loads in the browser. It does. Quit Mu. Refresh browser on Hello, Mu page. Start a Flask web site up from another IDE (VS Code).

* What you expected to happen:
I expected the server to respond with the site launched by VS Code

* What actually happened:
First I got a 404, then on paring the URL down to http://localhost:5000/, I got the Hello Mu page. So it looks like the Flask process kicked off by Mu is still running even though Mu is not.

* Why this difference is problematic (it may not be a bug!):
Normally I'd expect a process started within my IDE to get killed when the IDE quits.

* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 
Running Windows 10 with Mu 1.1.0.alpha.2

Please remember to attach a **copy of the full log files for Mu**.

2020-02-07 17:14:58,745 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2020-02-07 17:14:58,745 - root:114(run) INFO: uname_result(system='Windows', node='whippet', release='10', version='10.0.18362', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 9, GenuineIntel')
2020-02-07 17:14:58,745 - root:115(run) INFO: Python path: ['C:\\Program Files\\Mu\\pkgs', 'C:\\Program Files\\Mu', 'C:\\Program Files\\Mu\\Python\\python36.zip', 'C:\\Program Files\\Mu\\Python\\DLLs', 'C:\\Program Files\\Mu\\Python\\lib', 'C:\\Program Files\\Mu\\Python', 'C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']
2020-02-07 17:14:58,745 - root:116(run) INFO: Language code: en_US
2020-02-07 17:14:59,262 - mu.logic:587(__init__) INFO: Setting up editor.
2020-02-07 17:14:59,262 - mu.logic:609(__init__) INFO: Settings path: C:\Users\m\AppData\Local\python\mu\settings.json
2020-02-07 17:14:59,262 - mu.logic:610(__init__) INFO: Session path: C:\Users\m\AppData\Local\python\mu\session.json
2020-02-07 17:14:59,262 - mu.logic:611(__init__) INFO: Log directory: C:\Users\m\AppData\Local\python\mu\Logs
2020-02-07 17:14:59,262 - mu.logic:612(__init__) INFO: Data directory: C:\Users\m\AppData\Local\python\mu
2020-02-07 17:14:59,292 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2020-02-07 17:15:00,649 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\m\mu_code
2020-02-07 17:15:00,649 - mu.logic:684(restore_session) INFO: Restoring session from: C:\Users\m\AppData\Local\python\mu\session.json
2020-02-07 17:15:00,649 - mu.logic:685(restore_session) DEBUG: {'theme': 'day', 'mode': 'web', 'paths': ['C:\\Users\\m\\python\\hello.py'], 'envars': [['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': -1, 'y': -9, 'w': 1920, 'h': 1001}}
2020-02-07 17:15:00,649 - mu.logic:776(_load) INFO: Loading script from: C:\Users\m\python\hello.py
2020-02-07 17:15:00,665 - mu.logic:311(read_and_decode) DEBUG: Trying to decode with utf-8
2020-02-07 17:15:00,665 - mu.logic:314(read_and_decode) INFO: Decoded with utf-8
2020-02-07 17:15:00,665 - mu.logic:327(read_and_decode) DEBUG: Detected newline '\n'
2020-02-07 17:15:00,665 - mu.logic:860(_load) DEBUG: from flask import Flask, render_template

app = Flask(__name__)

@app.route("/")
def index():
    return render_template("index.html")

2020-02-07 17:15:01,609 - mu.logic:706(restore_session) INFO: Loaded files.
2020-02-07 17:15:01,609 - mu.logic:710(restore_session) INFO: User defined environment variables: [['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']]
2020-02-07 17:15:01,609 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False
2020-02-07 17:15:01,690 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\m\mu_code
2020-02-07 17:15:10,509 - mu.modes.web:164(start_server) DEBUG: from flask import Flask, render_template

app = Flask(__name__)

@app.route("/")
def index():
    return render_template("index.html")

2020-02-07 17:15:10,541 - mu.interface.panes:691(start_process) INFO: Running script: 
2020-02-07 17:15:10,541 - mu.interface.panes:696(start_process) INFO: Command args: []
2020-02-07 17:15:10,556 - mu.interface.panes:741(start_process) INFO: Unable to set Python paths. Python's USER_SITE not enabled. Check configuration with administrator.
2020-02-07 17:15:10,556 - mu.interface.panes:755(start_process) INFO: Running with environment variables: [['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ['FLASK_APP', 'hello.py'], ['FLASK_ENV', 'development'], ['PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions'], ('FLASK_APP', 'hello.py'), ('FLASK_ENV', 'development'), ('PYTHONPATH', 'C:\\Program Files\\Mu\\pkgs;C:\\Program Files\\Mu;C:\\Program Files\\Mu\\Python\\python36.zip;C:\\Program Files\\Mu\\Python\\DLLs;C:\\Program Files\\Mu\\Python\\lib;C:\\Program Files\\Mu\\Python;C:\\Users\\m\\AppData\\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions')]
2020-02-07 17:15:10,556 - mu.interface.panes:758(start_process) INFO: Working directory: C:\Users\m\python
[... cut ...] \\Local\\python\\mu\\site-packages;C:\\Program Files\\Mu\\pkgs\\IPython\\extensions']]
2020-02-07 17:25:08,023 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False
2020-02-07 17:25:08,109 - mu.logic:1266(change_mode) INFO: Workspace directory: C:\Users\m\mu_code
2020-02-07 17:25:36,091 - mu.logic:1161(show_admin) INFO: Showing admin with logs from C:\Users\m\AppData\Local\python\mu\Logs\mu.log
2020-02-07 17:25:36,091 - mu.logic:163(installed_packages) INFO: Packages found: []

devdanzin, 2020-12-17:
Confirmed on Windows, seems to leak two Python processes with command line `"-m flask run"` per Run/Stop invocation. Looks like we're doing things right, but don't account for children processes. We could figure out a `wmic` plus `taskkill` invocation (or a pure Python equivalent) if it's a Windows-only issue.

If it's a cross-platform issue (and maybe if we have enough use cases to justify a new dependency), `psutil` allows fixing this in half a dozen lines.

devdanzin, 2021-01-11:
Cannot reproduce the issue on Linux: three child processes are created (just like in Windows), but they're all killed (while in Windows two leak). The solution below can stop the leak, but I'd like to hear what people think we should do to fix this: use it for all OSes, only on Windows, fix Windows via improved `os.kill` instead...?

This is a Flask-native solution that works on Windows: [create a shutdown endpoint](http://web.archive.org/web/20190706125149/http://flask.pocoo.org/snippets/67) (from [SO](https://stackoverflow.com/a/64597254)).  Something like:

```python
from flask import request

BASE_URL = "http://127.0.0.1:5000"
SHUTDOWN = "/shutdown"

def shutdown_server():
    func = request.environ.get('werkzeug.server.shutdown')
    if func is None:
        raise RuntimeError('Not running with the Werkzeug Server')
    func()

@app.route(SHUTDOWN, methods=['POST'])
def shutdown():
    shutdown_server()
    return 'Server shutting down...'
```

If we can add that endpoint to user apps, we can later [fire a request](https://stackoverflow.com/a/64860348) against it and it'll kill the flask children processes (we can even run that in a loop to cleanup accumulated leaks if users quit Mu without cleaning flask instances up).

```python
from urllib.request import Request, urlopen
BASE_URL = "http://127.0.0.1:5000"
SHUTDOWN = "/shutdown"

def request_shutdown(url=BASE_URL + SHUTDOWN):
    req = Request(url, method="POST")
    res = urlopen(req)
    content = res.read()
    print(content)
```

dybber, 2021-01-28:
I think we shouldn't address this only for Flask mode. Potentially the problem also affects other modes? 

As I see it, it should be addressed in the ProcessPane implementation, such that we make sure all processes launched are also closed again. I'm not sure how you discover that Qt has been instructed to close, or it's enough to add a __del__ destructor for ProcessPane that closes `self.process` if it's not closed already?

Btw. right now it's rather ugly that the ProcessPane creates the QProcess object:
https://github.com/mu-editor/mu/blob/2d4b2b1214bfae248181294fc3dc8c8de409feea/mu/interface/panes.py#L875

and launches the process:
https://github.com/mu-editor/mu/blob/2d4b2b1214bfae248181294fc3dc8c8de409feea/mu/interface/panes.py#L932

but when closing the process again, it's the modes responsibility.
https://github.com/mu-editor/mu/blob/2d4b2b1214bfae248181294fc3dc8c8de409feea/mu/modes/python3.py#L231-L233

It should really be the mode signaling to the ProcessPane that it should close.

On Mac/Linux all subprocesses should be killed automatically, but it would be more elegant to just always make sure we close the processes we launch.

ntoll, 2021-07-11:
Fixed by merging #1319

closed 2021-07-11 by ntoll