mu-editor/mu #1005: Question: Elif

GoldBranches, 2020-02-13:
The program that I wrote must not be written properly! *steps I took: I wrote a program that would request someone's age and then reply appropriately * What I expected: When I type 23, I expect the response to be "Hang with people in their 20's! There are predators out there." * What actually happened: Regardless of the age I type, it says "May I ask you on a date? What have you learned?" * Why this difference is problematic: asking someone on a date even if that person is an inappropriate age. * Technical details like the version of Mu you're using, your OS version and other aspects of the context in which Mu was running: I am in OS 10.14.6 and the Mu is Alpha 1.1.0 alpha.2

Here is the program: 

print("May I ask how old you are?")
yourAge = input()
if "age" < "18":
    print("Your mother will do what you want, okay? Just go home.")
elif "age" < "27":
    print("Hang with people in their 20's! There are predators out there.")
elif "age" > "27":
    print("May I ask you on a date? What have you learned?")
else:
    print("You are a gooooood kid!")

ZanderBrown, 2020-02-13:
This really isn't the place for general python help, the internet is full of such places

But since your here...
```python
print("May I ask how old you are?")
yourAge = input()
if "age" < "18":
    print("Your mother will do what you want, okay? Just go home.")
elif "age" < "27":
    print("Hang with people in their 20's! There are predators out there.")
elif "age" > "27":
    print("May I ask you on a date? What have you learned?")
else:
    print("You are a gooooood kid!")
```

First issue is that your using the string `"age"` instead of the variable (`yourAge`)

Secondly you seem to be comparing strings expecting them to work as numbers, they won't!

You need to convert the users input to a number (here an integer will probably do) that you can compare with other numbers, something like
```python
print("May I ask how old you are?")
age = int(input())
if age < 18:
    print("Your mother will do what you want, okay? Just go home.")
elif age < 27:
    print("Hang with people in their 20's! There are predators out there.")
elif age > 27:
    print("May I ask you on a date? What have you learned?")
else:
    print("You are a gooooood kid!")
```

GoldBranches, 2020-02-14:
Thanks! The solution didn't quite work but I understand that this isn't the place for it!

closed 2020-02-14 by GoldBranches