mu-editor/mu #1007: 'Print' Error

GoldBranches, 2020-02-15:
* I was trying to apply sample code to generate numbers. 
*I copied the code and made adjustments only to the text string and to the equation.
* I expected different alphanumeric strings to print as text strings.
* What actually happened was that the print function kept printing the string without applying the math equation.
* Why this difference is problematic: The code works in the sample video that was given by the library of congress book, so I believe that it is the editor.
* I am in 10.14.6 Mojave (Mac) and Mu Editor 1.0.1 Alpha.2

**Here is the sample code from the book:** 

print('My name is')
i=0
while i<5
    print('Jimmy Five Times (' + str(i) + ')')
i=i+1

**And here are my adjustments:** 

print('here are the codes')
i = 1
while i < 200:
    print('aa5d-s2jf-8s62-tfes- (' + str(i) + ')')
i = (i + 1)**2

ZanderBrown, 2020-02-15:
Please always [paste code in codeblocks](https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks), especially when python is so dependent on whitespace

Would also be nice to have a link to this "library of congress book"

In your first block...

```python
print('My name is')
i=0
while i<5
    print('Jimmy Five Times (' + str(i) + ')')
    i=i+1
```
...you are missing a `:` from `while i<5` which should have been reported as a [syntax error](https://docs.python.org/3/tutorial/errors.html#syntax-errors) when running the code (or rather trying to)

The second block hovever is perfectly valid

```python
print('here are the codes')
i = 1
while i < 200:
    print('aa5d-s2jf-8s62-tfes- (' + str(i) + ')')
    i = (i + 1)**2
```
And produces
```
here are the codes
aa5d-s2jf-8s62-tfes- (1)
aa5d-s2jf-8s62-tfes- (4)
aa5d-s2jf-8s62-tfes- (25)
```
Now it might be [this isn't what you wanted](https://en.wikipedia.org/wiki/Logic_error) but it's valid code and Python has interpreted it correctly

... which in itself raises an important point: _Python_ has interpreted it correctly, not Mu

At the end of the day Mu is just a glorified text area, Mu doesn't actually "run" the code but rather says "oi python, run this" 

----

I would again like to remind you this is for problems with Mu, [other forums exist](https://www.python.org/about/help/) for general python help

GoldBranches, 2020-02-17:
Thanks, Mu and IDLE can alternate.

closed 2020-02-17 by GoldBranches

ZanderBrown, 2020-02-17:
Huh?