mu-editor/mu #1012: Possible issue with mu editor

GoldBranches, 2020-02-26:
If you are reporting a bug, we would like to know:

* What you were trying to do: I copied in the sample code and have been getting intermittently changing corrections. 
* What steps you took to make this happen: I typed in the sample code and then made corrections as directed by the editor
* What you expected to happen: the code to run
* What actually happened: the same help text would come up to point out an error
* Why this difference is problematic (it may not be a bug!): there is not change to the help text even though I fixed the code
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running: I am using Mojave Mac OS 10.14.6 and Mu 1.1.0.alpha.2

This is the code I type in: 
`if name ==" :`

And the recurring error is: "Syntax error. Python cannot understand this line. Check for missing characters!"

Thank you.

GoldBranches, 2020-02-26:
Sorry, I have made this error twice! The double quote " in the teaching code is actually a single quote '  '

closed 2020-02-26 by GoldBranches

ukBaz, 2020-02-26:
Hi @GoldBranches ,

This is a general Python programming issue and not an issue with the tool. 
When you are learning to code you would be better looking in other forums that will help you more. Only if it is clear that it is an issue with Mu should you come here:
Here is an example of the help you can get in other forums:
https://stackoverflow.com/a/9573283