mu-editor/mu #1013: I cannot open mu...

Sanyok6, 2020-02-26:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

ZanderBrown, 2020-02-26:
Looks like you hit submit too soon, you've forgotten to tell us about your problem :-)

tjguk, 2020-06-11:
Closing this as there's been no more information since the original report. Please open a fresh issue if the problem persists, adding as much information as you can to help us help you

closed 2020-06-11 by tjguk