mu-editor/mu #1016: microbit error

gowtham1234, 2020-03-22:
i keep getting this issue for some reason, and eventually it will have finished flashing but nothing much happens. 
it will leave me with the message "there was a problem flashing the microbit, please do to disconnect until flashing has completed. please check the logs for more info"

2020-03-22 17:18:10,942 - root:113(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2020-03-22 17:18:10,944 - root:114(run) INFO: uname_result(system='Darwin', node='gowthams-MacBook-Air-4.local', release='19.3.0', version='Darwin Kernel Version 19.3.0: Thu Jan  9 20:58:23 PST 2020; root:xnu-6153.81.5~1/RELEASE_X86_64', machine='x86_64', processor='i386')
2020-03-22 17:18:10,944 - root:115(run) INFO: Python path: ['/Applications/mu-editor.app/Contents/MacOS', '/Applications/mu-editor.app/Contents/Resources/python/lib/python36.zip', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/lib-dynload', '/Applications/mu-editor.app/Contents/Resources/python/lib/python3.6/site-packages', '/Applications/mu-editor.app/Contents/Resources/app', '/Applications/mu-editor.app/Contents/Resources/app_packages', '/Users/gowthamvigneshwar/Library/Application Support/mu/site-packages', '/Applications/mu-editor.app/Contents/Resources/app_packages/IPython/extensions']
2020-03-22 17:18:10,944 - root:116(run) INFO: Language code: en_KE
2020-03-22 17:18:11,508 - mu.logic:587(__init__) INFO: Setting up editor.
2020-03-22 17:18:11,509 - mu.logic:609(__init__) INFO: Settings path: /Users/gowthamvigneshwar/Library/Application Support/mu/settings.json
2020-03-22 17:18:11,509 - mu.logic:610(__init__) INFO: Session path: /Users/gowthamvigneshwar/Library/Application Support/mu/session.json
2020-03-22 17:18:11,510 - mu.logic:611(__init__) INFO: Log directory: /Users/gowthamvigneshwar/Library/Logs/mu
2020-03-22 17:18:11,510 - mu.logic:612(__init__) INFO: Data directory: /Users/gowthamvigneshwar/Library/Application Support/mu
2020-03-22 17:18:11,528 - mu.logic:625(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2020-03-22 17:18:11,768 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/gowthamvigneshwar/mu_code
2020-03-22 17:18:11,768 - mu.logic:684(restore_session) INFO: Restoring session from: /Users/gowthamvigneshwar/Library/Application Support/mu/session.json
2020-03-22 17:18:11,769 - mu.logic:685(restore_session) DEBUG: {'theme': 'day', 'mode': 'microbit', 'paths': ['/Users/gowthamvigneshwar/mu_code/000.py', '/Users/gowthamvigneshwar/mu_code/radio.py'], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 1, 'window': {'x': 144, 'y': 90, 'w': 1152, 'h': 720}}
2020-03-22 17:18:11,769 - mu.logic:776(_load) INFO: Loading script from: /Users/gowthamvigneshwar/mu_code/000.py
2020-03-22 17:18:11,770 - mu.logic:311(read_and_decode) DEBUG: Trying to decode with utf-8
2020-03-22 17:18:11,770 - mu.logic:314(read_and_decode) INFO: Decoded with utf-8
2020-03-22 17:18:11,771 - mu.logic:327(read_and_decode) DEBUG: Detected newline '\n'
2020-03-22 17:18:11,771 - mu.logic:860(_load) DEBUG: # Write your code here :-)
display.on()

2020-03-22 17:18:11,896 - mu.logic:776(_load) INFO: Loading script from: /Users/gowthamvigneshwar/mu_code/radio.py
2020-03-22 17:18:11,898 - mu.logic:311(read_and_decode) DEBUG: Trying to decode with utf-8
2020-03-22 17:18:11,898 - mu.logic:314(read_and_decode) INFO: Decoded with utf-8
2020-03-22 17:18:11,899 - mu.logic:327(read_and_decode) DEBUG: Detected newline '\n'
2020-03-22 17:18:11,899 - mu.logic:860(_load) DEBUG: # Write your code here :-)
from microbit import *
import radio

radio.on()
radio.config(address=0xBEEFFEEB)

while True :
    details = radio.receive_full()
    if button_a.was_pressed():
        radio.send("MINI")
    if details:
        pin0.write_digital(1)
        msg, rssi, timestamp = details
        display.scroll(rssi)
        pin0.write_digital(0)

2020-03-22 17:18:11,917 - mu.logic:706(restore_session) INFO: Loaded files.
2020-03-22 17:18:11,917 - mu.logic:710(restore_session) INFO: User defined environment variables: []
2020-03-22 17:18:11,917 - mu.logic:714(restore_session) INFO: Minify scripts on micro:bit? False
2020-03-22 17:18:12,004 - mu.logic:1266(change_mode) INFO: Workspace directory: /Users/gowthamvigneshwar/mu_code
2020-03-22 17:23:05,177 - mu.logic:950(save_tab_to_file) INFO: Saving script to: /Users/gowthamvigneshwar/mu_code/radio.py
2020-03-22 17:23:05,190 - mu.logic:951(save_tab_to_file) DEBUG:  Write your code here :-)# This program was written by Giri Venkataramanan# ECE 210 Lab 6# When you press the PB, the RGB LED sequences between Dark, Red, Green, Blue and mixed colors# During the colored states, the POT can be used to set the brightness of each color# Pins 1-2-3 are to RGB of common cathode LED drive# Pin 4 to Pot# Pin 7 to reading the push-button switch with debounced hardwarefrom microbit import *display.off()# Defining the state identiificationIDLE, RED, GREEN, BLUE, WHITE = range(5)# Initializing the statesstate = IDLE# Defining the idle state functiondef handle_idle_state():    global state    sleep(50)    pin1.write_analog(0)    sleep(50)    pin2.write_analog(0)    sleep(50)    pin3.write_analog(0)    if pin7.read_digital():        state = RED        sleep(500)# Defining the red state functiondef handle_red_state():    global state, angle_R    angle_R = pin4.read_analog()    sleep(50)    pin1.write_analog(angle_R)    pin2.write_analog(0)    pin3.write_analog(0)    if pin7.read_digital():        state = GREEN        sleep(500)# Defining the green state functiondef handle_green_state():    global state, angle_G    angle_G = pin4.read_analog()    sleep(50)    pin2.write_analog(angle_G)    pin1.write_analog(0)    pin3.write_analog(0)    if pin7.read_digital():        state = BLUE        sleep(500)# Defining the blue state functiondef handle_blue_state():
    global state, angle_B    angle_B = pin4.read_analog()    sleep(50)    pin3.write_analog(angle_B)    pin1.write_analog(0)    pin2.write_analog(0)    if pin7.read_digital():        state = WHITE        sleep(500)# Defining the white state functiondef handle_white_state():    global state, angle_R, angle_G, angle_B    sleep(50)    pin1.write_analog(angle_R)    sleep(50)    pin2.write_analog(angle_G)    sleep(50)    pin3.write_analog(angle_B)    if pin7.read_digital():        state = IDLE        sleep(500)# Main state machinewhile True:    if state == IDLE:        handle_idle_state()    elif state == RED:        handle_red_state()    elif state == GREEN:        handle_green_state()    elif state == BLUE:        handle_blue_state()    elif state == WHITE:        handle_white_state()
2020-03-22 17:23:05,198 - mu.logic:1295(autosave) INFO: Autosave detected and saved changes in /Users/gowthamvigneshwar/mu_code/radio.py.
2020-03-22 17:23:09,955 - mu.logic:1060(check_code) INFO: Checking code.
2020-03-22 17:23:09,964 - mu.logic:1066(check_code) INFO: {0: [{'message': 'Syntax error. Python cannot understand this line. Check for missing characters!', 'line_no': 0, 'column': 0, 'source': ' Write your code here :-)# This program was written by Giri Venkataramanan# ECE 210 Lab 6# When you press the PB, the RGB LED sequences between Dark, Red, Green, Blue and mixed colors# During the colored states, the POT can be used to set the brightness of each color# Pins 1-2-3 are to RGB of common cathode LED drive# Pin 4 to Pot# Pin 7 to reading the push-button switch with debounced hardwarefrom microbit import pin15, pin2, pin0, pin1,  pin3, pin6, pin4, i2c, pin5, pin7, pin8, Image, pin9, pin14, pin16, reset, pin19, temperature, sleep, pin20, button_a, button_b, running_time, accelerometer, display, uart, spi, panic, pin13, pin12, pin11, pin10, compassdisplay.off()# Defining the state identiificationIDLE, RED, GREEN, BLUE, WHITE = range(5)# Initializing the statesstate = IDLE# Defining the idle state functiondef handle_idle_state():    global state    sleep(50)    pin1.write_analog(0)    sleep(50)    pin2.write_analog(0)    sleep(50)    pin3.write_analog(0)    if pin7.read_digital():        state = RED        sleep(500)# Defining the red state functiondef handle_red_state():    global state, angle_R    angle_R = pin4.read_analog()    sleep(50)    pin1.write_analog(angle_R)    pin2.write_analog(0)    pin3.write_analog(0)    if pin7.read_digital():        state = GREEN        sleep(500)# Defining the green state functiondef handle_g [... cut ...] n    sleep(50)\\n    pin3.write_analog(angle_')", "f(b'B)\\n    if pin7.read_digital():\\n        state = IDLE\\n        slee')", "f(b'p(500)\\n\\n# Main state machine\\nwhile True:\\n    if state == IDLE:\\n ')", "f(b'       handle_idle_state()\\n    elif state == RED:\\n        handle')", "f(b'_red_state()\\n    elif state == GREEN:\\n        handle_green_state')", "f(b'()\\n    elif state == BLUE:\\n        handle_blue_state()\\n    elif ')", "f(b'state == WHITE:\\n        handle_white_state(

dybber, 2020-06-13:
Hi @gowtham1234
Thanks for the bug report. Could I get you to edit your bug report to format the code? Right now it's very hard to read and replicate the problematic code, as all comments are interpreted as Markdown headings, and indentation is removed.

In the Github editor, select the lines of code and click the < > button, or manually insert ``` on the line before as well as the line after the code example.

carlosperate, 2020-08-03:
I think I've seen this issue in a faulty micro:bit, that was mostly working, but for some reason took a long time to reunumerate and be accessible as a MSD or serial port.

If that is the problem the simplest solution would be to increase the wait period between flashing MicroPython and trying to send the files via serial. We should probably do a bit better than try a few times with an increasing delay and fail after 3 attempts.

@gowtham1234 just in case this might be similar, do you a second micro:bit you could try with the same Mu installation?

carlosperate, 2021-03-04:
I didn't notice from the log this is a permission issue, @gowtham1234 could you try the steps from the "Disk Permissions in OSX Catalina" section in this link? https://codewith.mu/en/howto/1.0/install_macos#catalina-permissions

carlosperate, 2021-03-16:
The "operation not permitted" issue listed in the log should be fixed in the latest release: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.2 🎉 

I'll close this as resolved, but please do reopen it if you are still having this issue with the v1.1.0-beta.2.

closed 2021-03-16 by carlosperate