mu-editor/mu #1020: MU Crashes when I click the REPL button

Axelsele, 2020-03-26:
Hey all,
Super super beginner. 
I downloaded MU, and then I tried to click the REPL button and the whole program just quit (crashed). 
Not sure what's required for you to help me but i have the problem report. 

PASTED BELOW (sorry it being super long!) 
Hope someone can help me out. 
(i downloaded python 3.8.2, then MU, then upon opening MU i clicked on REPL and it crashed)
Thanks!

Process:               python3 [12875]
Path:                  /Applications/mu-editor.app/Contents/MacOS/../Resources/python/bin/python3
Identifier:            mu.codewith.editor.mu-editor
Version:               1.0.3 (1)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           python3 [12875]
User ID:               501

Date/Time:             2020-03-26 18:51:16.021 +0100
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Anonymous UUID:        B359845B-DFAA-C1A0-89C4-F4E1B7A68806

Sleep/Wake UUID:       C53B3630-BFF5-45F6-B755-C6A6CD2F6E8A

Time Awake Since Boot: 340000 seconds
Time Since Wake:       4000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000c300003a
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [12875]

VM Regions Near 0xc300003a:
--> 
    __TEXT                 000000010e1d6000-000000010e393000 [ 1780K] r-x/rwx SM=COW  /Applications/mu-editor.app/Contents/Resources/python/bin/python3

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   QtCore                        	0x000000010f5b08f5 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 181
1   QtCore                        	0x000000010f60bf2f QTimerInfoList::activateTimers() + 991
2   libqcocoa.dylib               	0x00000001158d79e2 QCocoaEventDispatcherPrivate::activateTimersSourceCallback(void*) + 18
3   com.apple.CoreFoundation      	0x00007fff33649b21 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
4   com.apple.CoreFoundation      	0x00007fff33649ac0 __CFRunLoopDoSource0 + 103
5   com.apple.CoreFoundation      	0x00007fff336498d4 __CFRunLoopDoSources0 + 209
6   com.apple.CoreFoundation      	0x00007fff33648740 __CFRunLoopRun + 1272
7   com.apple.CoreFoundation      	0x00007fff33647bd3 CFRunLoopRunSpecific + 499
8   com.apple.HIToolbox           	0x00007fff3219e65d RunCurrentEventLoopInMode + 292
9   com.apple.HIToolbox           	0x00007fff3219e39d ReceiveNextEventCommon + 600
10  com.apple.HIToolbox           	0x00007fff3219e127 _BlockUntilNextEventMatchingListInModeWithFilter + 64
11  com.apple.AppKit              	0x00007fff3080feb4 _DPSNextEvent + 990
12  com.apple.AppKit              	0x00007fff3080e690 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1352
13  com.apple.AppKit              	0x00007fff308003ae -[NSApplication run] + 658
14  libqcocoa.dylib               	0x00000001158d8773 QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2579
15  QtCore                        	0x000000010f5ac99f QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 431
16  QtCore                        	0x000000010f5b0f22 QCoreApplication::exec() + 130
17  QtWidgets.abi3.so             	0x00000001121ed172 meth_QApplication_exec_(_object*, _object*) + 82
18  python3                       	0x000000010e22e40b _PyCFunction_FastCallDict + 491
19  python3                       	0x000000010e2b2037 call_function + 439
20  python3                       	0x000000010e2ae988 _PyEval_EvalFrameDefault + 24952
21  python3                       	0x000000010e2b2e7c _PyEval_EvalCodeWithName + 3436
22  python3                       	0x000000010e2b3801 fast_function + 657
23  python3                       	0x000000010e2b2011 call_function + 401
[... 559 lines cut ...]
Stack                             19.6M        8 
VM_ALLOCATE                        440K       21 
__DATA                            28.9M      399 
__DATA_CONST                        20K        1 
__FONT_DATA                          4K        1 
__LINKEDIT                       367.4M       84 
__OBJC_RO                         32.0M        1 
__OBJC_RW                         1780K        2 
__TEXT                           223.6M      383 
__UNICODE                          564K        1 
mapped file                      371.3M       30 
shared memory                      640K       15 
===========                     =======  ======= 
TOTAL                              1.2G     1292 
TOTAL, minus reserved VM space     1.2G     1292 

Model: MacBookPro14,1, BootROM 203.0.0.0.0, 2 processors, Dual-Core Intel Core i5, 2,3 GHz, 8 GB, SMC 2.43f7
Graphics: kHW_IntelIrisGraphics640Item, Intel Iris Plus Graphics 640, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3332443250462D3039
Memory Module: BANK 1/DIMM0, 4 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C3531324D3332443250462D3039
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x170), Broadcom BCM43xx 1.0 (7.77.106.3 AirPortDriverBrcmNIC-1435.3)
Bluetooth: Version 7.0.2f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 41.3

carlosperate, 2020-08-03:
Thank you for the report @Axelsele, I hope you don't mind I've done a small edit to your original messaging to format the log in a more readable way.

Did this happen as soon as Mu was installed, and the first time you opened Mu? Or instead, were you able to use Mu at least once before this happened?

I see some unicode mentions in the stack trace, do you have your operating system configure in a non-english language?

carlosperate, 2020-08-03:
Also, this looks similar to https://github.com/mu-editor/mu/issues/584. I haven't looked into it properly, but it'd be good to link up these two issues in case it's the same.

carlosperate, 2021-03-29:
Hi @Axelsele,

We've updated how the Python REPL works in 1.1.0-beta.2, would you be able to install the latest release and see if that works?
https://github.com/mu-editor/mu/releases/

Thanks!

carlosperate, 2021-04-26:
We haven't received any response to this ticket since its creation and as the Jupyter kernel configuration has been updated I believe this issue should be fixed in the latest release.

Thanks for the original report @Axelsele, please feel free to reopen if you are still encountering this issue in the latest beta release.

closed 2021-04-26 by carlosperate