mu-editor/mu #1023: REPL not working

1104045928, 2020-04-07:
I install the mu-editor with python 3.6.9 on Ubuntu system.
After I installed it, I find that I cannot type anything in REPL!
And nothing is shown in the interactive shell. It is totally blank.
[image]

carlosperate, 2020-08-03:
Thank you for the report @1104045928.

Could you send us a copy of the Mu log? that would help us identify the issue.
If you could open the Mu editor, then open the REPL (as shown in this creenshot), and the copy the log from the settings that would help us find the issue at the end of the log.

1104045928, 2020-08-03:
Sorry I didn't get anything. I didn't send report to you, I think.

carlosperate, 2020-08-03:
Sorry, I'm not sure I quite understood. Do you still have this issue with the Mu editor? If you are, would you be able to send us the log?

1104045928, 2020-08-03:
No, thank you!

carlosperate, 2020-08-03:
Okay, thanks for the quick reply! I'll close the issue then

closed 2020-08-03 by carlosperate

Moey27, 2020-10-13:
hey I have the same problem

Moey27, 2020-10-13:
@carlosperate  
Ive got the same problm, the blank Repl and cant type anything. How'd you fix it?

carlosperate, 2020-10-13:
Hi @Moey27,

As this issue is close, could you open a new one indicating what issue you are encountering, which version of Mu you are runnning, what mode is selected, the log etc? The new issue template indicates all the things that will help us diagnose the issue.