mu-editor/mu #1025: My program won't run even when I close and re-open the program. I also remembered to save it

piuser123, 2020-04-11:
If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

carlosperate, 2020-07-25:
Hi @piuser123, sorry it took this long to respond, all the Mu maintainers do this in the (limited) spare time we have.

Could you elaborate a bit more on how you are trying to run your program? how does it fail? perhaps share your program as well so that we can replicate, and please share the Mu log as well?

carlosperate, 2020-11-09:
Thank you for the report, but we don't have much information to investigate. I'll close this issue as inactive, but please feel free to reopen if you can provide a bit more info on the crash.

closed 2020-11-09 by carlosperate