mu-editor/mu #1028: Column select, spaces, unstable, crash.

MvGulik, 2020-04-19:
1) New file.
2) Add single line with only spaces. (_~64 spaces should do_) (_same with tabs_)
3) Initiate column select (_CTRL+Mouse-Select_) on that lines (_at or near beginning of line_).
4) Extend selection with mouse towards the end of the line.
5) Tends to crash at some point.

+ On multiple blank lines. Extending the selection over more than one line first, seems to bypass this issue. (_as long as multiple lines are kept selected_)

Terminal output:
`Logging to /home/<user>/.cache/mu/log/mu.log` (_not included, did not contain relevant/related data_)
`free(): invalid next size (fast)`
`Aborted (core dumped)`

```
Process 6851 (mu-editor) of user 1000 dumped core.
Stack trace of thread 6851:
#0  0x00007f4c3f5c7e97 __GI_raise (libc.so.6)
#1  0x00007f4c3f5c9801 __GI_abort (libc.so.6)
#2  0x00007f4c3f612897 __libc_message (libc.so.6)
#3  0x00007f4c3f61990a malloc_printerr (libc.so.6)
#4  0x00007f4c3f620f60 _int_free (libc.so.6)
#5  0x00007f4c2b8587f0 n/a (/home/<user>/test_venv/lib/python3.6/site-packages/PyQt5/Qt/lib/libQt5XcbQpa.so.5)
#6  0x00000000036283e0 n/a (n/a)
```

+ Suspected issue source: QScintilla (_at least v2.11.3 & 2.11.4_)

(_Linux Mint 19.3 cinnamon_)

(_See also https://github.com/matkuki/ExCo/issues/33 => Other effected code editor**s**_)

----

pip list:
```
Package          Version   
---------------- ----------
appdirs          1.4.3     
backcall         0.1.0     
certifi          2020.4.5.1
chardet          3.0.4     
colorzero        1.1       
cycler           0.10.0    
decorator        4.4.2     
docutils         0.16      
eric-ide         20.4      
gpiozero         1.5.1     
guizero          1.1.0     
idna             2.9       
ipykernel        5.2.1     
ipython          7.13.0    
ipython-genutils 0.2.0     
jedi             0.17.0    
jupyter-client   6.1.3     
jupyter-core     4.6.3     
kiwisolver       1.2.0     
Markdown         3.2.1     
matplotlib       2.2.2     
mu-editor        1.0.3     
nudatus          0.0.4     
numpy            1.18.2    
parso            0.7.0     
pexpect          4.8.0     
pgzero           1.2       
pickleshare      0.7.5     
pigpio           1.46      
Pillow           7.1.1     
pip              20.0.2    
pkg-resources    0.0.0     
prompt-toolkit   3.0.5     
ptyprocess       0.6.0     
pycodestyle      2.4.0     
pyflakes         2.0.0     
pygame           1.9.6     
Pygments         2.6.1     
pyparsing        2.4.7     
PyQt5            5.14.1    
PyQt5-sip        12.7.2    
PyQtChart        5.14.0    
PyQtWebEngine    5.14.0    
pyserial         3.4       
python-dateutil  2.8.1     
pytz             2019.3    
pyzmq            19.0.0    
QScintilla       2.11.4    
qtconsole        4.3.1     
requests         2.23.0    
semver           2.9.1     
setuptools       39.0.1    
six              1.14.0    
tornado          6.0.4     
traitlets        4.3.3     
urllib3          1.25.9    
wcwidth          0.1.9     
```

MvGulik, 2020-04-22:
Matkuki of Ex.Co editor has found and added additional information about this issue.

carlosperate, 2020-04-24:
Thanks for the report @MvGulik!

Can't replicate in macOS with v1.1.0-alpha.2, but sounds like this could be a C/C++ issue in Qscintilla on Linux: https://github.com/matkuki/ExCo/issues/33#issuecomment-617360648

We still need to reproduce in Mu on Linux to make sure it's the same problem, but I'd be surprised if we can't reproduce (at least in debian-based distros).

@MvGulik are you planing to report this as well in the Qscintilla mailing list?

MvGulik, 2020-04-24:
> @MvGulik are you planing to report this as well in the Qscintilla mailing list?

Considering eric6' s developer already opened a related thread on it, and also seems to have had more Qscintilla mailing list contacts. I figured I first give him to opportunity to follow-up on it. On top of that, he has better experience and skill for this. (_kinda think you all have in this case_)

But ... if need be, yes I will.
Although, as I'm not having any Qscintilla experience(_code-wise_), so I'm not sure if I can provide what was asked at the Qscintilla mailing list. (_at least not yet_)

(_Mmm, I see eric6's dev is at least not on holiday ... but has not reacted to my update-post yet_)

MvGulik, 2020-04-24:
> I'm not sure if I can provide what was asked at the Qscintilla mailing list.

But than again, Matkuki's reports might do the trick too.
Will forward those (_including links_) in a days or so to the Qscintilla mailing list. (_if you like me to wait with that, just let me know_)

MvGulik, 2020-04-24:
Actually ... it turns out eric6' s dev already posted a follow-up post on the Qscintilla mailing list.
https://www.riverbankcomputing.com/pipermail/qscintilla/2020-April/001451.html

carlosperate, 2020-05-02:
Thanks @MvGulik, looks like at the moment Qscintilla dev is still in the "trying to replicate" phase. We'll keep an eye.