mu-editor/mu #1029: PyQt5 5.14.2 replaced with PyQt5 5.14.1 on pip install (?)

MvGulik, 2020-04-19:
Kinda wondering why running `pip install mu-editor` (_with active venv_) would replaced PyQt5 5.14.2 with PyQt5 5.14.1 ?

`mu-editor        1.0.3 `

```
  Attempting uninstall: pyqt5
    Found existing installation: PyQt5 5.14.2
    Uninstalling PyQt5-5.14.2:
      Successfully uninstalled PyQt5-5.14.2
    Running setup.py install for nudatus ... done
    Running setup.py install for tornado ... done
    Running setup.py install for backcall ... done
Successfully installed Pillow-7.1.1 PyQtChart-5.14.0 appdirs-1.4.3 backcall-0.1.0 certifi-2020.4.5.1 chardet-3.0.4 colorzero-1.1 cycler-0.10.0 decorator-4.4.2 gpiozero-1.5.1 guizero-1.1.0 idna-2.9 ipykernel-5.2.1 ipython-7.13.0 ipython-genutils-0.2.0 jedi-0.17.0 jupyter-client-6.1.3 jupyter-core-4.6.3 kiwisolver-1.2.0 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.4 numpy-1.18.2 parso-0.7.0 pexpect-4.8.0 pgzero-1.2 pickleshare-0.7.5 pigpio-1.46 prompt-toolkit-3.0.5 ptyprocess-0.6.0 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pygments-2.6.1 pyparsing-2.4.7 pyqt5-5.14.1 pyserial-3.4 python-dateutil-2.8.1 pytz-2019.3 pyzmq-19.0.0 qtconsole-4.3.1 requests-2.23.0 semver-2.9.1 six-1.14.0 tornado-6.0.4 traitlets-4.3.3 urllib3-1.25.9 wcwidth-0.1.9
```

tmontes, 2020-04-29:
Hey @MvGulik...

There's an explanation for that, yes. In short, Mu 1.0.1 is "packaged" in such a way that it strictly requires PyQt5 5.14.1.

Why would it be that strict, could you ask? Locking dependencies strictly helps ensure that all users get the same exact software versions, known to work. That is the case you are observing.

Let me know if this makes sense.

PS: Thanks for using Mu. :)

MvGulik, 2020-04-30:
> There's an explanation for that, yes. In short, Mu 1.0.1 is "packaged" in such a way that it strictly requires PyQt5 5.14.1.

Roger.

The reason I asked was that I encountered some table on some page that _seemed_ to specified PyQt5 5.14.2 as being supported.

(_seemed: Could not find the related page again. Think it was a Mu-site page. But I'm not sure about that at the moment (due to missing proof)_)

(_Feel free to close. Will potentially reopen if I manage to run into the related page again and it makes sense to reopen_)

---

PS: bad/old link.
At page: https://codewith.mu/en/howto/1.0/install_with_python
Old link: `https://codewith.mu/en/howto/install_raspberry_pi`
Probably should be: `https://codewith.mu/en/howto/1.0/install_raspberry_pi`

tmontes, 2020-04-30:
Thanks for the bad-link identification. I just created a PR to fix that.

Closing. Feel free to open other issues for topics that you consider relevant.

closed 2020-04-30 by tmontes