mu-editor/mu #1030: Left/right window tiling/snapping (Linux), mu-minimal-window-width.

MvGulik, 2020-04-20:
On Linux Mint (_although assuming its a general Linux feature across the board, optional/based on desktop settings_)
When dragging a window (_by its title bar_) to one of the side's or corners of the desktop the window would snap to that desktip side, and resize itself appropriate. (_1/2 of desktop along selected side, or 1/4 of desktop size for corners_)

Although this works for the mu-editor for the top and bottom desktop sides, this don' t works for the left and right side (_and all corners of course_).

This seems directly related to the fact that mu-editor has a set-fixed-minimal-width size that is bigger than the new size the window would have after the auto resize. (_ie min > 1/2 desktop related-side size_) (_1600 x 1200 in my case_)

Overall I think its general beneficial (_and consisted_) to be able to use auto-tiling/snapping for all desktop sides.

It seems to make sense to see if that minimum window-with setting could be auto-set/-adjusted based on the current desktop size. (_ie window-min < (or <=) 1/2 desktop related-side size_)

(_I' m sure there is a bit more to it than just this to make auto-tiling/snapping work smoothly across the board, and that some other mu-editor parts (like mu-buttons auto-resizing) need to be taking in account_)

(_Linux Mint 19.3 cinnamon_)

tmontes, 2020-04-20:
@MvGulik,

Thanks for your report. AFAICT, such window sizing limitations have been relieved in the context of #757 with PR #775 which may be already integrated into the current alpha. Not 100% sure.

Can you confirm which version you're running and, if not the alpha, if you get what you want with the alpha release?

MvGulik, 2020-04-20:
Was running the stable `mu-editor 1.0.3` version. (`pip install mu-editor`)

I'm currently not sure about installing/trying the alpha version. The issue being that when it comes to building stuff from source, I'm a bit out my convert/experience zone. (_will give it some additional thoughts/reading-up and see how far I might get._)

MvGulik, 2020-04-20:
Ok, that went smoother that anticipated. :)

Yes. Mu 1.1.0.alpha.2 can be snapped to left and right side of the desktop.

But, in my case (1600 x 1200), it can't be snapped to the top and bottom side of the desktop any more. (minimal mu-window height > ~600)

I myself find left+right more useful than top+bottom. But others might disagree with that. ;-)

tmontes, 2020-04-21:
Glad that you managed to get the alpha up and running smoothly enough.

Regarding your experience, not being able to have the Mu window snapped to the top/bottom-half of the screen, I wonder if that limitation is due to some unforeseen details...

Facts:
* Mu's window size is set to being no smaller than 1/2 the screen width/height.
* The code that does that is here: https://github.com/mu-editor/mu/blob/e3b7191e223a02fbff296fa0ecc834382e16e2b4/mu/interface/main.py#L1043
* If your screen is 1200px in height, Mu's minimal height will be set to 600px.

Thesis:
* Your desktop environment includes some kind of horizontal panel / menu / dock that consumes vertical screen space.
* When asked to snap to top/bottom your desktop environment takes that into account and tries to set the window height to half the screen height *minus* that panel's height.
* Thus, if you have a 1200px tall display with a 100px tall horizontal panel, snapping to top/bottom would have the desktop environment try to set Mu's window height to 550px (half of 1200px - 100px) which is less than the minimum 600px and, by definition, will always be less than half the total screen height that Mu sets as its minimum height.

Could this be what's going on? Maybe.

If you feel adventurous, you may try to locate the code I linked to, but on your installed version, and play a bit with it (you will need to quit/relaunch Mu between attempts!) -- look for a file named `main.py` under a directory structure like `mu/interface/main.py`. It is a "plain text" file that you can edit with Mu itself. :-)

Maybe you could try setting the minimum window height to 1/3rd of the screen height, instead, and see if that makes top/bottom snapping work!

MvGulik, 2020-04-21:
Yea, the fact that the task-bar might play a role here crossed my mind too, but I did not follow up on that idea.

Result of testing that part.
When I move my taskbar from the bottom to the left side. It instantly disables snapping for the mu-window to the left and right side (_as expected_).
However snapping to top and bottom did not become available after this (_not as expected_). Also not after restarting the mu-editor.

This feels a bit like it might be a internal Linux/Mint ">" / "<" vs ">=" / "<=" thing. (_probably a to simplified view_)

O dear.
Ps: I was assuming that seeing a snapping preview also meant that that ment the actually snapping would work.
For `Mu 1.1.0.alpha.2` this seems to be not true. Although I get a snap-preview when I drag the mu-window to the left or right desktop side, the actual snapping don' t works. Ie: when the actual snapping should take place, the Mu-window just stays as it is and don' t changes it position or size as expected. (_tested on a restarted mu-editor_) (_should probably go into a new issue report_)

Source code editing:
I don' t mind taking a look as the mu-source code and tinker a bit with it, although I doubt my python skill level is up to par for any useful results. (_My Linux/Mint skills are unfortunately below what I would call basic level_)
But than again. If one don't tries the new/difficult things, one will not progress. ;-)

MvGulik, 2020-04-21:
Edge snapping behaviour results for `Linux Mint 19.3 cinnamon`. (_with `mu-editor 1.1.0.alpha.2`_)

1) For Left & Right edge: If the minimum width value is exactly half the size of the desktop. It will show a snapping preview, but the window will not actively snap (_window just stays at the same position with the same size_). ie: works properly with `(screen_width // 2) - 1`.

2) For the Top and Bottom edges things are a bit different.
For one it has no intermediate value for which only the snapping preview will be activated, but no actual snapping will be done (_technically this would of course be the default expected behaviour_).
And second. Instead of subtracting one, I needed to subtract 5 to get snapping to activate. (_I don' t have a clue where those extra 4 points are coming from_)

3) To account for a potential taskbar, on top of the above parts, just subtracting half its size worked fine for all edges.

In relation to this I think mu-editor should set its minimal window sizes somewhere below half the desktop size. (_ie: add(_subtract_) some general safety margin_)

_Personally I would just divide by 3 instead of 2 and be done with it._ ;-)

tmontes, 2020-04-29:
@MvGulik,

Please clarify: when you set Mu's window minimum width to 1/3 of the screen width/height then you can snap Mu 1.1.0.alpha2 windows any of the left, right, top, or bottom edges of the screen. Is that it?

Thanks for your assistace and patience. :)

MvGulik, 2020-04-30:
> when you set Mu's window minimum width to 1/3 of the screen width/height then you can snap Mu 1.1.0.alpha2 windows any of the left, right, top, or bottom edges of the screen. Is that it?

Correct.

Fluffkin, 2024-04-25:
> * Mu's window size is set to being no smaller than 1/2 the screen width/height.
>
>     * The code that does that is here: [...]

This isn't a great idea for ultrawide monitor users. (Or presumably ultra tall portrait monitor users).
If a minimum size is a requirement, perhaps take the smallest dimension of width or height, then set half that to be the minimum height AND width for the window?