mu-editor/mu #1031: `pip3 install mu-editor` fails due to PyQtChart/setup.py?

CheeseL0ver, 2020-04-21:
Trying to run `pip3 install mu-editor` fails on different computers with different on MXLinux version 19 (Debian 10.3 (buster) ) for me due to for PyQtChart.
 
I try the following command and I get this error and output:

```
Installing build dependencies ... done
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python3.7/tokenize.py", line 447, in open
        buffer = _builtin_open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/tmp/pip-install-3_otje6c/PyQtChart/setup.py'
```

I searched the past issues for mu and found this issue #626 but that was for Ubuntu 18 and was two years ago. I was in the middle of recording a beginner Python series with the editor, as I love it for beginners, but after upgrading my distro to the current version I have not been able to reinstall. You all have any ideas on what could be wrong?

I appreciate any help you guys can give!

carlosperate, 2020-04-21:
Looking at the trace it looks like this might be an issue with the PyQtChart package itself and not Mu.
Have you tried searching on Google to see if other users are having issues with PyQt and PyQtChart?

A quick search brings this up, it's worth a look and further googling: https://stackoverflow.com/questions/59711301/install-pyqt5-5-14-1-on-linux

carlosperate, 2020-04-21:
Oh yeah, this is very likely related to the post-release changes they did to the PyQt wheels in PyPi (they released 5.12, and then updated the wheel a few months later): https://github.com/mu-editor/mu/issues/958

CheeseL0ver, 2020-04-21:
Thank you all @carlosperate for the link. The excepted solution did not work for me BUT another answer helped me solve it. For me I did the following:

-  I manually uninstalled pyqt5 using pip: `python3.7 -m pip uninstall pyqt5`
- I installed pyqt5 binary: `python3.7 -m pip install pyqt5 --only-binary pyqt5`
- I installed mu: `python3.7 -m pip install mu-editor`

Installing the binary for pyqt5 was enough for the wheel to download the other binaries (I'm guessing) and so it avoided the need for the setup.py which was causing the issue. Thanks for the help!

closed 2020-04-21 by CheeseL0ver

cdesqaz, 2021-03-14:
hi everyone!
I just wondering about this issues, already read every answer, and at the end put attention on my terminal output en run the command with sudo. That's it. Installation run perfectly.
virtual environment created with python3.6:  python3  -m venv mu
source ~/myvenv/mu/bin/activate
sudo pip3 install mu-editor

Successfully installed Pillow-8.1.2 PyQt5-sip-12.8.1 PyQtChart-5.14.0 appdirs-1.4.4 colorzero-1.1 cycler-0.10.0 gpiozero-1.5.1 guizero-1.1.1 ipykernel-5.5.0 jupyter-client-6.1.12 jupyter-core-4.7.1 kiwisolver-1.3.1 matplotlib-2.2.2 mu-editor-1.0.3 nudatus-0.0.5 numpy-1.19.5 pgzero-1.2 pigpio-1.78 pycodestyle-2.4.0 pyflakes-2.0.0 pygame-1.9.6 pyparsing-2.4.7 pyqt5-5.14.1 pyserial-3.4 pytz-2021.1 qscintilla-2.11.4 qtconsole-4.3.1 semver-2.13.0 tornado-6.1
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$  mu-editor
Registrando en /home/masua/.cache/mu/log/mu.log
QApplication: invalid style override 'gtk' passed, ignoring it.
	Available styles: Windows, Fusion
Gtk-Message: 09:39:53.450: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 09:40:11.517: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: 09:47:19.852: GtkDialog mapped without a transient parent. This is discouraged.
QObject::killTimer: Timers cannot be stopped from another thread
QObject::~QObject: Timers cannot be stopped from another thread
QThread: Destroyed while thread is still running
Abortado (`core' generado)
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$ [IPKernelApp] WARNING | Parent appears to have exited, shutting down.
^C
(mu) masua@w2l-pc:~/0l2p0/Python/ide/mu$

carlosperate, 2021-03-15:
Hi @cdesqaz, this looks like a different problem. Could you open a new issue with all the things that you've tried and the terminal output?

cdesqaz, 2021-03-16:
> Hi @cdesqaz, this looks like a different problem. Could you open a new issue with all the things that you've tried and the terminal output?

Sorry my bad. 
This It were my first failed terminal output:
pip3 install mu-editor ==>

Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-3jbwjd_l/PyQtChart/'

carlosperate, 2021-03-16:
@cdesqaz if you could open a new issue that will help us triage it, otherwise it will be missed as this issue is currently closed.