mu-editor/mu #1034: Error displaying image in Pygame Zero Mode on some Windows 10

abixadamj, 2020-04-29:
Hi, I teach teachers Pygame Zero, and today some of them (2 from group of 10) has reported strange behavior when running simple app:

```
TITLE = "Moja gra ;-)"
WIDTH = 640
HEIGHT = 426

def draw():
    screen.blit("tlo.jpg", (0, 0))
```
Of course there was tlo.jpg file located in images folder with size 640x426.

```
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "runpy.py", line 193, in _run_module_as_main
  File "runpy.py", line 85, in _run_code
  File "C:\Program Files\Mu\pkgs\pgzero\__main__.py", line 3, in <module>
    main()
  File "C:\Program Files\Mu\pkgs\pgzero\runner.py", line 93, in main
    run_mod(mod)
  File "C:\Program Files\Mu\pkgs\pgzero\runner.py", line 113, in run_mod
    PGZeroGame(mod).run()
  File "C:\Program Files\Mu\pkgs\pgzero\game.py", line 217, in run
    self.mainloop()
  File "C:\Program Files\Mu\pkgs\pgzero\game.py", line 256, in mainloop
    draw()
  File "gra_1.py", line 9, in draw
    screen.blit("tlo.jpg", (0, 0))
  File "C:\Program Files\Mu\pkgs\pgzero\screen.py", line 100, in blit
    image = loaders.images.load(image)
  File "C:\Program Files\Mu\pkgs\pgzero\loaders.py", line 120, in load
    self.validate_root(name)
  File "C:\Program Files\Mu\pkgs\pgzero\loaders.py", line 102, in validate_root
    subpath=self.subpath, type=self.TYPE, name=name
KeyError: "No 'images' directory found to load image 'tlo.jpg'."
---------- FINISHED ----------
```

I cannot get logs now, maybe tommorow...

tjguk, 2020-04-30:
This used to be because the images/ directory displayed by the [Images] button is relative to mu_code/ but pgzero is looking for it relative to the .py file yoiu're running. If these were different there was a problem. But that was fixed a while ago.... I think ;)

tjguk, 2020-05-02:
@abixadamj do we have any more info on this. I assume it was a passing bug with one student which is perhaps now hard to reproduce. If you do have the change, though, we could do with the logs and/or the full path to the code and to the "images" folder which opens when you click the [Images] button. Thanks

abixadamj, 2020-05-04:
@tjguk - I was exactly this... Mu-editor 1.0.3 - Today I had the same situation on Linux version. But thanks to you, I know how this should be repair ;-)

closed 2020-05-04 by abixadamj