mu-editor/mu #1039: Mu don't start on MAC OSX El capitan version 10.11.6

Franci85, 2020-05-09:
Hello, i installed MU on MAC OSX El capitan version 10.11.6 following the guide (https://codewith.mu/en/howto/1.0/install_macos) performing specially the task:
Step 5 - Start Mu
Open your Applications folder - click “Go”, “Applications” on the Finder menu.
Press the CTRL button and click on the Mu icon. You’ll see a dialog to check you want to open Mu. Click “Open”

MU application does not start, the icon Mu appears on the dock make small jumps and then disappears.
I check also the following guide [https://support.apple.com/en-in/guide/mac-help/mh40616/mac](https://support.apple.com/en-in/guide/mac-help/mh40616/mac).
On my system on Security & Privacy i switch from App Store and identified developers to everywhere.

Thank you in advance who will be able to give me support.
Ciao

d-roni, 2020-05-12:
Ditto, what Franci85 said! I am experiencing the same problem. My MU editor application doesn't start, even though I've followed the installation instructions exactly. I even tried d/loading the latest Alpha version [mu-editor_1.1.0-alpha.2_osx] from the Mu site [https://codewith.mu/en/download] to see (as suggested in another MU thread), but still encountering the same problem after installation. I'm also using MAC OSX El Capitan version 10.11.6. Any help would be greatly appreciated!! Thank you!

Choilly, 2020-05-24:
That's exactly the same problem I'm having!
I'm also using Mac OS X El Capitan version 10.11.6.
Is it the software itself having the problem? I've followed the instruction and tried so many other ways to open the Mu editor but none of them never worked :( 
The icon just pops once and disappears!!! SOOO FRUSTRATING!!!
Please help..

iraytrace, 2020-06-10:
I can confirm as well.  It looks like a packaging / build error.  Starting the application from a terminal window results in a complaint that it is trying to load things from /Users/travis/...  when there is no /Users/travis on the machine (my username isn't travis, nor is there such a user on my machine).

Specifically, open a terminal window
$ /Applications/mu-editor.app/Contents/MacOS/mu-editor

Traceback (most recent call last):
  File "./mu-editor", line 20, in <module>
    load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')()
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 480, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2324, in load
  File "/Users/travis/build/mu-editor/mu_portable_python_macos/python/lib/python3.6/site-packages/pkg_resources/__init__.py", line 2330, in resolve
  File "/Applications/mu-editor.app/Contents/Resources/app/mu/app.py", line 29, in <module>
    from PyQt5.QtCore import QTimer, Qt
ImportError: dlopen(/Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/QtCore.abi3.so, 2): Symbol not found: __os_activity_create
  Referenced from: /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore (which was built for Mac OS X 10.13)
  Expected in: /usr/lib/libSystem.B.dylib
 in /Applications/mu-editor.app/Contents/Resources/app_packages/PyQt5/Qt/lib/QtCore.framework/Versions/5/QtCore