mu-editor/mu #1046: REPL doesn’t have exit or quit functions

merwok, 2020-05-21:
Hello!  Many people call `exit()` or `quit()` at the regular Python prompt rather than `Ctrl-D` or other key combos on other operating systems.  The Mu console doesn’t have `exit` (also tried `import site` to have them added but that’s not possible).

dybber, 2020-05-23:
It does work in Python 3 mode for me (Mac OS X, Mu 1.1.0 alpha 2), which mode are you referring to, and what would you expect to happen, when calling `quit()`?

merwok, 2020-05-23:
I see that regular Python mode has IPython!  I have edited the title to mention that this is the CircuitPython REPL (in the Serial output pane).

dhalbert, 2020-05-23:
This is a function of CircuitPython and doesn't really have anything to do with Mu. Feel free to open an issue in https://github.com/adafruit/circuitpython with what you think it should do. Since there's no operating system and no shell to fall back, I'm not sure what `exit()` or `quit()` should do other than be an alias for ctrl-d.

merwok, 2020-05-23:
Yes, the same behaviour as Ctrl-D is what’s requested!

merwok, 2020-05-23:
Request moved one level upstream https://github.com/adafruit/circuitpython/issues/2950

closed 2020-05-23 by merwok