mu-editor/mu #1050: Version conflict with pyflakes and flake8

GiulioCentorame, 2020-05-28:
Not sure if a bug or not.
I installed the dev version of mu (1.10.a3) on Linux Manjaro using `git clone`, following the [quick start guide](https://mu.readthedocs.io/en/latest/). If I run `mu-editor` I get:

```
Traceback (most recent call last):

  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 584, in _build_master
    ws.require(__requires__)
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 901, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 792, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (pyflakes 2.1.1 (/home/giuliocentorame/.local/lib/python3.8/site-packages), Requirement.parse('pyflakes<2.3.0,>=2.2.0'), {'flake8'})

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/giuliocentorame/.local/bin/mu-editor", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3253, in <module>
    def _initialize_master_working_set():
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3236, in _call_aside
    f(*args, **kwargs)
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3265, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 586, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 599, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/home/giuliocentorame/.local/lib/python3.8/site-packages/pkg_resources/__init__.py", line 792, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (pyflakes 2.1.1 (/home/giuliocentorame/.local/lib/python3.8/site-packages), Requirement.parse('pyflakes<2.3.0,>=2.2.0'), {'flake8'}) 

```

whereas with `python run.py` it launches without any problem.

Here's also mu's log:

```
2020-05-28 16:57:31,908 - root:122(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2020-05-28 16:57:31,908 - root:123(run) INFO: uname_result(system='Linux', node='giulio-pc', release='4.19.122-1-MANJARO', version='#1 SMP Sun May 10 14:02:27 UTC 2020', machine='x86_64', processor='')
2020-05-28 16:57:31,908 - root:124(run) INFO: Python path: ['/home/giuliocentorame/mu', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/giuliocentorame/.local/lib/python3.8/site-packages', '/home/giuliocentorame/mu', '/usr/lib/python3.8/site-packages', '/home/giuliocentorame/.local/share/mu/site-packages', '/home/giuliocentorame/.local/lib/python3.8/site-packages/IPython/extensions']
2020-05-28 16:57:31,908 - root:125(run) INFO: Language code: en_GB
2020-05-28 16:57:32,005 - mu.logic:644(__init__) INFO: Setting up editor.
2020-05-28 16:57:32,005 - mu.logic:666(__init__) INFO: Settings path: /home/giuliocentorame/.local/share/mu/settings.json
2020-05-28 16:57:32,005 - mu.logic:667(__init__) INFO: Session path: /home/giuliocentorame/.local/share/mu/session.json
2020-05-28 16:57:32,006 - mu.logic:668(__init__) INFO: Log directory: /home/giuliocentorame/.cache/mu/log
2020-05-28 16:57:32,006 - mu.logic:669(__init__) INFO: Data directory: /home/giuliocentorame/.local/share/mu
2020-05-28 16:57:32,027 - mu.logic:682(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2020-05-28 16:57:32,142 - mu.logic:1378(change_mode) INFO: Workspace directory: /home/giuliocentorame/mu_code
2020-05-28 16:57:32,142 - mu.logic:746(restore_session) INFO: Restoring session from: /home/giuliocentorame/.local/share/mu/session.json
2020-05-28 16:57:32,142 - mu.logic:747(restore_session) DEBUG: {'theme': 'day', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': 137, 'y': 77, 'w': 1092, 'h': 614}}
2020-05-28 16:57:32,142 - mu.logic:768(restore_session) INFO: Loaded files.
2020-05-28 16:57:32,142 - mu.logic:771(restore_session) INFO: User defined environment variables: []
2020-05-28 16:57:32,142 - mu.logic:777(restore_session) INFO: Minify scripts on micro:bit? False
2020-05-28 16:57:32,161 - mu.logic:1378(change_mode) INFO: Workspace directory: /home/giuliocentorame/mu_code
2020-05-28 16:57:32,223 - mu.logic:813(restore_session) INFO: Starting with blank file.
2020-05-28 16:57:35,414 - mu.logic:1266(show_admin) INFO: Showing admin with logs from /home/giuliocentorame/.cache/mu/log/mu.log
2020-05-28 16:57:35,415 - mu.logic:200(installed_packages) INFO: Packages found: []
```
and the `make check` summary:

```
tests/modes/test_microbit.py::test_flash_device_has_latest_firmware_encounters_serial_problem_unix
tests/modes/test_microbit.py::test_flash_with_attached_device_has_latest_firmware_encounters_problem
tests/modes/test_microbit.py::test_flash_with_attached_device_has_latest_firmware
tests/modes/test_microbit.py::test_flash_device_has_latest_firmware_encounters_serial_problem_windows
tests/modes/test_microbit.py::test_flash_with_attached_known_device_and_forced
tests/modes/test_microbit.py::test_flash_forced_with_attached_device_as_not_windows
tests/modes/test_microbit.py::test_flash_force_with_attached_device_as_windows
tests/modes/test_microbit.py::test_force_flash_user_specified_device_path
tests/modes/test_microbit.py::test_force_flash_empty_script
  /home/giuliocentorame/mu/mu/modes/microbit.py:271: DeprecationWarning: Function 'semver.parse' is deprecated. Deprecated since version 2.10.0.  This function will be removed in semver 3. Use the respective 'semver.VersionInfo.parse' instead.
    semver.parse(version_info["release"])

tests/modes/test_microbit.py: 10 tests with warnings
  /home/giuliocentorame/mu/mu/modes/microbit.py:289: DeprecationWarning: Function 'semver.compare' is deprecated. Deprecated since version 2.10.0.  This function will be removed in semver 3. Use the respective 'semver.VersionInfo.compare' instead.
    semver.compare(board_version, uflash.MICROPYTHON_VERSION)

tests/interface/test_panes.py::test_JupyterREPLPane_set_theme_night
  /home/giuliocentorame/.local/lib/python3.8/site-packages/qtconsole/jupyter_widget.py:561: DeprecationWarning: JupyterREPLPane._style_sheet_changed is deprecated in traitlets 4.1: use @observe and @unobserve instead.
    def _style_sheet_changed(self):

-- Docs: https://docs.pytest.org/en/latest/warnings.html

----------- coverage: platform linux, python 3.8.2-final-0 -----------
Name                         Stmts   Miss  Cover   Missing
----------------------------------------------------------
mu/__init__.py                  13      0   100%
mu/app.py                       80      0   100%
mu/debugger/client.py          200      0   100%
mu/debugger/runner.py          252      1    99%   90
mu/debugger/utils.py            11      0   100%
mu/interface/__init__.py         2      0   100%
mu/interface/dialogs.py        280      0   100%
mu/interface/editor.py         271      0   100%
mu/interface/main.py           596      0   100%
mu/interface/panes.py          753      0   100%
mu/interface/themes.py         198      0   100%
mu/logic.py                    783      0   100%
mu/modes/__init__.py             8      0   100%
mu/modes/api/__init__.py         9      0   100%
mu/modes/api/adafruit.py         1      0   100%
mu/modes/api/esp.py              1      0   100%
mu/modes/api/flask.py            1      0   100%
mu/modes/api/microbit.py         1      0   100%
mu/modes/api/pi.py               1      0   100%
mu/modes/api/pygamezero.py       1      0   100%
mu/modes/api/python3.py          1      0   100%
mu/modes/api/shared.py           1      0   100%
mu/modes/base.py               222      0   100%
mu/modes/circuitpython.py       57      0   100%
mu/modes/debugger.py           158      0   100%
mu/modes/esp.py                 94      0   100%
mu/modes/microbit.py           295      0   100%
mu/modes/pygamezero.py          63      0   100%
mu/modes/python3.py            168      0   100%
mu/modes/web.py                103      0   100%
mu/resources/__init__.py        15      0   100%
----------------------------------------------------------
TOTAL                         4639      1    99%

```
I haven't found any error in the programme so far, but I am just wondering if it's normal behaviour or it ended up in some dependency hell. Cheers!

bletvaska, 2020-06-02:
to simply solve this issue, comment the lines `32` and `34` of file `setup.py`

GiulioCentorame, 2020-06-02:
That solves it, thank you!

closed 2020-06-02 by GiulioCentorame