mu-editor/mu #1051: Mu editor won't start [Windows 10]

ABridger56, 2020-05-30:
Hello all,

New face here. I'm reading through "Automate the Boring Stuff" and using Mu on the side to get some coding reps in as I read about them. 

I can't figure out why, but today, Mu won't start when I click on it.

I've tried opening it regularly, running it as Administrator; I've tried restarting my computer, restarting the install process for Mu, and I've even tried fully un-installing and re-installing Mu, with no luck. 

I expected the program to open normally, as it has in the days before, but no luck. Today, I can't get Mu to start.

I can't really learn how to write in Python without writing scripts, so this is a bit problematic. (I'm trying to start a new line of text for each * in the bug writing guideline) 

I'm running Windows 10, on 64-bit OS (although I'm pretty sure all Windows 10 are now on x64 but that's irrelevant), and running the installer named "mu-editor_1.1.0-alpha.2_win64.exe" 

I can't attach a log file for Mu because, simply, the error is getting Mu to load, not a specific coding problem.

Any help would be greatly appreciated - I'm only trying to learn.

Cheers

__________________________________________________________________________

If you are reporting a bug, we would like to know:

* What you were trying to do,
* What steps you took to make this happen,
* What you expected to happen,
* What actually happened,
* Why this difference is problematic (it may not be a bug!),
* Technical details like the version of Mu you're using, your OS version and
  other aspects of the context in which Mu was running. 

Please remember to attach a **copy of the full log files for Mu**. You can get
the logs by clicking on the cog icon in the bottom right of the editor window.
Click on the logs and use CTRL-A to select all, then CTRL-C to copy and CTRL-V
to paste the contents into the issue.

Thank you for contributing to Mu! :-)

federicomm, 2020-06-06:
[image]
[image]
I have done the same installation and also the python master-mu and all two ( the exe and the python are running) 
The following is a copy of mu.log .
I'll check later the exe installation hoping to find the info that can help you to solve the problem
2020-06-05 22:22:57,949 - root:122(run) INFO: 

-----------------

Starting Mu 1.1.0.alpha.2
2020-06-05 22:22:57,949 - root:123(run) INFO: uname_result(system='Windows', node='Monaldi', release='10', version='10.0.17134', machine='AMD64', processor='Intel64 Family 6 Model 69 Stepping 1, GenuineIntel')
2020-06-05 22:22:57,949 - root:124(run) INFO: Python path: ['C:\\Users\\federico\\ESP32_uPython\\mu-master', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\python37.zip', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\DLLs', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32', 'C:\\Users\\federico\\AppData\\Roaming\\Python\\Python37\\site-packages', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\site-packages', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\site-packages\\win32', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\site-packages\\win32\\lib', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\site-packages\\Pythonwin', 'C:\\Users\\federico\\AppData\\Local\\python\\mu\\site-packages', 'C:\\Users\\federico\\AppData\\Local\\Programs\\Python\\Python37-32\\lib\\site-packages\\IPython\\extensions']
2020-06-05 22:22:57,949 - root:125(run) INFO: Language code: it_IT
2020-06-05 22:22:58,090 - mu.logic:644(__init__) INFO: Setting up editor.
2020-06-05 22:22:58,090 - mu.logic:666(__init__) INFO: Settings path: C:\Users\federico\AppData\Local\python\mu\settings.json
2020-06-05 22:22:58,090 - mu.logic:667(__init__) INFO: Session path: C:\Users\federico\AppData\Local\python\mu\session.json
2020-06-05 22:22:58,090 - mu.logic:668(__init__) INFO: Log directory: C:\Users\federico\AppData\Local\python\mu\Logs
2020-06-05 22:22:58,090 - mu.logic:669(__init__) INFO: Data directory: C:\Users\federico\AppData\Local\python\mu
2020-06-05 22:22:58,168 - mu.logic:682(setup) INFO: Available modes: python, circuitpython, microbit, esp, web, debugger, pygamezero
2020-06-05 22:22:58,559 - mu.logic:1379(change_mode) INFO: Workspace directory: C:\Users\federico\mu_code
2020-06-05 22:22:58,559 - mu.logic:746(restore_session) INFO: Restoring session from: C:\Users\federico\AppData\Local\python\mu\session.json
2020-06-05 22:22:58,559 - mu.logic:747(restore_session) DEBUG: {'theme': 'night', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': -1, 'y': -8, 'w': 1366, 'h': 705}}
2020-06-05 22:22:58,559 - mu.logic:768(restore_session) INFO: Loaded files.
2020-06-05 22:22:58,559 - mu.logic:773(restore_session) INFO: User defined environment variables: []
2020-06-05 22:22:58,559 - mu.logic:779(restore_session) INFO: Minify scripts on micro:bit? False
2020-06-05 22:22:58,606 - mu.logic:1379(change_mode) INFO: Workspace directory: C:\Users\federico\mu_code
2020-06-05 22:22:58,684 - mu.logic:813(restore_session) INFO: Starting with blank file.
2020-06-05 22:23:04,512 - mu.modes.base:269(find_device) INFO: Found device on port: COM15
2020-06-05 22:23:04,512 - mu.modes.base:270(find_device) INFO: Serial number: 
2020-06-05 22:23:04,527 - mu.modes.esp:199(toggle_files) INFO: Toggle filesystem on.
2020-06-05 22:23:43,275 - mu.interface.panes:478(dropEvent) INFO: Getting 'rgb_hex.py' from micro:bit. Copying to 'C:\Users\federico\mu_code\rgb_hex.py'.
2020-06-05 22:24:21,522 - mu.modes.esp:203(toggle_files) INFO: Toggle filesystem off.
2020-06-05 22:24:22,584 - mu.modes.base:269(find_device) INFO: Found device on port: COM15
2020-06-05 22:24:22,584 - mu.modes.base:270(find_device) INFO: Serial number: 
2020-06-05 22:24:22,600 - mu.modes.base:327(add_repl) INFO: Started REPL on port: COM15
2020-06-05 22:24:22,600 - mu.modes.base:307(toggle_repl) INFO: Toggle REPL on.
2020-06-05 22:25:30,212 - mu.logic:1242(quit) DEBUG: Session: {'theme': 'night', 'mode': 'esp', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2, 'window': {'x': -1, 'y': -8, 'w': 1366, 'h': 705}}
2020-06-05 22:25:30,212 - mu.logic:1243(quit) DEBUG: Saving session to: C:\Users\federico\AppData\Local\python\mu\session.json
2020-06-05 22:25:30,212 - mu.logic:1257(quit) INFO: Quitting.

federicomm, 2020-06-06:
[image]
[image]
[Win10_set.txt](https://github.com/mu-editor/mu/files/4739802/Win10_set.txt)
Attached you can find a picture showing the mu.log ( check. the first lines relative to various env paths)  and a txt file showing the environment set of my Win10 .
I hope that these new info can help you to solve the Mu problem.

dafafachrudin, 2020-07-17:
This is the exact same problem that i have, even the same book. Im completely new to this so if someone can explain whats going on it would be great.

federicomm, 2020-07-18:
I can only integrate my previous post about this issue with the pictures
showing my Windows 10 and Mu 1.1.0.alpha.2 configuration used by the
executable version and also by the mu-muster version.
Hoping that this can be of some help.

dafafachrudin, 2020-07-18:
> I can only integrate my previous post about this issue with the pictures showing my Windows 10 and Mu 1.1.0.alpha.2 configuration used by the executable version and also by the mu-muster version. Hoping that this can be of some help.
> […](#)

Im completely new with coding so i have no idea what to do with your pictures

federicomm, 2020-07-18:
Can you detail the steps that you have followed to install the Mu
1.1.0.alpha.2 (executable ? ) on Windows 10 ?

federicomm, 2020-07-18:
The correct procedure to install Mu on Windows can be found at:

https://codewith.mu/en/howto/1.0/install_windows

How to install Mu on Windows with the Official Installer.

Have you followed this procedure?

dafafachrudin, 2020-07-18:
I followed it exactly, if someone can provide an alternative than please tell me, i really wanna continue learning

federicomm, 2020-07-18:
1st -I also have followed the Web procedure to install Mu on Windows 10. To
understand where is the problem, in my opinion, we need to locate (  if
they exist) the significant differences between your and my installion (
for this reason I have posted some pictures in the previous email). For
example, from the cmd shell, if you type python wich is the output?
You must have as output the python version ( determined by the Windows 10
environment vars PATH).
2nd - I don't know if exists a different procedure to install Mu.

federicomm, 2020-07-19:
I can understand your disappointment, but I can't help you more than that.

 I uninstalled the Mu program from Windows 10 which worked perfectly on my
PC.
 So I started a new installation of Mu using the following file:

  mu_editor_1.1.0-alpha.2_win64.exe

 The only requirement that I selected during the installation process is:
Install only for me

 All other requirements have remained the default ones.

 At the end of the installation process I launched Mu;  it ran perfectly
without problems.

ABridger56, 2020-07-19:
> Hi @ABridger56,
> 
> Thanks for the bug report! I'm sorry you are having issues opening Mu, that sounds like an odd issue indeed.
> When you uninstalled and reinstalled Mu, were you installing the same version you had installed before? Were you running 1.1.0-alpha.2 successfully before the issue started?
> Have you encountered similar issues with any other software on your system?
> 
> Thanks @federicomm as well for the additional information 👍
> @ABridger56 if you could find the log from the Windows paths as indicated by @federicomm and attach the file here it could help us identify the issue.

Hi Carlos,

thanks for the reply. I'm not sure how to record the log from the windows paths in the same manner that @federicomm did; It looks like the log might be through Mu, which I wouldn't be able to access. If there's a way to record the log without using Mu that I'm not aware of, let me know & I'll try to post that soon.

In reply to your question about which version I installed - yes, I was running the alpha version successfully before the issue started. I could run it for a few days (I can't remember exactly how long now as it was a few weeks ago), and one morning, without changing anything (to my knowledge), it wouldn't run. I tried re-installing the alpha version, as well as un-installing that and installing the more stable version. Unfortunately, no luck.

federicomm, 2020-07-20:
I don't know if the following can be of some help , in other words if we
can have a Traceback for the posted case, anyway :

To verify if we can have an error log output in the Win 10 cmd line screen,
 I have introduced at the start of the run() function in the app.py module
a phantom module aaa

From Windows 10 cmd line shell I have selected the dir where the mu is
stored(*) :
C:\Users\federico\AppData\Local\Mu\pkgs\mu

1-type:
python

2-from the python prompt :
>>>from app import run

3-then
>>>run()

the result was:

C:\Users\federico\AppData\Local\Mu\pkgs\mu>python
Python 3.7.1rc2 (v3.7.1rc2:6c06ef7dc3, Oct 13 2018, 14:49:27) [MSC v.1914
32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from app import run
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\federico\AppData\Local\Mu\pkgs\mu\app.py", line 121, in run
    import aaa # Error - only to test the output on Win10 cmd line screen
ModuleNotFoundError: No module named 'aaa'
>>>

(*) this was the installer default choice.
The user name is clearly a variable.

federicomm, 2020-07-20:
The  suggestion of my previous post is possible  because I have various
python dir set in the Win10 environment vars for mu-muster.
For the only Mu 1.1.0.alpha.2 executable ( that use python3.6 ) the
following command should be executed in the cmd shell to show ( if any)
errors.

C:\Users\federico\AppData\Local\Mu>Python\python.exe Mu.launch.pyw

Traceback (most recent call last):
  File "Mu.launch.pyw", line 32, in <module>
    run()
  File "pkgs\mu\app.py", line 121, in run
    import aaa # phantom module
ModuleNotFoundError: No module named 'aaa'

C:\Users\federico\AppData\Local\Mu>

On 20 Jul 2020 13:43, "Federico Monaldi" <federicomonaldi72@gmail.com>
wrote:

I don't know if the following can be of some help , in other words if we
can have a Traceback for the posted case, anyway :

To verify if we can have an error log output in the Win 10 cmd line screen,
 I have introduced at the start of the run() function in the app.py module
a phantom module aaa

From Windows 10 cmd line shell I have selected the dir where the mu is
stored(*) :
C:\Users\federico\AppData\Local\Mu\pkgs\mu

1-type:
python

2-from the python prompt :
>>>from app import run

3-then
>>>run()

the result was:

C:\Users\federico\AppData\Local\Mu\pkgs\mu>python
Python 3.7.1rc2 (v3.7.1rc2:6c06ef7dc3, Oct 13 2018, 14:49:27) [MSC v.1914
32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from app import run
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\federico\AppData\Local\Mu\pkgs\mu\app.py", line 121, in run
    import aaa # Error - only to test the output on Win10 cmd line screen
ModuleNotFoundError: No module named 'aaa'
>>>

(*) this was the installer default choice.
The user name is clearly a variable.

federicomm, 2020-07-21:
I suspect that the problem is linked to the Windows 10 background updates. It is possible to restore the Windows 10 state at the time when Mu was OK?

carlosperate, 2020-07-21:
@ABridger56 
 
> Hi @carlosperate ,
> 
> thanks for the reply. I'm not sure how to record the log from the windows paths in the same manner that @federicomm did; It looks like the log might be through Mu, which I wouldn't be able to access. If there's a way to record the log without using Mu that I'm not aware of, let me know & I'll try to post that soon.
> 
> In reply to your question about which version I installed - yes, I was running the alpha version successfully before the issue started. I could run it for a few days (I can't remember exactly how long now as it was a few weeks ago), and one morning, without changing anything (to my knowledge), it wouldn't run. I tried re-installing the alpha version, as well as un-installing that and installing the more stable version. Unfortunately, no luck.

The log should be located in `C:\Users\<YOUR_USERNAME>\AppData\Local\python\mu\Logs` (just replacec `<YOUR_USERNAME>` with your Windows username).

> I suspect that the problem is linked to the Windows 10 background updates. It is possible to restore the Windows 10 state at the time when Mu was OK?

I'm not familiar with the latest Windows updates, I haven't heard of anything that should have affected Mu. Do you know of anything that is included that could be causing issues?

federicomm, 2020-07-22:
Mine is a supposition based on the fact that the failure to run Mu is due
to the inability to import run from app.py.
As result we can't have  an mu.log file

In my opinion,  put simply, the problem isn't in the python code that needs
to be run - it's previous.  We have two executables in the Python dir,
pythonw.exe and python.exe
 My guess is that for security reasons Windows 10 may have introduced some
form of control over the DLLs ( vcruntime140.dll for example ), which if
not satisfied interrupts the execution of the program.
 And it is to empirically verify this hypothesis that I asked (to the
author of the first post) if it was possible to restore the state of
Windows 10. It is in fact the timing of the phenomenon - before Mu was
loaded normally and then - suddenly it is impossible to do it,  that
justifies my suspicion.
 In my case, in order to avoid the background updates of Windows 10, the
connection to the network of my PC can only take place through the enabling
by me -Tethering Mobile hotspot - of my smartphone -

federicomm, 2020-07-25:
A simple and quick method to confirm or refute the hypothesis that the two executables pythonw.exe and phython.exe do not complete their execution and stop before to process python code, is the following ( see the attachment 0000.png):
 1- cmd.exe
 2- cd \
 3- cd C: \ Users \ <your username> \ AppData \ Local \ Mu \ Python \
 4- python.exe
 if all is OK you must have opened a python console - (in the attachment 0000_a.png Mu positive  run from cmd shell- in my case!)
 if not the python.exe stopped for a reason that must be discovered.
[image]
[image]

ABridger56, 2020-07-27:
Hi @federicomm ,

I just tried re-installing Mu; afterwards, I tried entering "cd Users\(my name)\AppData\Local\Mu\Python", as well as "cd Users\(my name)\AppData\Local\Mu\Python\python.exe"

The command prompt returned "The system cannot find the path specified" for both commands

federicomm, 2020-07-27:
Hi Bridger56,
the output or error message  that you have can only be justified by a
different Mu location of your installation.
In the 4 attachments the sequence, that I have followed to verify if the
console python executable does not stop before to process the first line of
python code, is detailed.

federicomm, 2020-07-28:
Hi @ABridger56

Because the file Mu.launch.pyw is only in the Mu installation directory,
you can find the Mu installation path from the Windows 10 command prompt:

Microsoft Windows [Versione 10.0.17134.1304]
(c) 2018 Microsoft Corporation. Tutti i diritti sono riservati.

------- long time search for Mu installation DIR -----

C:\Windows\System32>where /R C:\ Mu.launch.pyw
C:\Users\federico\AppData\Local\Mu\Mu.launch.pyw
C:\Windows\System32>

----- short time - if you know that Mu is installed in the Users dir tree
-----

C:\Windows\System32>where /R C:\Users Mu.launch.pyw
C:\Users\federico\AppData\Local\Mu\Mu.launch.pyw
C:\Windows\System32>

federicomm, 2020-07-28:
Hi @ABridger56
Rereading  your message  :
I tried entering "cd Users(my name)\AppData\Local\Mu\Python", as well as
"cd Users(my name)\AppData\Local\Mu\Python\python.exe"

I think that from the command prompt we should  use :
cd C:\Users\(my name)\AppData\Local\Mu\Python", as well as "cd C:\Users\(my
name)\AppData\Local\Mu\Python\python.exe

federicomm, 2020-07-28:
Errata-corrige
I think that from the command prompt we should  use :

cd C:\Users\(my name)\AppData\Local\Mu\Python
to enter in the Python dir

 and ,and,  to run the console python:
 C:\Users\(my name)\AppData\Local\Mu\Python\python.exe

On 28 Jul 2020 20:15, "Federico Monaldi" <federicomonaldi72@gmail.com>
wrote:

Hi @ABridger56
Rereading  your message  :
I tried entering "cd Users(my name)\AppData\Local\Mu\Python", as well as
"cd Users(my name)\AppData\Local\Mu\Python\python.exe"

I think that from the command prompt we should  use :
cd C:\Users\(my name)\AppData\Local\Mu\Python", as well as "cd C:\Users\(my
name)\AppData\Local\Mu\Python\python.exe

federicomm, 2020-07-31:
Windows 10 work-around to run Mu-

I have done this installation on my Window 10 platform - the result is that
I can run Mu also in this mode.

Remark !! The default Mu executable installation( thanks to Carlos Pereira
Atencio) remains and is the best one. This must be considered  only
a temporary work-around if we face problems with the default Mu
installation .

Take into consideration that the until now unknown reason of the failure to
run Mu with the proper installation,
doesn't guarantee the success of the proposed work-around on your Windows
10 platform.

1- Install Python v3.6.3 64bit ( clearly if your previous Mu istallation
was a 64 bit version )
   ( don't change the default settings)

2- check the Windows 10 Path environment variable  to be sure that python36
path  is included ( before any
   other - if exist - python installation )
3- Test the python36 opening the cmd shell ( the black icon on your
Desktop) and typing python after
   the shell promt C:\Windows\System32> you should see the following output:

Microsoft Windows [Versione 10.0.17134.1304]
(c) 2018 Microsoft Corporation. Tutti i diritti sono riservati.

C:\Windows\System32>python
Python 3.6.3 (v3.6.3:2c5fed8, Oct  3 2017, 18:11:49) [MSC v.1900 64 bit
(AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>

4 - If the step 3 is OK,  exit from the python console typing  after the
python prompt >>>
    exit()

5 - If the test in the step 3 is positive continue with the step 6,
otherwise stop any action
    If you stop here, that means ( in my opinion)  that the problem is in
your Windows 10 platform !!!!
6 - Using the FileManager, from the location where the default Mu has been
installed
    ( C:\User\your_user_name\AppData\Local\Mu )
    enter in the dir
C:\Users\federico\AppData\Local\Mu\pkgs
select all its files and copy them in the python36 dir  :

C:\Users\your_user_name\AppData\Local\Program\Python\Python36\lib\site-package

Why the step 6 ?
Because we have all the correct  mu-editor dependencies(*) for the python -
3.6.3. 64bi - version in the dir:
C:\Users\federico\AppData\Local\Mu\pkgs.

To make clear each step I have added some autoexplaining pictures.

(*) use : pip show mu-editor
to know its  the requirements.